sebanc/brunch #1: brunch: ChromeOS was not found on any device

DiogoSilva48, 2020-02-07:
The image creation went fine but after booting I get this issue, I guess wasn't supposed to happen.
[image]

sebanc, 2020-02-07:
Hi,

Thank you for giving it a try, I am actually looking for testers.

It seems like your CPU is not compatible with the "octopus" recovery image which I currently use during the build process. Could you please provide me with your pc/laptop configuration details ?

Also, I see that on your github that you are familiar with croissant project, is there a recovery image that worked with your computer ? (note: at this point using another recovery image will give the exact same result as the issue lies within the initramfs)

DiogoSilva48, 2020-02-07:
Hi, I'm happy to help you test, for sure. I'm on the Project Croissant team actually.

I used to use Eve mainly but since some builds ago I had to change to Atlas, which works so so,  because eve stopped booting. And to be fair Octopus never booted with croissant.

Also, my laptop is a bit old i5-3230M, also a Nvidia 710M (which seems irrelevant for ChromeOS but sometimes can cause problems 😅).

sebanc, 2020-02-07:
I can confirm that octopus works fine on both my Surface Pro 5 and Surface Go (only devices I own) but it may not be the most compatible image (as I hoped it would be).

I was also using octopus because (as far I tested) it has the widest gpt partitions, which means that you can use any image during install. However, there are other ways to create the gpt partition so we can use another recovery image as the base and just change the gpt partition creation process.

I still think unibuild versions are better as we can define ensure all hardware is supported (notably sensors).

With atlas, as the device is not convertible, there are chances that the tablet mode / accelerometer will not be supported. (To be confirmed)

So, my global opinion on this would be to try finding a unibuild image with better compatibility (maybe coral be more compatible).

With that said, the build process should work with any recovery image, so you can try building the framework manually with atlas. Else, I should be able to upload a test image based on atlas during the weekend.

sebanc, 2020-02-07:
Actually, have you already tried the coral recovery image on your device ?

DiogoSilva48, 2020-02-07:
I think Coral wasn't booting with croissant as well, lately a lot of recoveries stopped booting for some reason. I also have tried to boot Atlas with your kernel but still installing manually to the rootA and didn't boot as well, same issue happened with ATB and not FydeOS. Something weird is happening .
BTW, do you have discord, telegram or any instant messaging app, I feel kinda awkward talking here 😂

sebanc, 2020-02-07:
Sure it would be better to switch to telegram, you can find me under the same username (sebanc in lowercase)

sebanc, 2020-02-07:
Just to say, I edited my previous post, the initial username I provided was incorrect. (I do not use telegram very often)

sebanc, 2020-02-07:
I just uploaded a script which has the only purpose to show if a ChromeOS image is compatible with your device:
https://gist.github.com/sebanc/b5e0f7b312030ee936d25a8d6417e061#file-chromeos_image_tester-sh

The script will output "ok" if the image is working or something like "chromeos_image_tester.sh: line 19:  7003 Illegal instruction     (core dumped)" if the image is not working.

When you have some time, I would need your feedback on the below images:
- coral
- nami
- nautilus
- rammus
- reef
- reks
- snappy

DiogoSilva48, 2020-02-27:
Working fine!!

closed 2020-02-27 by DiogoSilva48

andreidiaconu, 2020-05-19:
> Working fine!!

Hi! Do you remember what you did to get it working? Did you try different images until it worked? If yes, which one worked? Thanks.

andreidiaconu, 2020-05-19:
@DiogoSilva48 ^ Thanks again.

Gajalshankar, 2020-07-01:
Rammus recovery not working 
I tried older version 80 and 83

Specs- Intel i5 8250U with Nvidia MX150
[image]