sebanc/brunch #2: How is TPM handled?

erenoglu, 2020-02-15:
I read the readme but could not find any info on how the TPM Is handled. Is this trying to use the HW TPM2.0 or using the swtpm module?

Thanks

sebanc, 2020-02-15:
Hi,

TPM is handled via swtpm method. swtpm is integrated in the kernel and binaries are generated during the build process.

I will add something in the documentation when i review it.

erenoglu, 2020-02-15:
Thanks. Are you using an upstream kernel or 4.4 or 4.14 or 4.19 that some other Chromebooks use? 
I'm asking as I compile my own 4.19 kernel and use on an eve image with swtpm as well (need min 4.19 to have my thunderbolt docking station work). 
I tried we well 5.4 but could not get it to turn on graphics (on eve or nocturne).

Lastly, although my Linux and windows can successfully own and use my hardware swtpm 2.0, Chromeos could not, whatever I did, so I also reverted to swtpm method. (I actually boot and come to login with hw TPM but then it always want to restore my account which fails we password is not right)

sebanc, 2020-02-15:
Currently I build both 4.14 and 4.19 versions as i am trying to fix things for users facing issues. In both cases, the kernels are based on chromiumos source for release 79.
If you want, i can probably include the 4.19 patches you need in the next r79 release.

Regarding tpm, its use seems quite different in ChomeOS and I generally prefer not touchig the hardware tpm.

erenoglu, 2020-02-15:
Thanks. I'm just building 4.19, latest release from here from time to time:
https://chromium.googlesource.com/chromiumos/third_party/kernel/+/refs/heads/chromeos-4.19

I can also send my .config file if needed.

sebanc, 2020-02-15:
Yes I'd be happy to test your config. Could you please attach it in this issue if you do not mind ?

erenoglu, 2020-02-21:
Hi @sebanc , sorry for late response due to travels. Here it is. This is specific to my system and I disabled stuff I didnt need. 
[config419working.txt](https://github.com/sebanc/brunch/files/4238180/config419working.txt)

sebanc, 2020-02-22:
Thank you, your config might really help with some issues i am currently investigating. I am closing this issue for now but do not hesitate to open another one if you have trouble or need clarification on something.

closed 2020-02-22 by sebanc

erenoglu, 2020-02-22:
You're welcome. I can't use your project yet as my ChromeOS install resides with a corporate win10 installation. What I'd suggest, if you could do, is to bake-in latest qemu into the system and adjust user/group permissions to let chronos user launch a full virtual machine (ie windows 10). I'm doing it  by using qemu in chrouton at the moment.

sebanc, 2020-02-23:
Thanks for the suggestion, I think it is a very good idea and will look into it once I have fixed the native ChromeOS compatibility issues users are currently reporting.

sebanc, 2020-02-23:
Just to let you know, i wanted to see if it was possible building qemu for native ChromeOS and it seems I succeeded in integrating it to my current build with 4.14 kernel, it's probably unstable though and you have to use a vnc to connect to the vm. I will try to improve it over time. Moreover, for now, it has to be launched at root (need to fix permissions).

erenoglu, 2020-02-24:
Cool! Great job!  VNC is good, getting spice protocol to compile with it could be next step (a bit more complicated to do). I guess it wont be too hard to have it for 4.19 as well (has better USB-C and thunderbolt support)

sebanc, 2020-02-29:
Just to keep you up-to-date, I included qemu+spice support in the latest kernel 4.14 release.
Actually I will stop supporting 4.19 kernel for now as android apps are not working for many users with it but the chromeos 4.14 kernel version includes a lot of patches backported by google and seems to work fine for users with both USB-C and thunderbolt.

erenoglu, 2020-03-01:
OK thanks for informing me. I have no issues on my build with 4.19 on android apps btw. If 4.14 has the USB-C and Thunderbolt support then I can try.
Btw, do you also add Wireguard vpn support to the kernel?
https://www.wireguard.com/compilation/#building-directly-in-tree

sebanc, 2020-03-09:
Finally decided to release again 4.14 and 4.19 kernel versions. Both now have wireguard vpn support  builtin.

Did you manage to have qemu working without sudo ? I did not spend much time on this but tried to give chronos user the right groups and it did not work.

erenoglu, 2020-03-09:
Cool. In my setup, yes, I have qemu working w/o sudo. Chronos user need to have acess to /dev/kvm, disk image file location, and if you're using kernel mode networking with tap devices, you need to create a tap device owned by chronos (a bit more difficult), or add chronos to the group of those devices. (and do needed configurations).
Btw, I still did not install with your method, as I have windows 10 on this PC which I can't touch. If I create a disk image with your method, copy over its files to my ROOT-A partition, modify the writegpt.sh file to match my partitions, would it work?
Or would you maybe consider making an advanced installation script, which would let me choose my ROOT-A and STATE partitions (the two critical ones to boot). For boot, I modify my own grub myself.

sebanc, 2020-03-09:
Actually you might want to use the dual boot setup which only requires an ext4 or ntfs (without bitlocker) partition to store a chromeos disk image. It only creates a chromeos.img file on this partition which is booted using a specific grub config.

Last question, which command do you use to give chronos user access to /dev/kvm ? That's what is not working for me.

erenoglu, 2020-03-09:
that could be an option, I'll try that, but does that disk image also encompass the STATE partition and when the system boots, does it have access to rest of the system partitions (to reach the win10 VM on another partition for example).

For /dev/kvm, I might have just added chronos to the group that /dev/kvm is owned by. I'm on win10 now, can't check, but I have following in a backed up /etc/group file:

tun:!:413:crosvm,shill,chronos
virtaccess:!:418:crosvm,wilco_dtc,chronos

sebanc, 2020-03-09:
The stateful partition is included in the disk image as well, its size will depend on the size you give the image on your disk. Chromeos still has access to all your partitions, you can access them from /media/removable (if they are not bitlocker protected).

If your current setup is working fine, you might not necessarilly need to change it. Thank you for your suggestions which are very interesting.

Also, thanks for the pointers on the kvm group !

erenoglu, 2020-03-09:
Can the dual boot mode be contained in a partition instead of a disk image? I could then combine my ROOT-A and STATE partitions into one disk partition (CHROMEOS) with the size I want. I dont want to disk image method as I think it will add a performance penalty (even if small) due to file system layer.

sebanc, 2020-03-09:
It probably has a small performance penalty even if those who tested did not report any difference. Unfortunately, the way brunch works is not compatible with your setup (it uses all the chromeos partitions unlike croissant) so it cannot be installed as a partition.

erenoglu, 2020-03-09:
Btw, I had an issue for having qemu running with tap networking (fast), it works but I did not find a way to tap into a running dhcp and dns daemon (which works for android container and linux i guess). So I had to compile & run my own dnsmasq daemon serving as DHCP Server and DNS server for the Windows VM, running on the host TAP interface which I created at boot using an upstart script.

erenoglu, 2020-03-20:
Hi @sebanc , I tried today recent build, here's the error when I try to run qemu:

localhost / # qemu-system-x86_64 --help
qemu-system-x86_64: error while loading shared libraries: libspice-server.so.1: cannot open shared object file: No such file or directory_

Looks like some libraries are not included in the system. I searched for it but could not find.

sebanc, 2020-03-20:
Hi @erenoglu ,

Thank you for reporting this, indeed, I can reproduce this issue, it is an error in my build system...
I will try to fix this for the next release.