sebanc/brunch #3: clean install

av-attila, 2020-02-16:
Hi,
I just created a new image based on the reks image since I have a similar Lenovo Yoga. The comp boots from the usb drive to Chrome without any problem. The comp directly boots to Chrome now, never managed to get to any blue screen/grub. The comp has an unformated ssd. How do I install Chrome now to the ssd? Cloudready had a "install permanently to hdd" option. 
Cheers,
Atis

sebanc, 2020-02-16:
Hi,

Indeed, it is not well documented for now, I will try to improve it.

What you have to do is boot to chromeos, go to the chromeos shell, identify your hard drive in /dev and then run:
sudo bash chromeos-install -dst < your hdd. e.g. /dev/sdX >

Do not hesitate to tell me if it does not work as expected.

av-attila, 2020-02-17:
Thanks, that helped!

sebanc, 2020-02-18:
Just added the instructions for hdd install from ChromeOS so I close this issue. However, do not hesitate to reopen it if something is not clear.

closed 2020-02-18 by sebanc