sebanc/brunch #4: Camera

hacker1024, 2020-03-01:
This looks like a great project, but I've noticed a regression compared to Project Croissant.

I have a Dell Inspiron 7586 (with a 1280x720 front camera). This worked fine with Project Croissant with several base chromium builds, but it doesn't seem to work with brunch and the Octopus image. It's not detected.

sebanc, 2020-03-01:
Hi, 

Could you try with "eve" recovery image and see if it works ?

If it does not work, could you please post the content of "/etc/camera/camera_characteristics.conf" and "lib/udev/rules.d/50-camera.rules" ?

Thank you.

sebanc, 2020-03-03:
Hi,
I just uploaded a new "testing" release, could you try both camera support and tablet mode with it ?
Thank you

hacker1024, 2020-03-10:
> Hi,
> I just uploaded a new "testing" release, could you try both camera support and tablet mode with it ?
> Thank you

Sorry for the late response.
I can't see any testing releases, should I use the latest one?

hacker1024, 2020-03-11:
The latest release still doesn't work. Should I still try eve, and provide those files?

sebanc, 2020-03-14:
Could you try the new testing build ?

hacker1024, 2020-03-30:
The latest testing build works, thanks!

hacker1024, 2020-03-30:
What did you change?

sebanc, 2020-03-30:
For camera, i fixed a bug in my camera patch. Is touchpad working too ?

hacker1024, 2020-03-30:
Unfortunately not, I'm going to grab a dmesg tomorrow and post on that issue page.

sebanc, 2020-03-30:
Ok thanks for the feedback :) could you post the dmesg in your touchpad issue ? i will close this one as the camera is fixed.

closed 2020-03-30 by sebanc