sebanc/brunch #5: Tablet mode

hacker1024, 2020-03-01:
I have a Dell Inspiron 7586 2-in-1. When I fold the keyboard back, it disables, but tablet mode doesn't get activated. Is there a way to fix this?

sebanc, 2020-03-01:
Hi,

I will try to have a look at it but i am not sure i can make it work as some vendors have proprietary tablet mode implementations.
Could you please attach the output of "dmesg" and "lsmod" to this issue ?

Thank you.

hacker1024, 2020-03-01:
Sure. I remember seeing something about tablet mode in cros-config configuration files, I'll see if I can find it again as well.

hacker1024, 2020-03-01:
Linux definitely detects something, because the keyboard disables when folded back more than 180 degrees, which doesn't happen in macOS (which I dual boot).

sebanc, 2020-03-09:
Could you try tablet mode in the new releases  when you have the time ?

hacker1024, 2020-03-10:
Yep, will do

hacker1024, 2020-03-11:
The latest release works, thanks! Out of curiosity, how did you fix it?

sebanc, 2020-03-11:
I added tablet mode support to intel_hid driver.

[tablet_mode.patch.txt](https://github.com/sebanc/brunch/files/4316522/tablet_mode.patch.txt)

sebanc, 2020-03-14:
I am closing this issue as it appears to be fixed, do not hesitate to re-open it if needed.

closed 2020-03-14 by sebanc

juicypop, 2020-04-24:
Hello @sebanc 

first of all I'd like to thank you for the awesome work you did here. 
I installed according your readme and was up n' running in less than 30 minutes :)

I'm trying to turn an old Medio Akoya E3216 into a Chromebook for my grandma as she's used to Android. The Problem is I can't seem to figure out how to turn on or even verify that the tablet mode is working. This is a convertible notebook which should be ideal to use as a tablet too.

Could you please help me to activate it so it will work automatically?

Thanks in advance.

sebanc, 2020-04-24:
Hi,

Tablet mode if supported by the linux kernel should automatically happen when folding the device. However, there are a lot of missing tablet_mode drivers in the linux kernel which seems to be the case here (I could not find any online resource on tablet mode for medio devices.

There are actually 2 tablet modes (with/without keyboard). The simplest solution, if it works, would be to add "options=force_tablet_mode" to the kernel command line (after cros_debug) by running "sudo edit-grub-config" in crosh shell.

If the keyboard is correctly detected, the device will go from tablet_mode with keyboard (which is very similar to desktop mode, just an additional small icon on the bottom right of the taskbar) to full tablet mode when folded.

juicypop, 2020-04-24:
I just tried it but it appears that it still doesn't work for me.
Also my touchpad stopped working with "options=force_tablet_mode" enabled, no icon appeared in bottom right corner.

I saw in another related issue "screen auto rotate" that you asked for dmesg and lsmod, so i upload it right away. Hope you can tell me how to proceed from here on.

Oh an by the way, why does my ChromeOS rebuild rootfs each time I cold boot the laptop?

Thanks in advance.

[Archive.zip](https://github.com/sebanc/brunch/files/4529716/Archive.zip)

sebanc, 2020-04-24:
Thanks for the logs :)

For now I can mostly see that the tablet mode module is correctly loaded so I am not sure what goes wrong.

Do you have external (not keyboard) volume buttons on this device ? If so, are they working ?

juicypop, 2020-04-24:
No external buttons except a power button.
Other keys are simple FN keys which are working, some of them double mapped if f1-f12 aren't intended by nature of chrome os to have also their own system wide functions.

juicypop, 2020-04-26:
Hi @sebanc 
I don't want to rush you, but as asked on telegram I'd kindly ask you to provide me just a little more examples how to modify my driver.
I think it could help me if you could supply me with the original file you modified for the sourface to work properly and tell me where the modified version of the file is located now, so i can track the changes and understand a little bit more what you did with the file.

For now I was at least able to compile the kernel as it is, so I'm just missing now to start tinkering with my driver to hopefully get it working.

Thanks in advance.

frodriguez96, 2020-05-03:
Hi @sebanc 
I also have the same issue as @juicypop .
I have brunch installed on a Lenovo yoga pro 3-1370.
When I flip my display 180 degrees, my keyboard and mouse are disabled but does not switch into tablet mode in the OS. Also when I force enable tablet mode via grub edit, my keyboard and mouse will no longer work as well.

juicypop, 2020-05-05:
@frodriguez96 you can enable tablet mode also by pressing keyboard shortcut. I think it was Ctrl+shift+alt+t