sebanc/brunch #6: Trackpad not working

MaGaRCaN, 2020-03-01:
I'm using a Lenovo Ideapad 100-15IBY.
Trackpad moves very slow thecursor but not scrolling with to fingers or any other gesture.

Using CloudReady everything worked well.

sebanc, 2020-03-01:
Could you please add the output of both dmesg and lsusb ?

MaGaRCaN, 2020-03-01:
There you go.

[lsusb.txt](https://github.com/sebanc/brunch/files/4271676/lsusb.txt)
[demesg.txt](https://github.com/sebanc/brunch/files/4271677/demesg.txt)

sebanc, 2020-03-01:
Thank you, could you post also the output of "cat /sys/class/dmi/id/product_version" ?

MaGaRCaN, 2020-03-03:
Lenovo ideapad 100-15IBY

sebanc, 2020-03-03:
Sorry but i am having a hard time understanding what is the missing driver, could you post the outpout of "ls /sys/bus/acpi/devices/" ?

MaGaRCaN, 2020-03-03:
[devices.txt](https://github.com/sebanc/brunch/files/4283701/devices.txt)

sebanc, 2020-03-09:
Hi, I just pushed 2 new releases with kernel 4.19 and kernel 4.14. I tried to fix it on both but it should at least work in 4.19. Could you please try it when you have the time ?

MaGaRCaN, 2020-03-09:
Using Brunch r79 k4.19 stable 20200309 and latest octopus and same problem...

sebanc, 2020-03-09:
Could you try booting brunch, removing the file "/etc/gesture/40-touchpad-cmt.conf", then reboot and see if the trackpad is any better ?

MaGaRCaN, 2020-03-09:
Gestures are not working either.

sebanc, 2020-03-14:
Could you try the new testing build ?

MaGaRCaN, 2020-03-18:
Tried using Brunch r79 k4.19 testing 20200314 and Rammus 79.
Gestures does work (only scrolling) but are to sensible.
Trakpad mouse movement works, but it is not fluid and sometimes erratic.

Tried CloudReady and none of this happens.

PD: Bluetooth is not working either.

MaGaRCaN, 2020-03-18:
[dmesg.txt](https://github.com/sebanc/brunch/files/4350416/dmesg.txt)

sebanc, 2020-03-22:
The first r80 build I have just uploaded contains a bluetooth fix which might help with your issue. It will probably not fix the trackpad though.

sebanc, 2020-03-29:
Hi, there is a new trackpad fix in the r80 stable build I have just uploaded. Could you give it a try ?

sebanc, 2020-05-03:
Hi, in the absence of recent news on this issue I will close it but don't hesitate to re-open it if the new release still does not fix it.

closed 2020-05-03 by sebanc