sebanc/brunch #7: Touchscreen not working on Surface Pro 5 (2017)

dhernan3, 2020-03-02:
I'm trying Brunch, with both releases available today (stable and testing), with Octopus recovery image.
But touchscreen is not working.
What can I do to help you fix it? Or what is the thing I'm doing wrong?
Regards

sebanc, 2020-03-02:
Hi,
Indeed Surface patches have not been integrated in recent releases, the reason being i had to switch from kernel 4.19 (which had the surface patches but was not working for many) to kernel 4.14. The backportng of surface patches from 4.19 to 4.14 is ongoing and should be finished by the end of the week. I will udate this issue once the version with surface patches is available.

sebanc, 2020-03-09:
Hi, i just pushed a new kernel 4.19 release with ipts support but i just noticed battery indicator seems not to be working... I will fix it in the next release

sebanc, 2020-03-09:
Hi, I have just replaced k4.19 release with a new build which should be fully functional.

dhernan3, 2020-03-09:
I've noticed it.
I'm still building with the previous version, because I've changed to nautilus as it's the new base. I will let it to finish and then update (writting partition 3 between 500kib/s and 900kib/s)

dhernan3, 2020-03-10:
Still not working. I've using now Nautilus as base image and your latest 4.19 kernel.
The output dmesg shows:
   24.906507] ipu3-imgu 0000:00:05.0: loaded firmware version irci_irci_ecr-master_20161208_0213_20170112_1500, 17 binaries, 1212984 bytes
[   25.126295] intel_ipts mei::3e8d0870-271a-4208-8eb5-9acb9402ae04:0F: IPTS started
[   25.129670] DMAR: Allocating domain for mei::3e8d0870-271a-4208-8eb5-9acb9402ae04:0F failed
[   25.129677] intel_ipts mei::3e8d0870-271a-4208-8eb5-9acb9402ae04:0F: cannot allocate common resource
[   25.129681] intel_ipts mei::3e8d0870-271a-4208-8eb5-9acb9402ae04:0F: cannot allocate default resource
[   25.129684] intel_ipts mei::3e8d0870-271a-4208-8eb5-9acb9402ae04:0F: error in handling resp msg
[   25.184867] ish-hid {33AECD58-B679-4E54-9BD9-A04D34F0C226}: [hid-ish]: enum_devices_done OK, num_hid_devices=1
[   25.189959] hid-generic 001F:8087:0AC2.0002: hidraw1: <UNKNOWN> HID v2.00 Device [hid-ishtp 8087:0AC2] on 
[   27.314077] cgroup: run_oci (1516) created nested cgroup for controller "memory" which has incomplete hierarchy support. Nested cgroups may change behavior in the future.
[   27.314088] cgroup: "memory" requires setting use_hierarchy to 1 on the root
[   27.501726] mwifiex_pcie 0000:01:00.0: info: trying to associate to 'fisheroson' bssid 3c:98:72:b8:bf:01
[   27.641826] usbcore: deregistering interface driver btusb
[   27.650384] mwifiex_pcie 0000:01:00.0: info: associated to bssid 3c:98:72:b8:bf:01 successfully
[   27.656778] le_splitter HCI splitter inited
[   27.656855] usbcore: registered new interface driver btusb
[   27.656934] Bluetooth: hci_event.c:hci_cmd_status_evt() hci0: unexpected event for opcode 0x0000
[   27.659561] Bluetooth: hci_event.c:hci_cmd_status_evt() hci0: unexpected event for opcode 0x0000

sebanc, 2020-03-10:
Could you post your grub.cfg from partition 12 ?

dhernan3, 2020-03-10:
Here it is:

localhost ~ # mount /dev/loop15p12 /tmp/efi/
localhost ~ # cat /tmp/efi/
brunch.der  efi/        
localhost ~ # cat /tmp/efi/efi/boot/grub.cfg 
set timeout=2
menuentry "ChromeOS" {
  linux (hd0,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
  initrd (hd0,7)/initramfs.img
}
localhost ~ #

sebanc, 2020-03-10:
Sorry last thing, do you have a dual bot setup ?

dhernan3, 2020-03-10:
yes, dual boot (or triple boot really).
My configuration is:

Internal disk: Windows 10
USB: Dual boot for Ubuntu and ChromeOS

Default boot device: USB

sebanc, 2020-03-10:
ok could you post your the config from your dual boot grub.cfg ? I suspect you have intel_iommu=on in it, if so, could you try removing it ?

dhernan3, 2020-03-10:
Yes!! that was the problem! I didn't notice that change when creating new installation, so I hadn't changed my previous grub entry.

closed 2020-03-10 by dhernan3

sebanc, 2020-03-10:
Sorry, it was my mistake, I added that at some point but removed it after as it was creating issues...