sebanc/brunch #8: Grub entry for HDD image boot wrong if Linux not in English

dhernan3, 2020-03-03:
I'm testing the installation as a disk image to dual boot with Windows (internal storage), and Ubuntu and ChromeOs on my external usb stick.(Surface Pro 2017)
All went fine, but the script fails to show a correct grub entry for manual edit.
The install script, at the end, search for Filesystem (parameter for the sed command, to delete that line). But my linux is in Spanish, so the sed command doesn't found that word, because my system shows "S. ficheros" as the first line.
My system, which has as partition for the image /dev/sda3, has:
img_part=/dev/sda3
as grub parameter, after eliminating the incorrect line.
Then, it boots fine.

sebanc, 2020-03-09:
Hi, thanks for raising this bug, indeed it was not the correct way of doing this. This should be fixed in the new releases. Could you give try it again when you have some time ?

dhernan3, 2020-03-09:
I will try later today. I'm going to rebuild it with your latest kernel to check compatibility with my Surface Pro 2017.
Thanks.

dhernan3, 2020-03-09:
I've tested with the version released this morning, and bug resolved!

closed 2020-03-09 by dhernan3