sebanc/brunch #9: When I see the welcome GUI,it reboot.

ericma15, 2020-03-08:
First, thanks what you have done.
I use the script to build a image. And I boot it succesful.But everytime I saw chromeos welcome GUI,it sudden reboot.
I try use a usb disk and install on ssd. But nothing change.
I have tried these three images, eve,nocturne and octpus , nothing change.
And also I use chromefy build a image can succes boot and use with everything ok with image eve.

sebanc, 2020-03-08:
Thank you. Could you please provide me with your computer model ? Does it have dual i915/nvidia gpu ?

ericma15, 2020-03-08:
> Thank you. Could you please provide me with your computer model ? Does it have dual i915/nvidia gpu ?

no,it's lenovo yoga 730 13-IWL with intel i5-8265u intel 640 graphic card,and a nvme ssd

ericma15, 2020-03-08:
I've test with chromefy build ChromeOS, it fits well. All include pen and touchscreen worked well.

sebanc, 2020-03-08:
When you chromefy, which kernel are you using, ATB or fydeos ?

ericma15, 2020-03-08:
> When you chromefy, which kernel are you using, ATB or fydeos ?
both are okay, I built two. They all can work and perform well

ericma15, 2020-03-08:
I use https://chromium.arnoldthebat.co.uk/special//Camd64OS_R78-12499.B-Vanilla.7z and fydeos stable

sebanc, 2020-03-08:
Thanks for the additional information, I have an idea what the problem might be, I will try to correct it in my next build and inform you when it's out so that you can test.

ericma15, 2020-03-08:
> Thanks for the additional information, I have an idea what the problem might be, I will try to correct it in my next build and inform you when it's out so that you can test.

Ok, thank you very much

sebanc, 2020-03-09:
Hi, i just pushed 2 new releases (one with kernel 4.14 and one with kernel 4.19). Could you try them when you have the time ?

ericma15, 2020-03-09:
> Hi, i just pushed 2 new releases (one with kernel 4.14 and one with kernel 4.19). Could you try them when you have the time ?

OK,I'll try it now.

ericma15, 2020-03-09:
> Hi, i just pushed 2 new releases (one with kernel 4.14 and one with kernel 4.19). Could you try them when you have the time ?

No ,i try 4.19 with eve and octpus and 4.14 with octpus. It still have the problem.
I can see the brunch rebuild the fs.then screen shows chromeOS,when enter the setup page,before i try to click the button,it reboot.

sebanc, 2020-03-09:
Thanks for testing, then the problem is not what i thought it was. I will keep looking into it.

sebanc, 2020-03-09:
Could you try removing all files in the "patches" on partition 7 and reboot ? It would allow me to understand if the crash is related to the included patches.

ericma15, 2020-03-09:
ok，i'll try it.

ericma15, 2020-03-09:
The first time the screen sudden
 show some word like"factory mode
 " And a yellow "!". Then it reboot. Next time when boot to welcome page,it behave like I
Didn't remove patches.

ericma15, 2020-03-09:
Is there a way I can see the log?

sebanc, 2020-03-09:
Thanks for trying, it is the expected behaviour, it means it is not the patches.

Could you try to add "module_blacklist=hid-sensor-cros-compat" to the kernel command line in grub.cfg on partition 12 ?

ericma15, 2020-03-09:
Unfortunately, still reboot.

sebanc, 2020-03-09:
I don't see how we could get the log in that case...

ericma15, 2020-03-09:
I use camera slow motion caputure the screen,before it show chromeos
 Screen. I see "NMI watch dog :enabled". Maybe watchdog?

sebanc, 2020-03-09:
Could you try adding "nmi_watchdog=0" and if it does not work "fbcon=map:1" to the kernel command line ?

ericma15, 2020-03-09:
> Could you try adding "nmi_watchdog=0" and if it does not work "fbcon=map:1" to the kernel command line ?

Both are reboot. When use last, the screen don't print anything.

sebanc, 2020-03-09:
Chances are low but i just replaced the 4.19 release as the previous one had incorrect acpi patches, could you try it when you have some time ?

ericma15, 2020-03-09:
> Chances are low but i just replaced the 4.19 release as the previous one had incorrect acpi patches, could you try it when you have some time ?

Ok,I'll try it tomorrow. Really thank you.you are a nice developer.

sebanc, 2020-03-09:
Thanks, I hope we will manage to fix this. If I have other ideas, I will post them here but feel free to try them when you have the time.

ericma15, 2020-03-10:
> Thanks, I hope we will manage to fix this. If I have other ideas, I will post them here but feel free to try them when you have the time.

It still reboot. I upload my img build with chromefy.here is the link https://drive.google.com/open?id=1Jkc-fIw-WFK8Zy--7GYGUfKEL8KOH8o2. I found the param is cros_efi and i915.modeset=1. I added it but nothing changed.

sebanc, 2020-03-10:
Unfortunately, it seems the issue lies within the kernel so I cannot really use a croissant image to track it.
However, could you provide me with a "dmesg" and an "lsmod" from croissant ?
Thank you !

ericma15, 2020-03-10:
Hey, here is  msg. Beacuse my usb 3.0 disk has just broken now.So i use a low-speed usb2.0 disk.At this time, the android subsystem failed.
https://bhpan.buaa.edu.cn:443/link/8343264312388AA9DCA38064B435ABDC
https://bhpan.buaa.edu.cn:443/link/3B13096520D3DF6945DFB677C4C1E756

ericma15, 2020-03-10:
Hi, i do another test. I modify the chromeos_startup script in /sbin/. At the end of line. I add the
```
chmod -r 777 /mnt/stateful_partition/unencrypted/my
mount -o rw /dev/nvme0n1p4 /mnt/stateful_partition/unencrypted/my
/usr/bin/stdbuf -o0 /bin/dmesg > /mnt/stateful_partition/unencrypted/my/dmesg.txt 2>&1
/usr/bin/stdbuf -o0 /bin/echo 'a\n' > /mnt/stateful_partition/unencrypted/my/a.txt 2>&1
/usr/bin/stdbuf -o0 /bin/lsmod > /mnt/stateful_partition/unencrypted/my/lsmod.txt 2>&1
umount /mnt/stateful_partition/unencrypted/my
exit 0
```
I do this on brunch imag and croissant image. 
In croissant image, I can see these 3 files with content.
In brunch image,I see the 3 files have been establised. But all of them are 0B.
It's so strange.
BTW: the croissant i close it when show userlogin. And the brunch image reboot when it show 'Let's start it' button after chrome logo.

sebanc, 2020-03-11:
That's a very interesting test you did here, both should work the exact same so i am not sure what happens here but that's definitely helpful. I am thinking about what other tests we could do to identify the issue.

ericma15, 2020-03-11:
> That's a very interesting test you did here, both should work the exact same so i am not sure what happens here but that's definitely helpful. I am thinking about what other tests we could do to identify the issue.

I also doubt whether I use the wrong recovery image or the wrong cmd. I made it on the usb disk. On my another PC which with intel cpu and i915/nvidia, a optimus laptop. It behave the same,re boot.

sebanc, 2020-03-11:
Did you try "fbcon=map:1" on the i915/nvidia one ?

ericma15, 2020-03-11:
> Did you try "fbcon=map:1" on the i915/nvidia one ?

I add it. It now can boot. It seems that's my yoga problem.

sebanc, 2020-03-11:
Could you adding  "console= i915.modeset=1" to the kernel command line for the one that still does not boot ?

ericma15, 2020-03-11:
> Could you try adding "console= i915.modeset=1" to the kernel command line for the one that still does not boot ?

hey guys，i made a progress. I use a slow usb disk. So it run slowly, I change to tty2 and capture some message, some error.

sebanc, 2020-03-11:
The udev errors are harmless, the only abnormal thing in the log is related to i915 but it might be caused by the tty2 console. Could you try the kernel command line arguments in my previous comment ?

ericma15, 2020-03-11:
> The udev errors are harmless, the only abnormal thing in the log is related to i915 but it might be caused by the tty2 console. Could you try the kernel command line arguments in my previous comment ?

the change don't take effect. Still rebootm

sebanc, 2020-03-11:
Could you try renaming /etc/init/udev-trigger.conf to /etc/init/udev-trigger.test and reboot ?

ericma15, 2020-03-11:
> Could you try renaming /etc/init/udev-trigger.conf to /etc/init/udev-trigger.test and reboot ?

It doesn't reboot. Now my touchpad and touch screen not work, keyboard can.

sebanc, 2020-03-11:
Ok so it confirms it is a module causing a kernel panic when loading

sebanc, 2020-03-11:
Could you try probing "ideapad-laptop" first ? It might be the culpirit

ericma15, 2020-03-11:
> Could you try probing "ideapad-laptop" first ? It might be the culpirit

I don't really understand what to do. Here is what I do: I boot in the system with udev-trigger.test. switch to tty2 and run modprobe ideapad-laptop ,seems nothing happene

sebanc, 2020-03-11:
It's normal, it just means that it's probably not this module generating kernel panics. We would have to try the other ones that are used by your device

ericma15, 2020-03-11:
> It's normal, it just means that it's probably not this module generating kernel panics. We would have to try the other ones that are used by your device

ok，you can give me a list. I'll try them one by one.

sebanc, 2020-03-11:
It's the lsmod you have provided in an earlier post

ericma15, 2020-03-11:
> It's the lsmod you have provided in an earlier post

No one cause pannic,but some mod not found.

ericma15, 2020-03-11:
Hey, now I find udev excute rules 80-drivers.rules cause kernel panic. 
```
# do not edit this file, it will be overwritten on update

ACTION=="remove", GOTO="drivers_end"

ENV{MODALIAS}=="?*", RUN{builtin}+="kmod load $env{MODALIAS}"
SUBSYSTEM=="tifm", ENV{TIFM_CARD_TYPE}=="SD", RUN{builtin}+="kmod load tifm_sd"
SUBSYSTEM=="tifm", ENV{TIFM_CARD_TYPE}=="MS", RUN{builtin}+="kmod load tifm_ms"
SUBSYSTEM=="memstick", RUN{builtin}+="kmod load ms_block mspro_block"
SUBSYSTEM=="i2o", RUN{builtin}+="kmod load i2o_block"
SUBSYSTEM=="module", KERNEL=="parport_pc", RUN{builtin}+="kmod load ppdev"
KERNEL=="mtd*ro", ENV{MTD_FTL}=="smartmedia", RUN{builtin}+="kmod load sm_ftl"

LABEL="drivers_end"

```

So it seems some drivers cause this.

sebanc, 2020-03-11:
Wow you are good at debuging :) So it means that with stock rootfs, if you remove this file it is working ?

ericma15, 2020-03-11:
> Wow you are good at debuging :) So it means that with stock rootfs, if you remove this file it is working ?

No,the rule is to load kofile. Now I try to modify it to show which module it try to load.

sebanc, 2020-03-11:
You should probably try deleting the file first and see if it boots with stock rootfs.

ericma15, 2020-03-11:
> You should probably try deleting the file first and see if it boots with stock rootfs.

I try. Now it doesn't kernel panic.But my touchpad still can't work.

sebanc, 2020-03-11:
I am working on touchpad right now, i hope it will be fixed in the next release. If you can identify the module responsible for the kernel panic that would be really great

ericma15, 2020-03-11:
hey， i did it. I found all mod I should prob. Some can't find.But after I load all mods, it doesn't crash and I can use it.

ericma15, 2020-03-12:
I found the error module is hid-sensor-cros-compat.
But I ban it in linux kernel options seems not take effect.How strange.

ericma15, 2020-03-12:
i solve it. It seems add to kernel param can't prevent udev load it. So add it in blacklist.conf.

sebanc, 2020-03-12:
Thank you for all the time you took to investigate this issue. 
I think i understand what the issue is now and I will try to fix it in the next release.

ericma15, 2020-03-12:
> Thank you for all the time you took to investigate this issue.
> I think i understand what the issue is now and I will try to fix it in the next release.

Thanks what you do. Now I can use it by patch a blacklist.conf. Now there are other problem.
camera can't work.and d.process.media take two much cpu. That's what happended on my crossaint.

sebanc, 2020-03-14:
Thanks to your investigations, i have fixed the issue in the new testing build, camera should also be fixed. Could you try it ?

ericma15, 2020-03-14:
> Thanks to your investigations, i have fixed the issue in the new testing build, camera should also be fixed. Could you try it ?

Hey I try it. Now it can boot. Camera still can't use. Also ,  d.process.media seems take too much cpu.(This caused by chromeos mount my disk auto. It's a chromeos bug.) Also, I write a ugly patch to enable tablet mode on my laptop. But orientaion still can't work.
Here is my temporal fix.
[ideapad-laptop.txt](https://github.com/sebanc/brunch/files/4333110/ideapad-laptop.txt)

sebanc, 2020-03-14:
Which recovery are you using ?

[35 more replies]

closed 2020-03-27 by ericma15