sebanc/brunch #12: Touchpad tap to click

hacker1024, 2020-03-11:
On Project Croissant, on two devices that I tested, touchpad tap to click never worked properly - it would cause the mouse cursor to only start moving a few seconds after starting to swipe.

The first release of brunch amazingly fixed this for me, but the latest release has broken it again.

sebanc, 2020-03-11:
Are you using 4.14 or 4.19 kernel ?

sebanc, 2020-03-11:
Could you try adding "options=disable_touchpad_fix" to the kernel command line and see if it works better ?

sebanc, 2020-03-14:
Could you try the new testing build ?

sebanc, 2020-03-29:
A new trackpad fix has been included in the r80 stable release i have just uploaded.

If it still does not fix your issue, could you please share a dmesg ?

I am not sure you are still using brunch, therefore without feedback from you on this issue by April 5th 2020, i will close it.

hacker1024, 2020-03-30:
Sorry, I've been pretty busy lately.
I'll try as soon as I get the chance.

sebanc, 2020-03-30:
No worries, it was just to verify that you were still interested in this issue but take the time you need.

hacker1024, 2020-04-29:
The latest testing r81 works just fine, thanks.

closed 2020-04-29 by hacker1024