sebanc/brunch #13: Image creation can take a long time for large file sizes

BrianAllred, 2020-03-13:
Creating a 32GB image file was taking upwards of an hour on my laptop (i7-7Y75 and 256 GB NVMe). Recommend using `fallocate` instead of `dd` if possible. I replaced line 287 with `fallocate -l ${image_size}G "$destination"` and the image creation was instant.

May make a PR if I get time.

sebanc, 2020-03-14:
Thank you very much for pointing me to this. I did not knew `fallocate` which indeed which seems perfectly suited for this task.

I have to update my code as it is not up-to-date, you don't really need to make a PR, i will include it.

Thanks !

sebanc, 2020-03-14:
Your suggestion has been integrated in the new testing release, I still have to cleanup and update my code on github though.

sebanc, 2020-04-13:
The code has been updated with r80.

closed 2020-04-13 by sebanc