sebanc/brunch #14: How to update chromeos ?

santhoshr, 2020-03-13:
Thanks for these releases, works great on my Surface Book 2, I have few queries,

i) how to update chromeos, my version shows v79 to be latest from any channel ?
ii) which kernel to use 4.19 or 4.14 ? any advantage with one over another
iii) laptop overheats is there any mechanism like Linux TLP, can be brough over here to disable CPU throttling 
iv) for the other issue of buttons not working, is there any init script place I can have all there modprobe instructions that gets loaded on startup (like bash_profile) to enable hardware volume power keys ?

sebanc, 2020-03-14:
Hi,

1) If you want to use the native updater, you have to add "options=enable_updates" to the kernel command line in grub.cfg on partition 12. However, using with the native updater might break things (android, bluetooth, ...) as the brunch kernel is not updated at the same time as the rootfs. The proper way is to use the chromeos-update script when the brunch r80 package is out (once google will have released the recovery image).

2) Use kernel 4.19, kernel 4.14 is only maintained as many users have issues with android apps on 4.19 but i will stop building 4.14 once i fix the android issue.

3) Do you have a nvidia dGPU ? In general, chromeos it is not resource hungry and should not overheat the laptop unless under heavy crostini use.

4) The hardware buttons issue will hopefully be fixed by next release which should be out later today.

santhoshr, 2020-03-14:
Thanks for the reply.
1. ok
2. ok
3. No I have i5 7300U with intel HD620, the heat comparison is in with Windows, where it is hardly warm, charging could be one reason, another in linux where clipboard(tablet portion) is always hot similarly after installing TLP package, it reduced.
4. great

When I ran your disk-img build, it worked great, so I was thinking about increased speed when it ran as part of main partitions, so I moved windows partitions to the end, spent lot of time with gdisk/fdisk, recreated partition tables, and installed the disk-img to real partitions, but the performance is similar or even slightly slower, do you know why it happens ? is disk-img build is different from installing directly to disc using script ? or no difference can be perceived in nvm disk for both types of booting ?

sebanc, 2020-03-14:
For 3. I would like you to tell me how it works with the new build i am going to release. Also make sure you use nautilus recovery image, there is a known issue with ipu3 recovery images (atlas, nocturne...) where the android camera service runs in a loop trying to find ipu3 cameras which uses a lot of cpu.

Regarding the disk image, I don't really understand what you call main partitions and real partitions. Where did you install install it first ? If I understand well, in the end you created a dedicated partition for it, what was the format ntfs or ext4 ?
What i can tell already is that there is no difference between image creation with the script and moving it somewhere else.

sebanc, 2020-03-14:
Regarding point 3, could you please try adding "intel_pstate=disable" to the kernel command line in grub.cfg on partition 12 and see if it is better ?

santhoshr, 2020-03-14:
As of now the clipboard is not even warm for me to check!, have feeling it gets hot when it is charging and I'm using it, since I mostly do stuff in linux or chrome, I didn't notice it in windows, will test it and let you know if that is the case. Also when it overheats in chromeos, I will add the intel_pstate and let you know. New build I will update it any case, and let you know of issues. 

Also using the kernel option, I was able to update the chrome stable version from v79 to v80, without any issues.

Regarding partitioning, used below command to generate .img file
sudo ./chromeos-install.sh -src < path to the ChromeOS recovery image > -dst < path to the ChromeOS image on the chosen partition > -s < ChromeOS image size in GB >

The performance was great when using grub boot from disk image, later I flashed the disk image .img file exactly to my NVMe disk thinking this would speed up the booting, but to my surprise, this did not make any difference, I thought using disk image should incur some time penalty, in reality it is not.

santhoshr, 2020-03-14:
One more thing, my hardware keys Volume and Power button started working, without doing anything, not sure updating from the stable channel caused this.

sebanc, 2020-03-14:
> The performance was great when using grub boot from disk image, later I flashed the disk image .img file exactly to my NVMe disk thinking this would speed up the booting, but to my surprise, this [...]

Indeed, this has been confirmed by all users until now, the performance appears to be the same.

sebanc, 2020-03-14:
> One more thing, my hardware keys Volume and Power button started working, without doing anything, not sure updating from the stable channel caused this.

Could you try the new testing build, normaly it should be fixed ?

ericma15, 2020-03-14:
> Thanks for the reply.
>
> 1. ok
> 2. ok
> 3. No I have i5 7300U with intel HD620, the heat comparison is in with Windows, where it is hardly warm, charging could be one reason, another in linux where [...]

Hey, you can use top command to see what consume cpu. For me,it seems some chrome OS bug cause this. https://www.reddit.com/r/chromeos/comments/b1oud3/usrbinmountpassthrough_eating_cpu/ .
Chrome os automatic mount my other partiyions.

santhoshr, 2020-03-14:
> Hey, you can use top command to see what consume cpu. For me,it seems some chrome OS bug cause this. https://www.reddit.com/r/chromeos/comments/b1oud3/usrbinmountpassthrough_eating_cpu/ .
> Chrome os [...]

Could be I had fedora live usb mounted sometimes. Also read somewhere there was a bug even with top, could be very old chromeos version.

santhoshr, 2020-03-14:
@sebanc already updated my system with testing build, buttons seems working. I noticed just now, the hp stylus I was using, stopped working, previous installations it was working fine.

santhoshr, 2020-03-15:
@sebanc testing build enables buttons but rotation no longer works

sebanc, 2020-03-15:
With this build, hid sensors are disabled by default (the sensor module is causing kernel panics for some), to enable them add "options=enable_sensors" to the kernel command line in grub.cfg on partition 12.

santhoshr, 2020-03-15:
@sebanc rotation also works now, thanks

santhoshr, 2020-03-15:
This is something not sure it is chcromeos & device compatibility issue, my chromecast speakers are not detected at all, tried enabling different flags also, do you have any suggestion ? they were detected in android apps.

santhoshr, 2020-03-15:
Not sure it is known bug, changing chrome://flags and restarting session immediately resulting in freeze

sebanc, 2020-03-22:
I think the issues reported here have been corrected. I am closing the issue but don't hesitate to reopen it if needed.

closed 2020-03-22 by sebanc

sebanc, 2020-03-25:
Just found a small workaround regarding the cpu workload generated by media scanner, you can use "module_blacklist=fuse,ntfs-3g" to avoid the scanning of your disks by the android media scanner but they will also not be visible in chromeos.

sebanc, 2020-03-25:
Forget my last comment, it is just back after a suspend/resume cycle

ericma15, 2020-03-25:
I have a temporal solution. disabling "Media Storage" in anfroid. It may affect some app in android. But as my test, play store and many other app I use still work well. There is another way that may take effect. Use .nomedia to avoid media storage scan sd card.

sebanc, 2020-03-25:
@ericma15 Thanks for sharing those workarounds, it is very helpful :)

sebanc, 2020-03-29:
Hi, just to let you know the android media server high CPU consuption should be fixed in the r80 stable release I have just uploaded.

ericma15, 2020-03-29:
Hey, I try it. It actually works. How did you do it?

sebanc, 2020-03-29:
Thanks for the feedback :)

I used an udev rule to prevent the automatic mounting all internal drives:
ACTION=="add|change", SUBSYSTEMS=="block", ATTRS{removable}=="0", ENV{UDISKS_PRESENTATION_HIDE}="1"

ericma15, 2020-03-29:
Thx, now I understand. It's good. I try a usb disk. It can work.

ericma15, 2020-03-29:
Oh, but I have a disk box. which can insert a hdd. I connect it. It seems system think it doesn't removeable.

ericma15, 2020-03-29:
here is my fix,
'''
ACTION=="add|change", SUBSYSTEMS=="block", ENV{ID_BUS}!="usb", ENV{UDISKS_PRESENTATION_HIDE}="1"
'''
just for reference

sebanc, 2020-03-29:
I was writing an answer but indeed your solution seems perfect :) Thanks, I will integrate it in the next release

sebanc, 2020-03-29:
I will keep the ATTRS{removable}=="0" though, unless it creates an issue

ericma15, 2020-03-29:
Ok. I think my probelem is really rare. Maybe I formmart my disk in a wrong way.

ericma15, 2020-03-29:
In fact, in general situation, I think removable is better.

sebanc, 2020-03-29:
I mean with both ATTRS{removable}=="0" and ENV{ID_BUS}!="usb") it should work for your disk no ?

ericma15, 2020-03-29:
That disk removeable is 1 and ID_BUS is usb. I think it will not work.

sebanc, 2020-03-29:
If it is removable=1, the udev rule should never have had any impact, did you see a difference compared to before ?

ericma15, 2020-03-29:
I'll try it.

sebanc, 2020-03-29:
Yes please, i think the rule should still work for your disk with ATTRS{removable}=="0"

ericma15, 2020-03-29:
You'll right. works fine. Maybe it's the best solution.

sebanc, 2020-03-29:
Nice, it should fix all usb devices :)