sebanc/brunch #16: Default brightness on boot

santhoshr, 2020-03-16:
By design, Chrome OS defaults brightness to predefined higher value, could you make it remember previous settings on boot ? or provide some boot parameters to control the brightness value ?

cedricju, 2020-03-24:
Works fine on r80 release... Lenovo X1 Carbon/DELL XPS15

sebanc, 2020-03-30:
Hi,

On the latest brunch (20200329), you can use the below commands so that backlight is always set at a specific level on boot (replace 40 in this exemple by your expected backlight percentage):

```
sudo mkdir -p /var/brunch/bootscripts
echo "backlight_tool --set_brightness_percent=40.0" | sudo tee /var/brunch/bootscripts/backlight.sh
sudo chmod 0755 /var/brunch/bootscripts/backlight.sh
```

Please note however that the backlight slider will not reflect the change.

santhoshr, 2020-03-30:
seems working, but lower values are not reflected well, using slider I can make it very dark but using values like 20, 10, 5, still can't match slider when lowering. Still brilliant effort, thanks.

Crescendo-BLYAT, 2020-07-04:
try this:
1. launch shell
2. type: sudo nano /usr/share/power_manager/board_specific/has_keyboard_backlight
3. set the value to 0
4. save & reboot

Crescendo-BLYAT, 2020-07-06:
Temp solution from @sebanc via Discord:

I will think about has_ambient_light_sensor but in the meantime, I think it would be easier for you to create a bootscript to customize all those values to your needs.

1) create the directory /var/brunch/bootscripts
2) create a shell script with the below content in that folder (ending in .sh)

echo 0 > /usr/share/power_manager/board_specific/has_ambient_light_sensor
echo 0 > /usr/share/power_manager/board_specific/has_keyboard_backlight

3) Make the bootscript executable (chmod 0755 .....)

Bootscripts persist on both brunch and chromeos updates (but won't survive a powerwash)

closed 2026-04-29 by github-actions[bot]