sebanc/brunch #17: Screeb auto-rotation

cobernicus, 2020-03-19:
Hi. First of all, thank you very much for your work. On my Lenovo D330 everything works well except auto-rotate. I tried "options=enable_sensors" but this did not work for me. My screen is always in portrait orientation. What can I do ?

[image]
[image]

sebanc, 2020-03-19:
Hi,

The screen will only rotate in chromeos when in tablet mode, what you can do first is change the default desktop orientation in settings.

When you fold your laptop, does it trigger tablet mode ? (the desktop should change a bit)
It 's only once you are in tablet mode that the sensors can work.

If something is still not working, could you attach a dmesg and an lsmod to this issue ?

Thank you.

cobernicus, 2020-03-19:
Yes without keyboard Chrome os enable tablet mode but auto rotation is not working.

> could you attach a dmesg and an lsmod to this issue ?

Yes if you tell me how

sebanc, 2020-03-19:
Go to chromeos shell:

then type:
- "cd ~/Downloads"
- "dmesg > dmesg"
- "lsmod > lsmod"

You will then have dmesg and lsmod files in your Downloads folder, could you attach them to this issue ?

cobernicus, 2020-03-19:
[lenovo_d330.zip](https://github.com/sebanc/brunch/files/4356395/lenovo_d330.zip)

sebanc, 2020-03-20:
Hi,

I am sorry but your laptop is not equipped with the standard hid sensors but with a specific driver "bmc150_accel" which is unsupported.

closed 2020-03-22 by cobernicus