sebanc/brunch #18: How long should the first boot take?

duke7553, 2020-03-21:
When booting brunch, it went through all the steps for mounting the filesystem, but has stayed put at these last few lines for nearly 15 minutes.

duke7553, 2020-03-21:
[image]

duke7553, 2020-03-21:
This device is a Surface Laptop 3

duke7553, 2020-03-21:
Update: Trying to switch to a tty with CTRL ALT Fx gave me this output:
[image]

sebanc, 2020-03-22:
Hi,

There is clearly an issue here, could you try to press e when you see briefly the grub menu at startup, add "module_blacklist=intel-ipts" after cros_debug and then press F10 to boot ?

If it does not work, could you attach a short video of the full boot process ?

duke7553, 2020-03-22:
Will do right now. Thanks

duke7553, 2020-03-22:
@sebanc https://www.youtube.com/watch?v=HcK6jTm-bB4

sebanc, 2020-03-22:
Which recovery image are you using ?

duke7553, 2020-03-22:
@sebanc I'm using rammus with the newest brunch package.

sebanc, 2020-03-22:
Could you try adding "module_blacklist=intel-ipts i915.alpha_support=1" to the kernel command line
and upload an other video if it does not boot ?

duke7553, 2020-03-22:
@sebanc Sure, I'll get to it

duke7553, 2020-03-22:
@sebanc Adding those to the kernel loads the boot output for a few seconds then boots me into a black, backlit screen that flickers off every now and then.

sebanc, 2020-03-22:
I would say that your (very recent) intel gpu is not supported in kernel 4.19. I intend to build kernel 5.4 at some point which should be compatible with your device but for now i don't have an alternative solution.

I will update this issue when kernel 5.4 is available.

duke7553, 2020-03-28:
I tried copying a 5.4 kernel onto the USB with its modules, but it didn't quite work because I got a black screen after grub.

sebanc, 2020-04-13:
Hi, I have not moved to kernel 5.4 yet but I fixed an issue which might have prevented your laptop from booting. If you are still interested, you can give the new testing release a try.

duke7553, 2020-04-13:
You bet I'll give this a try! 😀

duke7553, 2020-04-21:
@sebanc That new release didn't seem to fix it

duke7553, 2020-04-21:
@sebanc I think v5.4 might be more compatible because I tried that copying that one to another Chromium OS spin and was able to boot to a live USB. It didn't boot after installing the OS though.

sebanc, 2020-05-10:
Hi,

Just updating this issue to say that moving to kernel 5.4 seems to be more complicated than expected as it introduces several issues and I don't have the time to maintain 2 different kernels currently.

Also, contrarily to what I first thought, some 10th gen intel are correctly booting brunch so I think this issue is just related to a specific patch that has to be backported to kernel 4.19.

Would you be interested in trying builds to hopefully solve the issue with kernel 4.19 ?

Another possibility would be for you to build kitakar's kernel and i can explain how to add it to brunch.

duke7553, 2020-05-10:
@sebanc Sure! I'd be happy to rapid-fire test builds for you.

duke7553, 2020-05-10:
@sebanc kitakar actually publishes binaries for the kernel, can I add those files to brunch?

sebanc, 2020-05-11:
Nice, let's try to fix this !

1st part so that you are not left out anymore, how to install kitakar's kernel:
- Create a brunch usb,
- Mount partition 7 of the brunch usb,
- Extract kitakar's kernel package somewhere,
- Copy /boot/vmlinuz from kitakar's kernel package to <brunch_partition_7>/kernel
- Extract modules.tar.gz into the directory <brunch_partition_7>/lib/
- Unmount partition 7 of the brunch usb.
- That should be it, try booting the usb

Limitations:
- Use 'eve' recovery if you want to be able to update chromeos
- Don't update brunch framework as it would replace the kernel (that should not be necessary for a while)
- Maybe some others i am not thinking of right now...

sebanc, 2020-05-11:
Regarding brunch kernel 4.19, I will build a new test release with a patch I have spotted.

LFmove, 2020-05-11:
Waiting for test the next release on my Surface Pro 7! 💪 
Thanks

sebanc, 2020-05-11:
@LFmove @duke7553 I have added you to a private repo on my git named "brunch-testing" that I use for issues that are difficult to debug and might break things for others.

When you have the time, could you try the release that's there ?

Very important: you have to add to the kernel command line "options=iwlwifi_backport", 2 possible ways to do it:
- Method1: Mount partition 12 of your usb in linux and add "options=iwlwifi_backport" after "cros_debug" in /efi/boot/grub.cfg.
- Method2: press "e" quickly at boot when you see grub and add "options=iwlwifi_backport" after "cros_debug" in /efi/boot/grub.cfg, then press F10 or CTRL-X to boot.

LFmove, 2020-05-11:
Tried,

Same thing (with the method 2), It stops.

Attached picture with the issue.
[image]

Available for other tests. 👍

sebanc, 2020-05-11:
Thanks ! Could you try booting with "module_blacklist=intel_ishtp_hid" ?

LFmove, 2020-05-12:
Sorry, i read it now.

Add the command with the others, or just this?

sebanc, 2020-05-12:
No worries,

The best would be to add "module_blacklist=intel_ishtp_hid options=iwlwifi_backport"

LFmove, 2020-05-12:
Ok, 10min and i try with :

Method2: press "e" quickly at boot when you see grub and add:

 options=iwlwifi_backport 

 cros_debug

/efi/boot/grub.cfg 

module_blacklist=intel_ishtp_hid options=iwlwifi_backport

then press F10 or CTRL-X to boot.

- Right?

sebanc, 2020-05-12:
Not sure:

Press "e" quickly at boot when you see grub:
1) locate the "cros_debug" text on the screen
2) add "module_blacklist=intel_ishtp_hid options=iwlwifi_backport" after the "cros_debug" text
3) press F10 or CTRL-X to boot.

LFmove, 2020-05-12:
Same error. Stops always at the same line (HDA Intel PCH HDMI/DP

But i see this lines (attached pic)
[image]

sebanc, 2020-05-12:
I doubt it could be the audio driver but we never know, could you try booting with "module_blacklist=snd_hda_intel options=iwlwifi_backport" ?

sebanc, 2020-05-13:
Forget my previous post, I just saw mostly the same log from someone who tried to boot chromeos with "nomodeset" which most likely means that this issue is related to the i915 driver not loading at all.

What's weird is that intel 10th gen laptops were able to boot brunch on kernel 4.19 but I guess these have a different gpu model.

I will try to see if I can find documentation on the SP7/SL3 gpu and to figure out the best way to move forward with this.

Meanwhile you can try kitakar's kernel.

LFmove, 2020-05-13:
Ok ok than you! Sorry but Yesterday i really can't try.

sebanc, 2020-05-13:
No worries, just try things when you have the time, I just wrote this because I saw the same log errors and the issue became much clearer.

I am currently backporting a few patches I found for i915 10th gen and will probably upload a new build that you can try whenever you have the time (no rush).

sebanc, 2020-05-13:
I have uploaded a new test release in the "brunch-testing" repository, could you try it with "options=iwlwifi_backport" added to the kernel command line when you have the time ?

LFmove, 2020-05-13:
Sure. ( Always after the "cros_debug" text, etc...?)

sebanc, 2020-05-13:
yes, just "options=iwlwifi_backport" after the "cros_debug" text

LFmove, 2020-05-13:
Ok, done It!

It's like finished, but when It reboot, you can ear a sound like ChromeOs starting, but the screen star flashing, "dark, grey, dark grey" (never stops)

Anyway...
I see some error in the Building operation. (Pic)
[image]

LFmove, 2020-05-13:
However, it is the first time that "builing" ends and the PC is restarted. (cool for me)

thorstenludwigs, 2020-05-14:
I did some testing with my surface Pro7, the problem I had that it went to a black flickering screen ("i915.alpha_support=1"). If I connect, an external Monitor, I get a blank white page on it. 

I tested kitakar's latest 5.4 kernel. It works, but I have Wifi and Bluetooth issues. 
Hope this helps a bit and thanks for your effort!

sebanc, 2020-05-14:
@LFmove The error while rebuilding rootfs is harmless, it's just an issue with the patch ordering when using "options=iwlwifi_backport", I will fix it.

@thorstenludwigs Thanks for the report, indeed it seems the problem with 4.19 kernel seems to be related to the internal display. I tried to identify the commit which fixes it in the i915 driver but there are too many.

For now, I will try to build an experimental 5.4 kernel that you can install manually.

thorstenludwigs, 2020-05-14:
Tanks for your help. I will compile a test kernel tomorrow as well.

kitakar5525, 2020-05-14:
Leaving note here that making SP7/SL3 wifi work with (my prebuilt) 5.4 kernel is being discussed also at here: https://github.com/kitakar5525/chromeos-kernel-linux-surface/issues/17

LFmove, 2020-05-21:
News for our Surface? 🙂

duke7553, 2020-06-03:
@sebanc Can you provide an update on v5.4 kernel?

sebanc, 2020-06-04:
Hi,

Both @kitakar5525 and I are working on 5.4 kernel with what seems to be relatively similar issues for now.

I will post my next bootable build here so that at least you have something to try.

sebanc, 2020-06-05:
Hi,

I have added everyone in this issue to a new private "brunch-testing" branch and uploaded a 5.4 build which appears to boot (at least on my Surface Go).

https://github.com/sebanc/brunch-next/releases

As previously mentioned, it's difficult to find the time to maintain 2 different kernels so I cannot guarantee regular/timely updates.

kitakar5525, 2020-06-06:
I also tried the 5.4 build (`brunch_r83_k5.4_testing_20200604.tar.gz`) and it's working on SB1 (NVMe disk).

So, I hope it also works on SP7/SL3. Regarding wifi, I heard it works with Linux 5.4's upstream iwlwifi (`CONFIG_IWLWIFI`) and that build uses this one, so hopefully wifi also works here.

@sebanc I'm curious if we can get SP7/SL3 wifi working with chromiumos' iwlwifi (`CONFIG_IWL7000`). I thought it's new enough but I heard it's unstable.
Do you know anything about the iwl7000?

By the way, it doesn't boot on Surface 3, saying that "brunch: ChromeOS was not found on any device, falling back to shell..."
This is maybe because mmc_block is built as a module (`CONFIG_MMC_BLOCK=m`). All of S3 models use eMMC.

sebanc, 2020-06-06:
Thanks for testing this release :)

Regarding CONFIG_IWL7000, it seems to be an alternative driver which means to have full-MAC support (not sure what that really means):
https://chromium.googlesource.com/chromiumos/third_party/kernel/+/2ce11a42c53960e9d40cd95dbf53ed0bdfe2bd4e
I enabled it at some point but it created issues for nearly all iwlwifi users, maybe it requires specific firmwares to work properly...

For the S3, nice catch ! I will correct it in a future build.

duke7553, 2020-06-07:
@sebanc Booting the brunch v5.4 image you published, it appears that WiFi does not work at all on my Surface Laptop 3.

sebanc, 2020-06-07:
@duke7553 that's weird, could you attach a dmesg (run "dmesg > ~/Downloads/dmesg.txt" and attach dmesg.txt to this issue.

Then, you can try booting with "options=iwlwifi_backport" added to the kernel command line and see if it helps.

duke7553, 2020-06-07:
@sebanc Sorry! It works now after installing it to the internal drive. Great work!

kitakar5525, 2020-06-07:
@sebanc 
> Regarding CONFIG_IWL7000,
> [...]
> I enabled it at some point but it created issues for nearly all iwlwifi users, maybe it requires specific firmwares to work properly...

I see... Thank you for the info.

Warelik, 2020-06-14:
Hi. Any update about this?

sebanc, 2020-06-14:
@Warelik For now you have to use the specific release here:
https://github.com/sebanc/brunch-next/releases

Warelik, 2020-06-15:
[lshw.pdf](https://github.com/sebanc/brunch/files/4781262/lshw.pdf)
Feels like everything works. The camera surprisingly works in crouton but does not work in chrome. There are some oddities in the work of android applications, but nothing critical so far.
Thank's @sebanc

sebanc, 2020-06-15:
@Warelik Thanks for the feedback :) There is currently some camera compatibility issues with rammus image, you can try to see if the camera works on "reef" which has the previous usb camera framework and should be more compatible.

weikevinhc, 2020-06-16:
Sorry to be a bother but can i get the invite to the brunch-next release again?

sebanc, 2020-06-16:
@weikevinhc validate the github invite email and you should have access now.

[208 more replies]

closed 2020-12-15 by sebanc