sebanc/brunch #19: No networks found

anubthomas, 2020-03-22:
Network Card - Qualcomm Atheros AR9565
Device - Lenovo Yoga 2 11
Chrome OS 79

sebanc, 2020-03-22:
Could you attach a dmesg ?

sebanc, 2020-03-22:
I changed something in the r80 build i have just posted. Could you confirm if your issue is still present and if it is provide the dmesg from this release ?

anubthomas, 2020-03-23:
[log.txt](https://github.com/sebanc/brunch/files/4370391/log.txt)
tried with r80 build. dmesg for the same attached.

sebanc, 2020-03-25:
The wifi module seems properly loaded, could you try to run "sudo rfkill unblock all" in cros shell ?

sebanc, 2020-03-25:
Could you also try 'echo "blacklist ideapad-laptop" | sudo tee /etc/modprobe.d/idea.conf' ?

anubthomas, 2020-03-25:
I tried 'sudo rfkill unblock all' and 'echo "blacklist ideapad-laptop" | sudo tee /etc/modprobe.d/idea.conf' and that did the trick. Now it works like expected. Thank you!
Closing the bug.

closed 2020-03-25 by anubthomas

sebanc, 2020-03-25:
Thanks for closing the issue.

This comment is just to say that the best way to make it permanent is probably to add "module_blacklist=ideapad-laptop" to the kernel command line in grub.cfg on partition 12 (after cros_debug)