sebanc/brunch #20: Startup in laptop mode

cobernicus, 2020-03-22:
My tablet always startup at laptop mode. For enable tablet mode I need connect and disconnect some keyboard. It make my tablet without keyboard almost useless. Can I force tablet mode at startup ?

Mayk728, 2020-03-23:
I'm using Ultima R80 (same with R78) on my surface 3 and it starts up in tablet mode with the keyboard already disconnected.

sebanc, 2020-03-26:
Hi, I have added the "force_tablet_mode" option in the new release which should let your device in tablet mode all the time. But to be honest, i completely forgot to test it.

If you want to give it a try, add "options=force_tablet_mode" to the kernel command line (after cros_debug) in grub.cfg on partition 12.

sebanc, 2020-03-27:
Tested this option and it works fine for me, let me know how it goes

sebanc, 2020-04-30:
I am cleaning the issues list, as this one is supposedly fixed for 1 month I will close it but you can re-open it if it still does not work.

closed 2020-04-30 by sebanc