sebanc/brunch #21: Crash on sleep ?

santhoshr, 2020-03-23:
Left my clipboard (tablet) open on SB2, when came back, pressing power button restarted the system, looks like system does not recover from sleep

cedricju, 2020-03-23:
I confirmed the same with the last R79 release. Lenovo X1 carbon touchscreen

cedricju, 2020-03-23:
It does not crash, it just hangs in Sleep mode...
BTW upgrade from R79 to R80 (Full brunch+Chromeos) is working fine. I feel a slighter higher speed on my config :)

santhoshr, 2020-03-23:
I too upgraded to R80 sleep bug still there, you can go to settings and disable sleep

cedricju, 2020-03-23:
That is what I did. Get about 8 hours battery now:). Pretty good indeed

sebanc, 2020-03-23:
Indeed, there is an issue in the last version, I will try to post a fixed one as soon as possible.

sebanc, 2020-03-26:
Hi, could you test suspend on the release i have just uploaded ?

cedricju, 2020-03-27:
It does not work on my X1 carbon and my Dell xps (The one with the internal
disk that is not seen) died yesterday. Waiting for a technician to change
the mainboard today.
But While my X1 carbon goes to sleep but does not wake up (The power led
blinks but cant be waked up), My Dell was waking up from Sleep but crashed
(seemed to hang!!).
It is know on Windows that XPS15 has a huge issue with Sleep mode, so i
would not focus soo much on this for now.
tx again for the hard work

cedricju, 2020-03-28:
I freshly installed the latest build and My X1 goes in sleep mode, but i cant wake it up. has to be (forced) turned off and turn on again

sebanc, 2020-03-29:
Could you try suspend again with the new r80 stable build ?

Thanks

santhoshr, 2020-03-30:
works very well with latest r80 stable build, I enabled sleep in settings, also enabled lock screen. Triggered using command from shell "sudo powerd_dbus_suspend", no crash, recovers very well both with and without lock screen.

Thanks.

cedricju, 2020-04-02:
I reinitialize my entire system with Rammus and latest stable build but i still cant wake up form sleep mode. Both X1 Carbon and XPS15. My X1 goes to sleep mode, my XPS does not. Can hear the CPU fan getting crazy for a couple of seconds then back to normal

sebanc, 2020-04-02:
@cedricju For the XPS15, could you try adding "mem_sleep_default=deep" to the kernel command line ?

cedricju, 2020-04-12:
I have to report that my 9575 xps was kind of dead when it came to sleep:):) I could not restart it at all. Had to wait for battery to die by forcing an "on/off" I hours:):). After that it restart normally... I don't think I want to try again. The xps15 is known to be a POS when it comes to sleep mode soninwiuld not focus on this too much. I will try on my x1 carbon... Next laptop will never be a Dell again;)

cedricju, 2020-04-15:
I have the same issue on my Lenovo X1. it does go to sleep. but i cant wake it up. Need to force an Shutdown by holding the power button in like 5 sec then restart normally. the battery life time is good so thus is not a big issue for me right now :)

sebanc, 2020-04-15:
@cedricju Could you test on both your laptops to add "options=suspend_s3" to the kernel command line ?

sebanc, 2020-05-03:
Hi, I would need newer and individual reports to properly work on them so I am closing this issue but do not hesitate to create another issue if the problem persists.

closed 2020-05-03 by sebanc

numfin, 2020-06-12:
@sebanc hi, i installed chrome os today (awesome work btw), and have issue with sleeping mode. What i need to do to solve it or help you to help me to solve it? :)