sebanc/brunch #22: Webcam not initialized

cedricju, 2020-03-23:
Hello,
Got a Dell XPS15. the Webcam does not want to initialized. Here is the dmesg log. Looks like it find it as a device but does not initialized it. The rest is working fine.
I tried on r79 and r80...
Sleep seemed to be working on this computer but when coming back from sleep, i cant reconnect to WIFI and after a minute or so, my laptop just hangs. It is not important to meat this level:)
Tx for the hard work!
[dmesg_webcam.log](https://github.com/sebanc/brunch/files/4371724/dmesg_webcam.log)

sebanc, 2020-04-07:
Hi @cedricju Could you try the new build I have just released for both camera and suspend ?

cedricju, 2020-04-08:
Will do

jjrojoc, 2020-04-11:
I have a HP-G6 and camera don't start
No tested slepping
I'll post some logs

cedricju, 2020-04-12:
Camera does not initialize either on latest build. Xps15 9575. Works fine on Lenovo X1

sebanc, 2020-05-03:
Hi, just to say I am still working on this issue for "rammus" recovery image but that most users reports that their cameras are working correctly with the "samus" recovery image.

sebanc, 2020-06-26:
The camera issue has been included in #331 "Main issues on which help is needed". I am therefore closing individual issues related to camera not working with rammus recovery.

closed 2020-06-26 by sebanc