sebanc/brunch #23: cant see internal HD

cedricju, 2020-03-24:
I cant seem to see my internal HDD. I can then not install from USB to HDD. been googling for hours trying to find a way to mount the internal disk where i have a free partition to install but no luck so far...

sebanc, 2020-03-25:
Hi, what would you like to do, full install on hdd or dual boot process ?

cedricju, 2020-03-25:
Actually both:)
I have done app that still require windows. I'll need a small windows
install and the rest for Chromeod.
Tx for the hardwork.
C

sebanc, 2020-03-25:
Ok so the best for you will probably be dual boot.

Could you run the following commands from chromeos shell:
```
cd ~/Downloads
ls -l /dev > devices.txt
lsblk > lsblk.txt
```

And then send me the devices.txt and lsblk.txt from your Downloads directory ?

cedricju, 2020-03-25:
[devices.txt](https://github.com/sebanc/brunch/files/4380150/devices.txt)
[lsblk.txt](https://github.com/sebanc/brunch/files/4380152/lsblk.txt)

here you go :)

sebanc, 2020-03-25:
It seems your hdd pass through a scsi interface and that needed modules are disabled in the kernel. I will try to extend scsi support in the next release.

cedricju, 2020-03-25:
cool :) merci!

sebanc, 2020-03-25:
Si possible, est-ce que tu pourrais m'envoyer un "dmesg" et un "lsmod" depuis une distribution linux ou chromium qui reconnait correctement ton hdd ? Ca m'aiderait beaucoup.

cedricju, 2020-03-26:
pas de probleme. je fais cet apres midi:)

cedricju, 2020-03-28:
i figured this out... My BIOS was configured to RAID (Default Dell BIOS value). i switched to AHCI and bingo, i can see my internal drive. you may close that case:)

sebanc, 2020-03-29:
Nice find ! and thanks for sharing the solution :)

closed 2020-03-29 by sebanc