sebanc/brunch #25: Playstore doesn't load

MaddTitan, 2020-03-27:
after initial login, i get an error "Unable to load playstore" any ideas on what could be causing it

sebanc, 2020-03-27:
What's your computer model ?

MaddTitan, 2020-03-27:
surface 3 non pro model

sebanc, 2020-03-27:
It's weird, i have had many 3 different reports of Surface 3 owners who have android apps working, could you try on a faster usb or with a dual boot install ?

Also, deleting google sync can help sometime:
https://support.google.com/chromebook/answer/2914794?hl=en

sebanc, 2020-03-27:
Also forgot to ask but which recovery image are you using ?

MaddTitan, 2020-03-28:
I'm using RAMMUS v79. 

I switched to Ultima and the play store works there, but then i get no battery indicator, no auto rotate spotty Bluetooth. With regards to the USB, I'm use a 3.0usb.

If you can recommend a more suitable recovery image for the Surface 3 that would be pretty well appreciated, because I'm really liking your work

sebanc, 2020-03-28:
Could you try the latest r80 testing (it has been reported to be stable) with rammus r80 ? And if it does not work, could you send me the output of dmesg ?

treypop, 2020-03-28:
The playstore loads for me but the interface freezes a lot. I'm on an Inspiron 15-3521 with a Pentium 2127U running Rammus 80 and the latest r80 testing.

sebanc, 2020-03-28:
@treypop Thanks for your feedback.

I would like to keep this issue focused on android not working on Surface 3 but if you open an other issue, i can try helping you with the freezing (not sure it will work though as all the chromebooks supporting android have a more powerful cpu)

MaddTitan, 2020-03-28:
I'll test it out and let you know

MaddTitan, 2020-03-29:
RAMMUS v80 and Brunch r80 installed and running stable so far. 

Play Store works
Apps work
Wifi Works
Bluetooth Works
Performance is a little stuttery, but i guess it's to be expected

No Battery Indicator
Auto-Rotate not working, my guess is sensors not enabled
Auto brightness not working, assuming same as above, sensors not enabled

sebanc, 2020-03-29:
Could you tell me which brunch package you downloaded ?

sebanc, 2020-03-29:
Could you try running the below command in cros shell, reboot and see if it fixes the battery indicator ?

```
echo 25000 | sudo tee /usr/share/power_manager/board_specific/battery_stabilized_after_startup_ms
```

kitakar5525, 2020-03-29:
Hi, I heard that the battery indicator on Surface 3 does not work with old UEFI firmware before (https://github.com/linux-surface/linux-surface/issues/60#issuecomment-577946157).

What's your UEFI version?
```bash
sudo dmidecode -t 0 | grep "Version:"
```

sebanc, 2020-03-29:
> Hi, I heard that the battery indicator on Surface 3 does not work with old UEFI firmware before ([linux-surface/linux-surface#60 (comment)](https://github.com/linux-surface/linux-surface/issues/60#issuecomment-577946157)).
> 
> What's your UEFI version?
> 
> ```shell
> sudo dmidecode -t 0 | grep "Version:"
> ```

Thanks for the help :)

MaddTitan, 2020-03-29:
> Could you tell me which brunch package you downloaded ?
> 
> Until 20200326 version, you had to add "options=enable_sensors" to the kernel command line for sensors to work.
> 
> I will try to fix the battery indicator.

Brunch package 20200326

MaddTitan, 2020-03-29:
> Hi, I heard that the battery indicator on Surface 3 does not work with old UEFI firmware before (https://github.com/linux-surface/linux-surface/issues/60#issuecomment-577946157).
> 
> What's your UEFI version?
> ```bash
> sudo dmidecode -t 0 | grep "Version:"
> ```

Version: 1.51116.218

How would i update if needed?

sebanc, 2020-03-29:
> > Could you tell me which brunch package you downloaded ?
> > Until 20200326 version, you had to add "options=enable_sensors" to the kernel command line for sensors to work.
> > I will try to fix the battery indicator.
> 
> Brunch package 20200326

Ok, then for sensors you can either add "options=enable_sensors" to the kernel command line (after cros_debug) by using "sudo edit-grub-config" or update to the 20200329 release.

MaddTitan, 2020-03-29:
> Ok, then for sensors you can either add "options=enable_sensors" to the kernel command line (after cros_debug) by using "sudo edit-grub-config" or update to the 20200329 release.

I'll try the command first and see if that work, if not I'll do the update

sebanc, 2020-03-29:
> > Hi, I heard that the battery indicator on Surface 3 does not work with old UEFI firmware before ([linux-surface/linux-surface#60 (comment)](https://github.com/linux-surface/linux-surface/issues/60#issuecomment-577946157)).
> > What's your UEFI version?
> > ```shell
> > sudo dmidecode -t 0 | grep "Version:"
> > ```
> 
> Version: 1.51116.218
> 
> How would i update if needed?

The best way is probably to install windows, perform the update and reinstall chromeos.

MaddTitan, 2020-03-29:
I figured, so I'm doing that know. Will give an update when it's all done

MaddTitan, 2020-03-30:
Surface 3 Firmware Updated
ChromeOS v80 installed
Brunch v80 installed

WiFi Works
Bluetooth Works
Play Store Works
Performance Improved
Auto-Rotate Works
Auto-Brightness seems inconsistent
Power Button Works for device sleep/wake
Power Drain is reasonable

Camera not working

At this stage, this build on the Surface 3 can be considered pretty good as a daily driver

sebanc, 2020-03-30:
Thanks for the feedback :)

Unfortunately, as far as i can see, there are no linux kernel drivers for S3 cameras, so I will not be able to implement this feature.

MaddTitan, 2020-03-30:
I can live without the camera on a tablet as they were only a recent addition to this style of device, and in practice are cumbersome to use.

Is there a chance the mic can work?

sebanc, 2020-03-30:
How did you test the mic ? If it is with "Ok google" you need to activate an option first in google assistant settings, it is disabled by default.

MaddTitan, 2020-03-30:
One last observation, Bluetooth audio works, was able to pair my earbuds and listen to music, but headphone jack seems inactive

kitakar5525, 2020-03-30:
(Battery indicator started working now? The version `1.51116.218` seems new enough. So, another problem might have existed)

MaddTitan, 2020-03-30:
> (Battery indicator started working now? The version `1.51116.218` seems new enough. So, another problem might have existed)

There is a newer update that is required, the latest is 1.51116.238.0, but you would have to reinstall windows, then download updates to get it.

MaddTitan, 2020-03-30:
> How did you test the mic ? If it is with "Ok google" you need to activate an option first in google assistant settings, it is disabled by default.

i have all the settings enabled, unless i am missing something simple?

sebanc, 2020-03-30:
Could you confirm the Ok google setting is "always on", I guess it is but it is just to confirm as this setting is off by default ?

[image]

MaddTitan, 2020-03-31:
> Could you confirm the Ok google setting is "always on", I guess it is but it is just to confirm as this setting is off by default ? (the red arrows are not from me, it is the only image i found on google)
> 
> [image]
> 

Confirmed as "always on" but nothing happens when I say the words. Tested with voice recording app, still nothing.

sebanc, 2020-03-31:
Ok, thanks for confirming, i will try to see if there are specific alsa ucm files necessary for SP3.

sebanc, 2020-04-01:
@kitakar5525 Do you know if microphone is working for S3 under linux ?

kitakar5525, 2020-04-01:
At a glance, it seems that recording is not working on Arch Linux, too... It may take some time to figure out what's going wrong, but I'll try.

sebanc, 2020-04-01:
@kitakar5525 Thanks a lot for the confirmation, i will have a look at it, it was just to know if I should look for a fix already implemented in linux distros or on the forums.

MaddTitan, 2020-04-01:
Not sure if i mentioned this before but audio via the headphone jack seems to not work either

sebanc, 2020-04-01:
Could you try replacing the files in /usr/share/alsa/ucm/chtrt5645 with those ones ?

[chtrt5645.zip](https://github.com/sebanc/brunch/files/4417001/chtrt5645.zip)

kitakar5525, 2020-04-02:
By the way, to check whether the microphone is working, this live demo is useful: https://github.com/lairdkruger/Twistline

kitakar5525, 2020-04-02:
@sebanc 
> Could you try replacing the files in /usr/share/alsa/ucm/chtrt5645 with those ones ?
> 
> [chtrt5645.zip](https://github.com/sebanc/brunch/files/4417001/chtrt5645.zip)

I also tried the file (the file seems to be from the last v1 version UCM of alsa-project/alsa-ucm-conf: https://github.com/alsa-project/alsa-ucm-conf/tree/6c56567a7cb31b183ea6b3df5a4cbdc1faeb5aac) on chromeos.
tl;dr: the mic still isn't working.

---
leaving here what I did.

#### v1 UCM from alsa-project/alsa-ucm-conf
First, even mic wasn't even recognized by the live demo. If I run the following command, the waveform appeared but it seems mic isn't working.
```bash
alsaucm -c chtrt5645 set _verb HiFi set _enadev Mic
```

#### UCM from plbossart/UCM (https://github.com/plbossart/UCM) (brunch uses this?)
First, the mic wasn't even recognized by the live demo, too.
After running the following commands, the waveform appeared and somewhat seems to be responding when I tap the place where the mic is located. But the waveform is too weak compared to Surface Book 1.
```bash
alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic
```
```bash
alsaucm -c chtrt5645 set _verb HiFi set _enadev LineOut
```
(`Mic` didn't work)

kitakar5525, 2020-04-02:
@MaddTitan You may need to run this command on startup for audio via the headphone jack to work:
```bash
amixer -c chtrt5645 set "Headphone Channel" on
```

MaddTitan, 2020-04-02:
> @MaddTitan You may need to run this command on startup for audio via the headphone jack to work:
> ```bash
> amixer -c chtrt5645 set "Headphone Channel" on
> ```

Thanks, it worked.

Would I have to run this at each start up?

kitakar5525, 2020-04-02:
> Would I have to run this at each start up?

Unfortunately, yes. I'm not sure why needed.

kitakar5525, 2020-04-02:
Even on Windows, the mic sound level is too low and noisy on my S3🤔 (affected by OEMB issue? not sure)
So, I can't really confirm that mic isn't working on Linux and chromeos.

@MaddTitan 
On chromeos, try each command:
- `alsaucm -c chtrt5645 set _verb HiFi set _enadev Mic`
and
- `alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic`

and let me know your thoughts about mic is working or not.

Also, if Windows is still available, let me know your thoughts about mic recording quality.

MaddTitan, 2020-04-02:
> > Would I have to run this at each start up?
> 
> Unfortunately, yes. I'm not sure why needed.

No problem, small price to pay for creature comforts.

sebanc, 2020-04-02:
@kitakar5525 Thanks for all the help !

I recently added  feature that may be useful in this case, in the latest brunch version you can create user init scripts by saving them to "/var/brunch/bootscripts".
The user init scripts will survive chromeos and brunch updates, they only have to finish with ".sh" and will be executed after modules loading finished.

So in this case, the below commands should do the trick:
```
sudo mkdir /var/brunch/bootscripts
echo 'amixer -c chtrt5645 set "Headphone Channel" on' | sudo tee /var/brunch/bootscripts/headphones.sh
sudo chmod 0755 /var/brunch/bootscripts/headphones.sh
```

MaddTitan, 2020-04-02:
> Even on Windows, the mic sound level is too low and noisy on my S3🤔 (affected by OEMB issue? not sure)
> So, I can't really confirm that mic isn't working on Linux and chromeos.
>
> @MaddTitan
> On [...]

Ok... So mic works, tested with Google assistant and it responded to the command

For some reason voice recorder apps don't seem to work, but if i trigger the assistant with the command, it responds while the recording is happening

MaddTitan, 2020-04-02:
In addition to the previous observe, enabling the mic kills the headphone jack. So I had to enable the headphones Jack again, bit the mic remained enabled.

MaddTitan, 2020-04-02:
> @kitakar5525 Thanks for all the help !
> 
> I recently added a feature that may be useful in this case, in the latest brunch version you can create user init scripts by saving them to "/var/brunch/bootscripts".
> The user init scripts will survive chromeos and brunch updates, they only have to finish with ".sh" and will be executed after modules loading finished.
> 
> So in this case, the below commands should do the trick:
> ```
> sudo mkdir -p /var/brunch/bootscripts
> echo 'amixer -c chtrt5645 set "Headphone Channel" on' | sudo tee /var/brunch/bootscripts/headphones.sh
> sudo chmod 0755 /var/brunch/bootscripts/headphones.sh
> ```

Can I test this out? 
Would I need to run commands from the Crosh Shell (Ctrl + alt + t) or Developer shell (Ctrl + alt +f2)?

sebanc, 2020-04-02:
the commands I provided are for cros shell but you can also do it from developer shell without the "sudo".

If you want to add the mic commands first, it would give the below commands:
```
sudo mkdir -p /var/brunch/bootscripts
echo 'alsaucm -c chtrt5645 set _verb HiFi set _enadev Mic' | sudo tee /var/brunch/bootscripts/asound.sh
echo 'alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic' | sudo tee -a /var/brunch/bootscripts/asound.sh
echo 'amixer -c chtrt5645 set "Headphone Channel" on' | sudo tee -a /var/brunch/bootscripts/asound.sh
sudo chmod 0755 /var/brunch/bootscripts/asound.sh
```

MaddTitan, 2020-04-02:
> the commands I provided are for cros shell but you can also do it from developer shell without the "sudo".
>
> If you want to add the mic commands first, it would give the below commands:
>
> ```
> sudo [...]

Will add the commands and restart the device and report the findings

MaddTitan, 2020-04-02:
Success!! we have a functional mic and headphone jack. Will test it out some more, but so far they seem to stable enough. Will do some more testing of the mic for voice recording purposes, but the Google Assistant is able recognize it.

kitakar5525, 2020-04-03:
`Mic` and `DMic` are each exclusive (https://github.com/plbossart/UCM/blob/145869c590d8fd1db9dc7ebeb1d4143879f3ff2b/chtrt5645/HiFi.conf#L299-L311 and https://github.com/plbossart/UCM/blob/145869c590d8fd1db9dc7ebeb1d4143879f3ff2b/chtrt5645/HiFi.conf#L349-L361)

So, you need to run ether one of working command (`alsaucm -c chtrt5645 set _verb HiFi set _enadev Mic` or `alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic`)

And for the internal speaker to work with working mic at the same time, add `set _enadev Speaker` to the command (for example: `alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic set _enadev Speaker`)

@MaddTitan By the way, are you sure the mic is really working?
Please try recording with mic with the following commands:
```
cd ~/Download
arecord arecord.wav # record to file named arecord.wav

# play the recording
aplay arecord.wav
```
For my S3, I can hear only noise 😢

MaddTitan, 2020-04-03:
> `Mic` and `DMic` are each exclusive (https://github.com/plbossart/UCM/blob/145869c590d8fd1db9dc7ebeb1d4143879f3ff2b/chtrt5645/HiFi.conf#L299-L311 and https://github.com/plbossart/UCM/blob/145869c590d8fd1db9dc7ebeb1d4143879f3ff2b/chtrt5645/HiFi.conf#L349-L361)
> 
> So, you only need to run ether one of the working command (`alsaucm -c chtrt5645 set _verb HiFi set _enadev Mic` or `alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic`)
> 
> And for the internal speaker to work with working mic at the same time, add `set _enadev Speaker` to the command (for example: `alsaucm -c chtrt5645 set _verb HiFi set _enadev DMic set _enadev Speaker`)
> 
> @MaddTitan By the way, are you sure the mic is really working?
> Please try recording with mic with the following commands:
> ```
> cd ~/Downloads
> arecord arecord.wav # record to file named arecord.wav
> 
> # play the recording
> aplay arecord.wav
> ```
> For my S3, I can hear only noise 😢

You are right, in my excitement i only accounted for the result i got for the Google assistant. When I record, I get noise as well.

kitakar5525, 2020-04-03:
I also want to confirm that, can Google assistant recognize what you say?

MaddTitan, 2020-04-03:
> I also want to confirm that, can Google assistant recognize what you say?

Yes it can recognize my commands, interestingly it doesn't respond by audio only by text output

sebanc, 2020-04-24:
Hi,

I have a report of wifi being unstable on S3 in issue #42, I would like to know if you have the same issue or if it is stable for you ?

MaddTitan, 2020-04-26:
> Hi,
> 
> I have a report of wifi being unstable on S3 in issue #42, I would like to know if you have the same issue or if it is stable for you ?

No issues here, never had issues with wifi or bluetooth

sebanc, 2020-04-26:
> > Hi,
> > I have a report of wifi being unstable on S3 in issue #42, I would like to know if you have the same issue or if it is stable for you ?
> 
> No issues here, never had issues with wifi or bluetooth

Thanks for the feedback !

sebanc, 2020-04-30:
I think this issue can be closed but don't hesitate to reopen it if something is not working.

closed 2020-04-30 by sebanc