sebanc/brunch #26: chrome://flags shows nothing

Mayk728, 2020-03-27:
Using recovery Ultima R80 with Brunch R79 stable k4.19. I'd like to access flags so that I could disable some things to improve performance on my device. Any suggestions? Thanks.

sebanc, 2020-03-28:
Hi, chrome://flags is not showing on TPM 1.2 recovery images, could you try the latest r80 testing (it has been reported to be stable) with rammus r80 ?

treypop, 2020-03-28:
go to the flags page in incognito mode that should pull it up

treypop, 2020-03-28:
it's worked for me on the latest build

cedricju, 2020-03-28:
Does not work for me on fresh installed new build (No Upgrade). Sleep mode is still not working on my X1 carbon. Will try on my Dell Xps15 later on tonight.

sebanc, 2020-03-31:
@cedricju chrome://flags will not work on your current recovery image which is tpm 1.2. The issues you had with rammus have been corrected, it would probably be better to switch back to it.

Mayk728, 2020-04-08:
Sorry about the late response. I got Rammus installed and it works, as well as other features that weren’t working on Ultima. Thanks!

closed 2020-04-08 by Mayk728

DennisLfromGA, 2020-06-01:
I just ran into this myself.

I have an old Acer C7 Chromebook / parrot / sandy bridge that AUEed Oct. 2017.
I installed the mr.chromebox UEFI firmware prior to trying any of the brunch stuff.

I started with a samus M80 / 12739.111.0 recovery image and brunch_r80_k4.19_stable_20200407.
I'm not sure if I checked `chrome://flags` or not at this point.

I enabled updates and updated to 83.0.4103.77 / 13020.55.0
Again, I'm not sure if I checked `chrome://flags` or not at this point.

I then updated the brunch framework with brunch_r83_k4.19_testing_20200528.
*This is when I first noticed that `chrome://flags` didn't load fully.*

I suppose it's an old TPM issue similar to what chromefy and arnodlthebat were running into also.
There was a [swtpm.tar](https://github.com/imperador/chromefy/issues/40#issuecomment-490380552) that seemed to work for them but I don't know if that would still be relevant.

I think that the samus recovery image is the best fit for this old thing so probably nothing can be done.

I only really needed it to enable `#crostini-disk-resizing` but I guess I may have to wait until it's enabled by default. I tried adding '`--crostini-disk-resizing`' to `/etc/chrome_dev.conf ` but that doesn't seem to work.

Anyway, this is a minor thing and I'm loving having Google Play, Crostini, dev mode, crouton, etc. on this device, never thought that would happen. I was using CloudReady before this and it was actually a pretty good experience too but now I have *everything*. ;-)