sebanc/brunch #27: Play store interface freezes a lot

treypop, 2020-03-28:
The playstore loads for me but the interface freezes a lot. I'm on an Inspiron 15-3521 with a Pentium 2127U running Rammus 80 and the latest r80 testing

sebanc, 2020-03-28:
Thanks for opening your own issue.

How are you using brunch ? (on usb, dual boot or full hdd install)

The first thing you can do (if not already done) is go to chrome://flags and enable hyperthreading in scheduler configuration.

treypop, 2020-03-29:
I'm running off of a HDD and I turned hyperthreading on and I still have the same problem.

sebanc, 2020-03-29:
Could you run the following commands (replace /dev/sda by your device if different):

```
mkdir -p ~/tmp
sudo mount /dev/sda7 ~/tmp
sudo mv ~/tmp/patches/99-io_scheduler.sh ~/tmp/patches/99-io_scheduler.bck
sudo umount ~/tmp
```

then reboot and confirm if it is better ?

treypop, 2020-03-30:
I'm still having the same problems as before. I'm pretty sure its a GUI error and that the GUI isn't updating as it should. I'v been able to get Play Music will play songs but the UI freezes so I know apps are working. If I pull up the shelf (I have it hidden) the GUI updates to what it's suppose to be displaying but then it will freeze again if the shelf hides again.

sebanc, 2020-03-31:
Could you try removing the ui parameter `!--enable-webgl-image-chromium` from /etc/chrome_dev.conf ?

Also, this might be caused by a pure chromeos bug, could you try updating to dev channel and see if it behaves correctly ?

treypop, 2020-03-31:
I'v updated and changed to the dev channel and commented out  `!--enable-webgl-image-chromium` and I am still having the same problem.

Silent-Assassin9, 2020-04-01:
I had the same issue and tried the recommended instructions to fix it, but the display on Playstore/android apps still glitched and/or froze with a black screen.
Strangely enough, changing the display size (via settings) has removed this issue and everything works perfectly.

p.s- Many thanks Sebanc for your work :). Alternative ChromeOS build methods wouldn't work for me, but this has been perfect so far :).

sebanc, 2020-04-02:
@Silent-Assassin9 Thanks for the help and for the feedback :)

treypop, 2020-04-02:
apt install dolphin-emu Thank you, that worked

treypop, 2020-04-02:
Changing the display size (via settings) has removed this issue and everything works perfectly.

closed 2020-04-02 by treypop