sebanc/brunch #29: No camera in Dell XPS 9365 2-in-1

enjewneer, 2020-03-30:
Using the r80 stable release with rammus recovery image, everything seems to work perfectly except the camera on my Dell XPS 9365. I believe this laptop is also very similar to eve, so I tried that as well with the same outcome.

I've attached the output of lsmod and dmesg (using rammus recovery as it's recommended), along with /lib/udev/rules.d/50-camera.rules and /etc/camera/camera_characteristics.conf if those are useful.

[lsmod.txt](https://github.com/sebanc/brunch/files/4403784/lsmod.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/4403786/dmesg.txt)
[50-camera.rules.txt](https://github.com/sebanc/brunch/files/4403787/50-camera.rules.txt)
[camera_characteristics.conf.txt](https://github.com/sebanc/brunch/files/4403788/camera_characteristics.conf.txt)

cedricju, 2020-03-31:
I had great sucess with AZUKA recovery on my 9575. Sensor working, Tablett mode starting automatically. I am using this build now on my X1 Carbon.
PS. this was with the 2603 build. Will test now with the 2903 stable build.

sebanc, 2020-03-31:
I don't think changing recovery image will help for the camera, the previous brunch camera patch had a script issue which has been fixed a few versions ago.

Now, the camera patch is most likely correct but i remain with one problem which seems to be related to commits from google on the chromeos kernel source.

All I can say for now is that I am working on it.

cedricju, 2020-03-31:
Sorry sorry my bad on this one. Someone was talking about sensors and tablet mode. My 2-1 Dell Xps15 was not doing it correctly with Rammus so i tried with AZUKA and it worked straight away :)

enjewneer, 2020-03-31:
Thanks for the hard work, @sebanc ! The project is amazing and I look forward to to using it once the camera for my device is working. Unfortunately, it's not quite a daily driver for me right now given all the video calls due to the pandemic.

@cedricju , thanks for the input! Rammus seems to be working fine for my tablet mode and sensors.

sebanc, 2020-03-31:
@enjewneer No worries I completely understand, i will update this issue once the camera issue appears to be fixed.

sebanc, 2020-05-03:
Hi, just to say I am still working on this issue for "rammus" recovery image but that most users reports that their cameras are working correctly with the "samus" recovery image.

enjewneer, 2020-05-08:
@sebanc I can confirm that the camera works as expected with the Samus recovery image. I would run with this, but I appear to not have access to any Chrome flags with Samus. Is this a known issue?

The camera is partially working with Rammus. I get a flickering black and white image in the camera app, along with a flashing red light on the physical camera.

All tested with 0508.

sebanc, 2020-05-09:
Yes it is a known issue with samus recovery image, chrome://flags can only be changed from guest mode.
I am still trying to fix camera in rammus image but for now I have not found how to do it.

fruchttiger87, 2020-06-17:
I have the same problem with my Dell Latitude 5285 2in1. Everything works great with rammus. Only the camera ist not detected. The infrared camera for Windows Hello shows a black/white and flickering picture in the camera app. Ist would be great to use the camera.

sebanc, 2020-06-17:
Apparently the camera works well with "reef" recovery image also which is very similar to rammus (no ownership issue)

sebanc, 2020-06-26:
The camera issue has been included in #331 "Main issues on which help is needed". I am therefore closing individual issues related to camera not working with rammus recovery.

closed 2020-06-26 by sebanc