sebanc/brunch #30: Dual Boot GRUB config does not work

justcallmedudedude, 2020-03-30:
I have been struggling for the three days trying to get the dual booting to work. I use Grub2 to manage a Ubuntu and Windows 10 dual boot. I have followed the USB install instructions, and ChromeOS boots and works flawlessly on the USB drive. I am using the Rammus 79 recovery image and Brunch r79 k4.19 stable 20200318.

After running the command:
`sudo bash chromeos-install -dst ~/tmp/chromeos.img -s 32`

the script spits out the following grub config:
`menuentry "ChromeOS (boot from disk image)" {
        img_part=/dev/sda5
        img_path=/chromeos.img
        search --no-floppy --set=root --file $img_path
        loopback loop $img_path
        linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
                cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
        initrd (loop,gpt7)/initramfs.img
}`

I then load this into the grub-configurator. When I reboot, I see the entry for the image. However, when I go to launch it, I always receive this error message:
`error: no such device: /chromeos.img.
error: out of memory.
error: disk 'loop,gpt7' not found.
error: you need to load the kernel first.'

sebanc, 2020-03-31:
I cannot help with grub-configurator as I never used it but for ubuntu my suggestion would be to create a file /etc/grub.d/41_chromeos with the following content:

```
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.

menuentry "ChromeOS" {
	img_part=/dev/sda5
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path options=enable_updates,enable_sensors,broadcom_wl
	initrd (loop,gpt7)/initramfs.img
}
```

Then run sudo update-grub and reboot.

justcallmedudedude, 2020-03-31:
> I cannot help with grub-configurator as I never used it but for ubuntu my suggestion would be to create a file /etc/grub.d/41_chromeos with the following content:
>
> ```
> #!/bin/sh
> exec tail -n +3 $0
> # [...]

That's exactly what I did previously (except I named the file `40_custom` instead of `41_chromeos`).
My `40_custom` file looked exactly like this:
```
#!/bin/sh
exec tail -n +3 $0
menuentry "ChromeOS (boot from disk image)" {
    img_part=/dev/sda5
    img_path=/chromeos.img 
    search --no-floppy --set=root --file $img_path 
    loopback loop $img_path 
    linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \ 
         cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
    (loop,gpt7)/initramfs.img 
}
```
I changed it to what you suggested:
```
#!/bin/sh
exec tail -n +3 $0 
menuentry "ChromeOS" {
 	img_part=/dev/sda5
 	img_path=/chromeos.img
 	search --no-floppy --set=root --file $img_path
 	loopback loop $img_path
 	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off \
 		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path options=enable_updates,enable_sensors,broadcom_wl
 	initrd (loop,gpt7)/initramfs.img
}
```
I then ran 
```
sudo update-grub
sudo reboot
```

selected `Chrome OS` and got the exact same error:
```
error: out of memory.
error: disk 'loop,gpt7' not found.
error: you need to load the kernel first.
```

sebanc, 2020-03-31:
Someone told me previously that the recent grub used in ubuntu had issues with loopback devices, could you try replacing ubuntu grubx64.efi with the one attached ? (same file that is included in efi_secure.img)

[grubx64.efi.zip](https://github.com/sebanc/brunch/files/4407450/grubx64.efi.zip)

justcallmedudedude, 2020-04-01:
That worked! Thank you so much!

closed 2020-04-01 by justcallmedudedude

jhedfors, 2020-06-27:
> Someone told me previously that the recent grub used in ubuntu had issues with loopback devices, could you try replacing ubuntu grubx64.efi with the one attached ? (same file that is included in efi_secure.img)
> 
> [grubx64.efi.zip](https://github.com/sebanc/brunch/files/4407450/grubx64.efi.zip)

I can confirm that this works for me as well!  When upgrading Linux Mint 19.3 -> 20, it ChromeOS stopped booting.

ReekyStive, 2020-12-04:
> Someone told me previously that the recent grub used in ubuntu had issues with loopback devices, could you try replacing ubuntu grubx64.efi with the one attached ? (same file that is included in efi_secure.img)
> 
> [grubx64.efi.zip](https://github.com/sebanc/brunch/files/4407450/grubx64.efi.zip)

Thanks! That works for me!