sebanc/brunch #31: cannot boot on surface pro 4 or 5

PierreCsn, 2020-04-04:
Hello i cannot figure out how to boot on surface pro 4 and 5, i have the same issue with project croissant :
i tried  sudo bash chromeos-install.sh  -src chromeos_12739.105.0_rammus_recovery_stable-channel_mp-v2.bin -dst /dev/sdc 
with 3rd party secure boot enabled
i tried  sudo bash chromeos-install.sh -l -src chromeos_12739.105.0_rammus_recovery_stable-channel_mp-v2.bin -dst /dev/sdc 
with no secure boot

I m not sure if tpm should be or not enabled, I tried both option and couldn't figure if it made a difference

I boot to usb and it tries to boot then it reboot to installed os, I managed to boot to a pop os usb to check if i managed to boot a 3rd party os and it booted. 0 luck with chromeos.

sebanc, 2020-04-04:
Hi, it should work on SP5 (i am using one) with or without both secure boot / TPM. The commands you use are correct, i would say the issue probably occurs during the media creation, could you please provide me with the output of the chromeos-install.sh script ?

PierreCsn, 2020-04-04:
All data on device /dev/sdc will be lost, are you sure ? (type yes to continue) yes
umount: /dev/sdc: non monté.
umount: /dev/sdc10: non monté.
umount: /dev/sdc11: non monté.
umount: /dev/sdc12: non monté.
umount: /dev/sdc2: non monté.
umount: /dev/sdc3: non monté.
umount: /dev/sdc4: non monté.
umount: /dev/sdc5: non monté.
umount: /dev/sdc6: non monté.
umount: /dev/sdc9: non monté.
51F840C5-BA5A-054E-AC3F-DB37D9CF7500
       start        size    part  contents
           0           1          PMBR (Boot GUID: 51F840C5-BA5A-054E-AC3F-DB37D9CF7500)
           1           1          Pri GPT header
           2          32          Pri GPT table
    19255304    13602760       1  Label: "STATE"
                                  Type: Linux data
                                  UUID: E4B9C9BF-D2DC-5C43-BC8D-624BA8E339A5
     2117638       65536       2  Label: "KERN-A"
                                  Type: ChromeOS kernel
                                  UUID: 87DEA2AC-3D5F-D541-94C6-3C7DC4477EBF
                                  Attr: priority=15 tries=15 successful=0 
    10866696     8388608       3  Label: "ROOT-A"
                                  Type: ChromeOS rootfs
                                  UUID: DF76E884-5EDB-2C4E-8720-DF665086B303
     2183174       65536       4  Label: "KERN-B"
                                  Type: ChromeOS kernel
                                  UUID: F5FE588D-E61B-CC48-A58B-C463BECCAE39
                                  Attr: priority=0 tries=15 successful=0 
     2478088     8388608       5  Label: "ROOT-B"
                                  Type: ChromeOS rootfs
                                  UUID: DCEAEB97-DF65-ED49-ACFA-4434EB5CF636
       16448           1       6  Label: "KERN-C"
                                  Type: ChromeOS kernel
                                  UUID: EF70BC23-C01B-F847-A04C-139BDDAAC4C0
                                  Attr: priority=0 tries=15 successful=0 
       16456     2097152       7  Label: "ROOT-C"
                                  Type: ChromeOS rootfs
                                  UUID: E4DB9AEE-8500-674A-9A3D-C6E1E7A64CB1
     2248712       32768       8  Label: "OEM"
                                  Type: Linux data
                                  UUID: 87E87012-53BA-FB43-AFF7-6D78B9CE8BE4
     2113608           1       9  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 3F594721-C772-2E4B-9335-A971D0F7C84F
     2113609           1      10  Label: "reserved"
                                  Type: ChromeOS reserved
                                  UUID: 000F7798-DDEC-B740-8FFE-6102CC68FCDC
          64       16384      11  Label: "RWFW"
                                  Type: ChromeOS firmware
                                  UUID: 07C26A83-1773-E948-8A62-4B5B43C286E8
     2412552       65536      12  Label: "EFI-SYSTEM"
                                  Type: EFI System Partition
                                  UUID: 51F840C5-BA5A-054E-AC3F-DB37D9CF7500
    32858079          32          Sec GPT table
    32858111           1          Sec GPT header
Writing partition 1
10,0MiO 0:00:00 [ 169MiB/s] [==================================================================>] 100%            
Writing partition 2
32,0MiO 0:00:00 [ 328MiB/s] [==================================================================>] 100%            
Writing partition 3
2,34GiO 0:01:37 [24,7MiB/s] [==================================================================>] 100%            
Writing partition 4
32,0MiO 0:00:03 [8,75MiB/s] [==================================================================>] 100%            
Writing partition 5
2,34GiO 0:08:14 [4,85MiB/s] [==================================================================>] 100%            
Writing partition 6
 512 O 0:00:00 [4,74MiB/s] [===================================================================>] 100%            
Writing partition 7
1,00GiO 0:00:01 [ 587MiB/s] [==================================================================>] 100%            
Writing partition 8
16,0MiO 0:00:00 [ 157MiB/s] [==================================================================>] 100%            
Writing partition 9
 512 O 0:00:00 [4,70MiB/s] [===================================================================>] 100%            
Writing partition 10
 512 O 0:00:00 [3,59MiB/s] [===================================================================>] 100%            
Writing partition 11
8,00MiO 0:00:00 [ 143MiB/s] [==================================================================>] 100%            
Writing partition 12
32,0MiO 0:00:00 [ 402MiB/s] [==================================================================>] 100%            
ChromeOS installed.

PierreCsn, 2020-04-04:
for your install for the sp5 did you use the -l or not ?

sebanc, 2020-04-04:
The -l argument is only needed for Surface Go, SP5 should boot with or without the "-l".

Otherwise, the media creation log seems fine.

Could you take a video of the boot process (from grub screen to the reboot) so that i can try to identify the issue ? (you can post it in this issue as zip file)

PierreCsn, 2020-04-04:
Maybe I have not been clear, I select USB device as the boot device, I see the 4 square logo like it's attempting to perform the task I asked and back to the 4 square logo and back to EFI menu or windows depending on what is the first boot device in the boot order. if I try to perform the same action with a windows installation from USB it works, same with a pop os, and I cannot figure out why every attempt with ChromeOS fail.
 I have a pro 4 and a pro 5 which I expect to be pretty similar apart from a CPU upgrade and they both have the same issue.
I m currently trying with a 2nd USB drive in case its a hardware problem, but the first one was tried with some old dell laptop and I witnessed a grub with the same drive.
I'll try without the -l since you said it is not necessary and I ll leave secure boot to Microsoft +3rd parties and TPM on

It's super frustrating as the process is not verbose at all, so I was hoping i missed some obvious step, but without any error and every test I did on the side, I feel totally stupid as other people seems to have gone way further than I did.
I've tried swiping USB device from the EFI, I ve tried selecting USB as boot device in the windows  repair menu. both way works out with other Os

sebanc, 2020-04-04:
Ah ok, so the issue is with the efi bios. Not sure it will change something but to boot from usb, i press  the volume volume down button during boot.

PierreCsn, 2020-04-04:
Can you give me efi version of you sp5?

sebanc, 2020-04-04:
234.2706.768

sebanc, 2020-04-04:
Did you try pressing the volume down button on boot, it does something different than just changing the boot order.

PierreCsn, 2020-04-04:
OK I have the same version.
I guess I'll try something stupid like login a live CD on Linux and try to make it install directly on the hard drive.

PierreCsn, 2020-04-04:
The volume down on boot seems to do the same for me. Do you get some kind of menu when you do that?

sebanc, 2020-04-04:
No menu with volume - during boot, it will just boot to the usb drive grub.

You can try to set Internal storage as first boot option and press volume - to boot from usb.

PierreCsn, 2020-04-04:
Thanks a lot for your help. There's something weird. I can see people with the same kind of issues in YouTube comments but no solutions. I'll try a recovery image from Microsoft.

sebanc, 2020-04-04:
Could you verify in the efi bios that on "boot configuration" tab:
- Enable alternate boot sequence is on
- Enable boot from usb device is on
- Enable boot configuration lock is off

PierreCsn, 2020-04-04:
Mindfuck news.
I inserted my USB in the dock instead of the surface itself and it's booting

sebanc, 2020-04-04:
Weird, I don't own a surface dock so i don't know about that, maybe there is a redirection to the dock when it is plugged

PierreCsn, 2020-04-04:
I guess. It's still was booting windows partition and pop_os I'll call that a victory. Anyway thanks a lot I'll restart the process cleanly

closed 2020-04-04 by PierreCsn