sebanc/brunch #32: Any way to map data to a physical partition?

jhedfors, 2020-04-04:
Any possibility to map Downloads, or better yet, the data (Partition1) to a physical partition?

sebanc, 2020-04-05:
Hi,

I assume this question refers to the dual boot process ?

The initial answer would probably be no as the data partition has to be the first partition of the chromeos device (loop device in the dual boot process), however if you tell me what you would like to achieve, it is probably possible to do it without mapping the data partition to a physical partition.

jhedfors, 2020-04-05:
Thanks for the response!

Pretty much the same use case as in the other issue, yes, I am dual boot: I am using this in a shared partition, and want to minimize the single-use dedicated space needed. I have figured out how to at least to add partitions to fstab, and that works to help keep Dowloads from filling up, but as I add Linux and Android apps, the data partition starts to fill up quickly.

I recall years ago, using "mount --bind" for this purpose in linux.

sebanc, 2020-04-06:
Indeed, mount binding would probably be the way to go to achieve this.

However, ChromeOS starts to read from data partition very early in the boot process and already has complex directory bindings so i don't think it would be good idea to perform the directory binding from ChromeOS.

Instead, i would suggest to give more space to the ChromeOS image file and create a boot script to mount the ChromeOS image data partition in linux and bind it to another directory.

jhedfors, 2020-04-06:
I appreciate the advice. When I mount the image file in Linux, I cannot seem to access the data directory, however. Encrypted? Any tips on how to do this?

sebanc, 2020-04-06:
Actually, I am not sure I correctly understood the exact need, depending on what you want to do there are 2 ways to do this:

1) You just want to be able to store files on the chromeos data partition from linux when needed:
Mount partition 1 under linux and create a new directory in it to store your data (indeed, you won't be able to access ChromeOS data as it is encrypted).

2) You want a shared folder to have access from ChromeOS to a linux folder, the following should work but is untested:
- Create a new directory under /var to mount your partition (e.g. /var/shared-partition)
- Run the below commands
```
sudo mount <your linux partition> /var/shared-partition/
sudo mount -o bind /var/shared-partition/<your shared folder path> ~/Downloads
```

jhedfors, 2020-04-06:
Thanks for the clarification with #1. I am already doing #2 successfully now. 👍😀

closed 2020-04-06 by jhedfors