sebanc/brunch #33: Can we adjust size of data?

jhedfors, 2020-04-05:
I found your entry to expand the data size, in reference to USB creation, and it appear to work on the physical partition the same (expanded to max available).

Can we shrink or expand to specific size?

Thanks for you brilliant work on this project!

sebanc, 2020-04-05:
Hi,

The below part of the resize-data script handles the expansion of the data partition:
cgpt add -i 1 -b $(cgpt show -i 1 -b "$source") -s $(( disk_size - $(cgpt show -i 1 -b "$source") - 48 )) -t $(cgpt show -i 1 -t "$source") -l $(cgpt show -i 1 -l "$source") "$source"

Specifically, " -s $(( disk_size - $(cgpt show -i 1 -b "$source") - 48 ))" is used to attribute all the remaining disk space to the data partition and could be replaced by a specific size (in sectors).

Expanding the partition is quite easy as only requires afterwards to also expand the filesystem. However, shrinking partitions would be much more complicated or require to format the filesystem.

With that said, I am not sure what would be your use case for not providing the data partition with all the remaining space ?

jhedfors, 2020-04-05:
Thanks for the info!

The use case is for placing the image not in a dedicated partition, but one shared by other OS. I have successfully done this, but found that I needed more space, but did not want to expand to take over the rest of this shared capacity.

Where can i find the resize-data script to edit it?

sebanc, 2020-04-06:
The resize-data script path is "/usr/sbin/resize-data", though now that i think i understand what you are trying to achieve, i don't think modifying this script will help. I will provide further explanation in your other issue.

closed 2020-04-06 by jhedfors