sebanc/brunch #34: surface 5 crash 100% of time when unplugged from doc

PierreCsn, 2020-04-05:
Hello. I know you don't have a surface dock but you might still want the info as overall the experience on chrome os on surface is outstanding and might attract more users. I noticed that every time I unplug the dock the surface crash and restart.
I have a 2nd dock at work so when I can I'll try to check if the same issue happens.
I noticed on boot with the dock plugged a warning for a bad USB cable. Maybe it's related.

sebanc, 2020-04-06:
Hi, thanks for reporting this. I think the ugly hack I implemented to detect tablet mode when type cover is detached is causing it.

Could you confirm that adding "usbcore.surface_tablet_mode=0" to the kernel command line fixes this issue ?

sebanc, 2020-04-13:
Hi, I made an attempt at fixing this issue in the new testing release, could you confirm if the crash when unplugging the surface dock still occurs ?

PierreCsn, 2020-04-13:
Problem fixed for the dock with the last update.
So far i noticed the camera not working at all, the battery estimate is super innacurate and that s about the only problems appart from android apps not working well but that's more a chromeos issue. do you want me to open some issues or should i consider you have them as well and work on them already ?

sebanc, 2020-04-13:
I am happy to hear that the dock issue is fixed.

Regarding the camera, there is no compatible driver in the linux kernel and it is out of my league so I can just say that if someone makes a driver, i will do my best to include it.

Regarding the battery estimate, I noticed it as well and will try to fix it.

closed 2020-04-17 by PierreCsn