sebanc/brunch #35: No external storage in "Files app"

Ebbot9876, 2020-04-06:
Hi, I have ASUS Transformer Book T200TA. (Rammus R80, Brunch R80 Stable)
External storage does not show up in Files App. 
It worked earlier with Nautilus R79, Brunch R79 Stable.

jhedfors, 2020-04-06:
Sorry to jump in here. Not sure if it is for the same reason as you, but I have been struggling with getting my external USB (2TB) HDD drive to show up as well. I can access in terminal however just not in Files app. Flash drives show up fine for me - what kind of external drive?

Ebbot9876, 2020-04-06:
All kind of external storage, SDCARD, USB-HDD, MMC. Nothing shows up in R80! 
When in R79 with kernel 4.14 everything was mounted under /media/removable/...

sebanc, 2020-04-06:
Hi, I confirm that it is a regression linked to an attempt at fixing the high cpu workload generated by the android media server. This issue should be fixed in the next brunch release which should arrive shortly (tomorrow if everything goes well).

Ebbot9876, 2020-04-06:
Thanks, i am looking forward tog next Brunch!

sebanc, 2020-04-07:
Hi, I have just released the new version which should fix those issues.

Ebbot9876, 2020-04-07:
I Cannes confirm, the newblue version works.

Ebbot9876, 2020-04-07:
I can confirm, the newblue version works. All media mounts to files    app.

jhedfors, 2020-04-07:
Yes, I second that -  I can see my 2TB drive. :)

closed 2020-04-08 by Ebbot9876