sebanc/brunch #37: Smart Lock not working

jhedfors, 2020-04-08:
The option to enable Smart Lock for unlocking with my associated phone is disabled.  I am curious if it just my issue, and if anyone has confirmed it working for them.

xeu100, 2020-04-09:
This sounds like you may be logged in with a g-suite account or have somehow enterprise enrolled the device. Smart lock was automatically enabled on mine when I installed it, however, I have prior to using this set it up on an actual Chromebook. If you're logged in with a work account or a g-suite account in general, it may prevent some settings from being toggled.

jhedfors, 2020-04-09:
My son may have logged into a version of chromium OS I had installed on this device at one point, but since has been wiped away. Any way to reset that?  Thanks!

xeu100, 2020-04-09:
In theory if you do a full reset, Google nor cloudready doesn't do something like that (unless it's either a Chromebook or you install cloudready again). However, even if you sign in with a g-suite account not as a primary account, it may change settings or force settings.

jhedfors, 2020-04-09:
Hmm... He definately has not logged into this computer since it has been wiped.  I wonder if it could be something else.  Does Smartlock require bluetooth by chance? I cannot get the phone to connect via bluetooth, so I wonder if that may be the issue. It sees other bluetooth devices, but not the phone.

xeu100, 2020-04-09:
Bluetooth connecting your phone is different than pairing it. If you can pair it, you should be able to use smart lock.

cedricju, 2020-04-09:
Actually I have 2 Gmail account, one private and one with work.
First login with my private account requires manual login. But after that,
just unlock your phone and it'll log you automatically:). That is not
possible with my Gsuite account

jhedfors, 2020-04-09:
@xeu100 It definately shows as a connected device, but the toggle for Smartlock is disabled. I havent tried to see of a different google account makes a difference.

xeu100, 2020-04-09:
That would be the problem. It doesn't play nice with rooted devices and I have seen people who have rooted talk about this issue. Hopefully, Google plays nice and releases an update.

jhedfors, 2020-04-09:
Strange. I created a new google account, and the same phone actually worked for that account, but not my primary. I can see both accounts connected under Chromebook on my phone.  Sorry, I will close this as it obviousl has nothing to do with this awesome ChromeOS build. :)

closed 2020-04-09 by jhedfors

jhedfors, 2020-04-10:
uI go this working, btw. After I signed out my phone from Google "my devices" page, I was able to sign back in, and add it back into Smart Lock.  I was able to enable it finally. :)