sebanc/brunch #38: Google Assistent not working

fruchttiger87, 2020-04-09:
Hi everybody,
Hi sebanc,
I love your framework. It is working very good on my Dell Latitude 5285 Tablet PC with rammus v80. Nearly everything works as expected. When I installed it the first time I was asked If and how I want to use the Google Assistent. I also had options for the Assistant in the settings. Now I installed it another time and I got no possibility to activate the Assistant. Neither during the first boot nor in the settings. I don't see and find the Assistant in the settings.
Does anyone has an Idea how to activate the Google Assistant. I really wan't to test and use it.

Thanks in advance. And thank you sebanc for you fantastic job here!!!!

Greets
fruchttiger87

sebanc, 2020-04-10:
Hi,

Did you change the system language since first install ? On ChromeOS, google assistant is only available in a few languages:

fruchttiger87, 2020-04-11:
Thank you very much for your answer. That was the solution. I am in Germany and the assistant doesn't work in Germany. So I close this issue because the assistant works.

closed 2020-04-11 by fruchttiger87

MadhiasM, 2021-05-17:
Actually, you can get it to work in German aswell.
https://chromebookz.de/google-assistant-chromebook-aktivieren-nutzen/

You need to add 'Deutsch (Deutschland)' instead of the generic 'Deutsch' and then restart.