sebanc/brunch #39: Fix ap6255 intel atom wifi driver

jjrojoc, 2020-04-11:
Please, add this file to /lib/firmware/brcm/ to fix wifi and bluetooth in inte atom, i'm using z8350 cpu
[brcmfmac43455-sdio.txt](https://github.com/sebanc/brunch/files/4464569/brcmfmac43455-sdio.txt)

sebanc, 2020-04-13:
Hi, your firmware has been included in the new test release and will be included in subsequent ones, could you confirm that it works as expected ?

jjrojoc, 2020-04-13:
of course, tonight i'll test new release, than thank you

jjrojoc, 2020-04-13:
Tested new release with chromeos rammus bootloop from chrome white first start

sebanc, 2020-04-14:
Sorry about that, I think one of my patches did not apply correctly, I will try to provide a fixed version quickly.

sebanc, 2020-04-14:
Hi, I have given you access to the brunch-testing repository. Could you test the release there and verify if the issue is fixed ?

jjrojoc, 2020-04-14:
Thank you, tonight i'll test
I think the mini pc is dead, don't pass from Intel screen

Ebbot9876, 2020-04-14:
I'm also having issues with brcmfmac firmware, I allways copy brcmfmac43340-sdio.txt to /lib/firmware/brcm after upgrade!

jjrojoc, 2020-04-14:
> I'm also having issues with brcmfmac firmware, I allways copy brcmfmac43340-sdio.txt to /lib/firmware/brcm after upgrade!

Yes, i copied in brunch stable version brcmfmac43455-sdio.txt inside rootc.img in /lib/firmware/brcm/ and this working fine, when i create a new chroime os image from brunch wifi firmware neede for mi mini pc is there and working

sebanc, 2020-04-15:
The latest testing version had both your firmwares included but was generating kernel panic with SDIO wifi adapters due to missing kernel patches. I will provide a new update today which will solve the issue.

jjrojoc, 2020-04-15:
Is strange this behavior, only happen in mini pc azw z83ii, 2gb RAM, 32;mmc
Fue run your texting release yesterday i have first, fix pendrive and
second, open chase and it have a internal usb, i think for recovery
propose, and putting pendrive in internal usb have started, can't boot from
any other usb or sdcard

sebanc, 2020-04-15:
Did it work before with other usb or sdcard in previous brunch releases ?

jjrojoc, 2020-04-15:
Yes, tested with usb entries before, perfectly

sebanc, 2020-04-15:
When you try booting from other usb / sdcard, do you see the grub menu appear or it just does not show the devices as bootable ?

jjrojoc, 2020-04-15:
i cant right now bios access and boot process for another usb entrie that
is not the internal, screen stay in intel logo and number 92 square right
down

sebanc, 2020-04-15:
If you put a linux distro image on your usb or sdcard, does it boot ?

jjrojoc, 2020-04-15:
magically i've been able to access bios, restore to default and started
from xubuntu installation in mmc, i don't understand nothing, i'm going
test another usb entries again

sebanc, 2020-04-19:
Hi, with some delay, I have just uploaded a new release which should work out of the box with your wifi adapters.

Ebbot9876, 2020-04-19:
Working fine now!

Ebbot9876, 2020-04-22:
Unfortunately, wifi is not stable after the upgrade! Wifi is disconnecting,I need to disable and enable wifi after a while.

sebanc, 2020-05-03:
Could you try the new testing release and tell me if you still have this issue ?

jjrojoc, 2020-05-03:
Of course, give me one day

jjrojoc, 2020-05-03:
Could be this recovery image better for atom x5?
Acer Chromebook 14 (CB3-431)
Codename = Edgar
This Chromebook use Intel atom x5 e8000

sebanc, 2020-06-02:
No news for a while on this issue so I will close it for now, don't hesitate to re-open it if needed.

closed 2020-06-02 by sebanc

jjrojoc, 2022-03-24:
Wifi is working flawsly in ap6255, thank you @sebanc