sebanc/brunch #40: Include brcmfmac43340-sdio.txt in /lib/firmware/brcm folder

rgunhk, 2020-04-12:
I am using Asus T100HA as my device. Everytime I update brunch framework I have to manually copy the captioned file into the folder in order to make the wifi driver to work. Would you please help to put the file into that folder so that the wifi would work out of the box? Thanks.

sebanc, 2020-04-13:
Hi, your firmware has been included in the new test release and will be included in subsequent ones, could you confirm that it works as expected ?

rgunhk, 2020-04-14:
I cannot confirm since it runs into a bootloop when the newest update is installed. Everytime the device reboots shortly after the chrome screen appeared. What should I do?

sebanc, 2020-04-14:
Sorry about that, I think one of my patches did not apply correctly, I will try to provide a fixed version quickly.

sebanc, 2020-04-14:
Hi, I have given you access to the brunch-testing repository. Could you test the release there and verify if the issue is fixed ?

rgunhk, 2020-04-14:
The testing repo's one has fixed the issue flawlessly as I can connect the wifi from the welcome screen. I am installing it to my hdd now. Thank you so much for fixing it!

closed 2020-04-14 by rgunhk