sebanc/brunch #41: Camera support for OV5648/OV2680

yutouyes, 2020-04-12:
Hello, my device Cube Mix Plus, integrated such cameras OV5648 as main camera and OV2680 as front camera. Neither of them couldn't be found in Chrome OS R80. 
The version of brunch I use is Brunch r80 k4.19 stable 20200407, and recovery image is rammus.
Hardware information is listed as below. 
[lspci.log](https://github.com/sebanc/brunch/files/4466364/lspci.log)
[lsmod.log](https://github.com/sebanc/brunch/files/4466365/lsmod.log)
[dmesg.log](https://github.com/sebanc/brunch/files/4466366/dmesg.log)

Thank you kindly!

sebanc, 2020-04-13:
Hi, I have added missing i2c camera drivers in the new testing build, could you try it and provide a another dmesg  if it does not work (taken just after a fresh start) ?

yutouyes, 2020-04-14:
> Hi, I have added missing i2c camera drivers in the new testing build, could you try it and provide a another dmesg if it does not work (taken just after a fresh start) ?

Tried tesing build with rammus R80, and camera still disappered.
[dmesg.log](https://github.com/sebanc/brunch/files/4472930/dmesg.log)

sebanc, 2020-04-14:
I see in the new dmesg that your cameras are ipu3 and I think they are not supported by the linux kernel.
Could you confirm that they do not work in linux ?

yutouyes, 2020-04-14:
> I see in the new dmesg that your cameras are ipu3 and I think they are not supported by the linux kernel.
> Could you confirm that they do not work in linux ?

Yes. I've tried manjaro 19 and they couldn't be found, either.

sebanc, 2020-04-14:
Unfortunately ipu3 cameras support is not yet included in the linux kernel. If at some point there are drivers, I will try to include them but for now there is nothing I can do. (I have 2 Microsoft Surface devices which have ipu3 cameras so I also hope they become supported at some point)

closed 2020-04-14 by yutouyes