sebanc/brunch #42: Microsoft Surface 3 (Non-Pro) WiFi Issues

GayDingo, 2020-04-13:
WiFi dies on the device at random times and cannot be revived without a reboot.

Running v80 Rammus w/ the 4.19 Kernel. 

This issue can be fixed on standard linux distros by following [these steps](https://www.reddit.com/r/SurfaceLinux/comments/9yt16i/surface_pro_3_ubuntu_update_on_wifi_issues/), but that directory does not exist in ChromeOS.

sebanc, 2020-04-13:
Hi,

The mentioned patch is directly included at kernel level, could you please attach 2 "dmesg" to this issue (one obtained just after boot, one just after a wifi crash) ?

Thanks.

GayDingo, 2020-04-21:
it's happened a few times but i havent been able to reproduce it in a while. I'll keep the issue open until it triggers again.

GayDingo, 2020-04-24:
Got it!

[dmesg_aftercrash.txt](https://github.com/sebanc/brunch/files/4526276/dmesg_aftercrash.txt)
[dmesg_afterboot.txt](https://github.com/sebanc/brunch/files/4526277/dmesg_afterboot.txt)

kitakar5525, 2020-04-24:
Unfortunately, even when we disabled the power_save, the mwifiex firmware still crashes on Surface 3 on heavy traffic or long idle.

Disabling ASPM L1 state can avoid this issue: https://github.com/linux-surface/linux-surface/wiki/Known-Issues-and-FAQ#mwifiex-module-crashing

---

Usually on Linux, disabling L1 state breaks S0ix (thus higher power consumption on suspend) but interestingly on chromeos, not break S0ix. So, once you disabled L1 state, you don't have to re-enable the L1 state for suspend, unlike what the wiki says.

sebanc, 2020-04-24:
@kitakar5525 Thanks a lot for your help once again !

@GayDingo Could you run the below commands in cros shell, reboot and see if it improves wifi stability:
```
sudo mkdir -p /var/brunch/bootscripts
echo "echo 0 > /sys/bus/pci/drivers/mwifiex_pcie/0000:0?:00.0/link/l1_aspm" | sudo tee /var/brunch/bootscripts/mwifiex.sh
sudo chmod 0755 /var/brunch/bootscripts/mwifiex.sh

kitakar5525, 2020-04-24:
and what is worse, we can't control much of ASPM easily until v5.5 series. We can only use the script method written in the wiki on v4.19 series kernel.

GayDingo, 2020-04-25:
> @kitakar5525 Thanks a lot for your help once again !
>
> @GayDingo Could you run the below commands in cros shell, reboot and see if it improves wifi stability:
>
> ```
> sudo mkdir -p [...]

i have ran these. I'll report back in a week to see how things go.  Thank you.

sebanc, 2020-04-25:
@GayDingo apparently the fix I posted will not work with kernel 4.19, i will try to find something else.

@kitakar5525 Sorry, just saw your comment, would completely disabling pcie aspm by adding to the kernel command line "pcie_aspm=off" work ? Is the impact on battery life noticeable ?

GayDingo, 2020-04-26:
> @GayDingo apparently the fix I posted will not work with kernel 4.19, i will try to find something else.
> 
> @kitakar5525 Sorry, just saw your comment, would completely disabling pcie aspm by adding to the kernel command line "pcie_aspm=off" work ? Is the impact on battery life noticeable ?

yep, i was about to post that the wifi ended up crashing again, thanks for the help, however. 

Also, i will mention mine is the LTE model, i'll test cellular once i get the chance to for y'all. Take care!

kitakar5525, 2020-04-26:
@GayDingo 

Here is the script I use to disable L1 state: [aspm-disable-L1-wifi-S3.txt](https://github.com/sebanc/brunch/files/4535176/aspm-disable-L1-wifi-S3.txt)
(rename it from .txt to .sh and execute with sudo. e.g. `sudo bash aspm-disable-L1-wifi-S3.sh`)

(redistributing script from https://wireless.wiki.kernel.org/en/users/documentation/aspm#enabling_aspm_with_enable_aspm with modification to disable L1 state for Surface 3 wifi to avoid firmware crashing, following the permission notice)

Hmm. I need to update the wiki in linux-surface later for clarity and copyright, including
- add the direct link to the script (which I posted here)
- add the link to the wireless.wiki.kernel.org that distributes the original script

kitakar5525, 2020-04-26:
@sebanc 

- I tried with kernel command line `pcie_aspm=off` but according to `lspci`[1], ASPM seems to be actually not disabled (?) I might be misunderstanding
- If I understand correctly, adding kernel command line `pcie_aspm.policy=performance` (or runtime: `echo performance | sudo tee /sys/module/pcie_aspm/parameters/policy`) has the same effect as `pcie_aspm=off`. And as expected, it broke S0ix. So, we usually don't want to to use `pcie_aspm.policy=performance` (and also `pcie_aspm=off` if I assume that it also has the same effect as `pcie_aspm.policy=performance`).

[1] You can show ASPM states by like this: `sudo lspci -nnvvv | grep -iE '^..:..\..|aspm'`

linux-surface are trying to avoid firmware crashing at kernel level but as I mentioned before, disabling L1 state usually breaks S0ix. So, we ensure that L1 gets re-enabled before suspend...

I'm curious why `pcie_aspm.policy=performance` breaks S0ix but not break S0ix when I use the script (aspm-disable-L1-wifi-S3.sh) on chromeos. Using the script on Linux breaks S0ix anyway.

sebanc, 2020-04-26:
@kitakar5525 Thanks a lot again for your support again ! I am getting tired of the mwifiex driver issues, I think I will disable aspm in the mwifiex driver in my next release even if it means breaking S0ix.

kitakar5525, 2020-04-26:
@GayDingo ah, you need to run that script on every boot or you can automate executing the script by copying that script into `/var/brunch/bootscripts` then adding executable permission.
For example:
```bash
sudo mkdir -p /var/brunch/bootscripts
sudo cp aspm-disable-L1-wifi-S3.sh /var/brunch/bootscripts/
sudo chmod 0755 /var/brunch/bootscripts/aspm-disable-L1-wifi-S3.sh
```

@sebanc 
Hmm. I'll let you know when some progress is made in this issue. I think just adding that script just works for now hopefully without breaking S0ix on chromeos, though.

sebanc, 2020-05-03:
Hi, I disabled aspm for the mwifiex driver in the new testing release, could you see if you still have this issue ?

sebanc, 2020-06-02:
I believe this issue is fixed so I will close it, don't hesitate to re-open it if needed

closed 2020-06-02 by sebanc