sebanc/brunch #43: Black scree on Android video app

thevoice89, 2020-04-13:
Hi,
brunch works perfectly but, for example, on youtube or sky go app, when I try to play a video there's a black screen. Apps work, all menus and functions, only videos are black.
is there an android codec to install or something else?
thanks

sebanc, 2020-04-13:
Hi, could you try the testing build I have just released ?

thevoice89, 2020-04-13:
> Hi, could you try the testing build I have just released ?

This night I'll try it.. thanks for your work!

sebanc, 2020-04-13:
If the new release does not fix it, could you try adding "!--enable-features=Pepper3DImageChromium" to at the end of the /etc/chrome_dev.conf file ?

thevoice89, 2020-04-13:
> If the new release does not fix it, could you try adding "!--enable-features=Pepper3DImageChromium" at the end of the /etc/chrome_dev.conf file ?

New brunch's version doesn't solve problem..
I would try second suggestion, but where is /etc/ ? Or better, how can I have access at this folder and modify this file? Directly from chrome os threw terminal or before flashing on SD card?

sebanc, 2020-04-14:
Press CTRL-ALT-T and type "shell" at the invite.
Then run "sudo nano /etc/chrome_dev.conf" and make the modification (use CTRL+SHIFT+V to paste text inside nano).
CTRL-X to exit
and reboot

thevoice89, 2020-04-15:
It works for YouTube..I think that It's a SkyGo problem..
By the way..thank you very much..I have a chrome box now!

sebanc, 2020-04-16:
Nice to know that youtube now works :) I will add the fix in the next version.

Regarding SkyGo I don't know the app but there are a few threads on reddit saying that it does not work on an external monitors.

thevoice89, 2020-04-16:
I have a little Nuc without monitor. My TV is my monitor..maybe he think that I use a notebook. Is there a file for identifying pc? Can I say at the system "ehi I'm a Chromebox not a Chromebook!"?

sebanc, 2020-04-16:
It seems to be for drm purpose that the display has been disabled on external monitors, I don't think you can trick that. (it does also not work on genuine chromeboxes as far as I can tell)

thevoice89, 2020-04-16:
They are going to release an upgrade for compatibility with rammus, but due to Covid , is currently on hold. I hope they solved this problem..

From: official sky community thread

sebanc, 2020-04-19:
Hi, I could not add the youtube fix by default as it breaks crostini, but since the new release you can add "options=youtube_fix" to the kernel command line so that the fix is automatically applied.

anubthomas, 2020-04-20:
Hi, I have the same issue with Youtube Kids android app. Youtube works fine. I can head the sound but a black screen for video.

thevoice89, 2020-04-21:
> Hi, I could not add the youtube fix by default as it breaks crostini, but since the new release you can add "options=youtube_fix" to the kernel command line so that the fix is automatically applied.

Ok thank you! But how can modify kernel code?

sebanc, 2020-04-22:
The fact that youtube works with  "!--enable-features=Pepper3DImageChromium" most likely means that your gpu does not support RGBA8888 native GpuMemoryBuffers. Sorry but as far as I can tell, only intel kernel devs could maybe do something about this and it is clearly out of my league.

sebanc, 2020-05-03:
Hi, I have fixed an issue with black videos in youtube in the new testing release, the issue was not related to "!--enable-features=Pepper3DImageChromium" but we never know, could you give it a try ?

thevoice89, 2020-05-08:
Yes of course!

thevoice89, 2020-05-13:
> Hi, I have fixed an issue with black videos in youtube in the new testing release, the issue was not related to "!--enable-features=Pepper3DImageChromium" but we never know, could you give it a try ?

It works! Unfortunately, even After Skygo update, It doesn't play videos

sebanc, 2020-05-13:
Nice to know that the youtube issue is fixed.

The Skygo app seems to be a specific issue, if really it is as Skygo support mentioned an issue with rammus device, you could try using another image like samus.

thevoice89, 2020-05-13:
But Samus Is for old devices,isn't it? I have a celeron n3350 processor, Is It ok?

sebanc, 2020-05-18:
Samus recovery image works fine in a general way, it only has a minor issues like chrome://flags have to be changed from guest mode.

Else, you can also try the octopus image.

closed 2020-06-02 by sebanc