sebanc/brunch #44: Changing Channel

jhedfors, 2020-04-13:
In your latest test release, you mention that it can be used on the Dev channel to try out version 83 new functionality.  Are there recovery images avalable for different channels, or do you have to just switch channel, and the apply the test Brunch build similar to the instuctions for automatic updates?

sebanc, 2020-04-13:
No, there are no recoveries for dev channel. If you want to try it, I recommend to update brunch first, then add options=enable_updates to the kernel command line, reboot and change channel to dev in settings.

jhedfors, 2020-04-13:
Cool. Thanks for the information!

closed 2020-04-13 by jhedfors

jhedfors, 2020-04-13:
I was fully expecting an automatic Powerwash when switching channels, but it did not. Confused, but pleasantly surprised... :)  Booted right up and Linux and Android apps both work!

sebanc, 2020-04-13:
Nice that it worked :)

Just for info, there will be a powerwash if you revert to stable.

jhedfors, 2020-04-13:
Got it. Thanks!