sebanc/brunch #45: WiFi not working

OCActivated, 2020-04-13:
Hello. Following the guide, I managed to make a bootable USB. The Chrome OS setup screen has opened, but I cannot use WiFi. I cannot see any network in Select network section.

anubthomas, 2020-04-14:
I had a similar issue. Could you try the steps mentioned in the below bug. It solved the issue for me.
https://github.com/sebanc/brunch/issues/19

sebanc, 2020-04-14:
@OCActivated What is your laptop model ?

OCActivated, 2020-04-14:
> @OCActivated What is your laptop model ?

Lenovo G50-30.

OCActivated, 2020-04-14:
I pressed Ctrl+Alt+F3 and it opened developer console but it wants a password.

sebanc, 2020-04-14:
Indeed the solution provided by @anubthomas seems to be the right one.

At boot, when you see the grub screen, quickly press "e".
Write "module_blacklist=ideapad-laptop" after cros_debug and then F10 to boot.
Verify if you have wifi.

If you have, finish the setup, open cros shell and run "sudo edit-grub-config" to add the "module_blacklist=ideapad-laptop" in a persistent way.

OCActivated, 2020-04-14:
Nope. I typed "module_blacklist=ideapad-laptop" after "cros_debug" but it didn't worked. I can only access internet by using ethernet cable. Also I'm getting error on the screen like this:
[image]

sebanc, 2020-04-14:
Ah great, since you can login, could you send me a dmesg (taken after a reboot):
Open cros shell
Type "dmesg > ~/Downloads/dmesg.txt"
And post here the dmesg.txt file from your downloads directory.

OCActivated, 2020-04-14:
It says "ERROR : unknown option: > ~/Downloads/dmesg.txt

sebanc, 2020-04-14:
I think you were not completely in cros shell, to open cros shell press CTRL-ALT-T and type "shell" at the invite.
Once you are there, run "dmesg > ~/Downloads/dmesg.txt"

OCActivated, 2020-04-14:
Okay, It worked. 
[dmesg.txt](https://github.com/sebanc/brunch/files/4474361/dmesg.txt)

sebanc, 2020-04-14:
Could you try to add "options=broadcom_wl" to the kernel command line ?

Go to cros shell again:
- run "sudo edit-grub-config"
- add "options=broadcom_wl" after cros_debug
- Press CTRL-X to quit and y to save when requested

Reboot your laptop

OCActivated, 2020-04-14:
Oof. Idk why but brunch started to rebuild rootfs again, my toaster takes a long time to do this.

OCActivated, 2020-04-14:
**Yeah, it worked!** I'm now successfully connected to my network. But I have 2 more problems. Shall I continue the issue here or open a new issue?

sebanc, 2020-04-14:
Yes, you can continue here, what are the other issues ?

OCActivated, 2020-04-14:
I get an error, as it appears in the photo I took.

> Nope. I typed "module_blacklist=ideapad-laptop" after "cros_debug" but it didn't worked. I can only access internet by using ethernet cable. Also I'm getting error on the screen like this:
> [image]

sebanc, 2020-04-14:
For the android apps issue, could you try to reset google sync:
https://support.google.com/chromebook/answer/2914794?hl=en

And then do a powerwash.

OCActivated, 2020-04-14:
Nice, now I can access to Google Play Store. Now, I want to wipe everything on the laptop and install Chrome OS to my laptop permanently. There are parted areas on the hard disk. I want to combine them all into one partition but how?

OCActivated, 2020-04-14:
Thank you so much. I did a powerwash and now i can access to Google Play Store. 👍

closed 2020-04-14 by OCActivated