sebanc/brunch #46: Crouton couldn't start xfce4: Cannot open /dev/tty0 (Permission denied) . Cannot open virtual console 2 (Permission denied)

yutouyes, 2020-04-14:
Device information:
```
chronos@localhost / $ sudo edit-chroot -all
name: xenial
encrypted: no
Entering /mnt/stateful_partition/crouton/chroots/xenial...
crouton: version 1-20200116214215~master:9017d640
release: xenial
architecture: amd64
xmethod: xorg
targets: xfce
host: version 12739.105.0 (Official Build) stable-channel rammus 
kernel: Linux localhost 4.19.98-brunch-sebanc #1 SMP PREEMPT Mon Apr 13 12:03:17 CEST 2020 x86_64 x86_64 x86_64 GNU/Linux
freon: no
Unmounting /mnt/stateful_partition/crouton/chroots/xenial...
```

I freshly installed Brunch r80 k4.19 testing 20200413 with rammus R80, and went straight to crouton via `sudo crouton -t xfce -u` but met error:

```
chronos@localhost / $ sudo startxfce4
(EE) 
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE) 
(EE) 
```

I fellow https://github.com/dnschneid/crouton/issues/3320 and entered 
```
chronos@localhost / $ sudo enter-chroot
Entering /mnt/stateful_partition/crouton/chroots/xenial...
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.
(xenial)yucr@localhost:~$ sudo chmod 666 /dev/tty0
[sudo] password for yucr: 
(xenial)yucr@localhost:~$ ls /dev/tty0 -lah
crw-rw-rw-. 1 root tty 4, 0 Apr 14 11:46 /dev/tty0
sudo usermod -a -G tty  yucr
```

/etc/X11/Xwrapper.config is set as allowed_users=anybody, and packages ```xserver-xorg-legacy``` ```xserver-xorg-legacy-hwe-16.04``` are not installed, so I have nothing to purge.

Then ran startxfce4, another error here:
```
(EE) 
Fatal server error:
(EE) xf86OpenConsole: Cannot open virtual console 2 (Permission denied)
(EE) 
(EE) 
```
Xfce4 couldn't start. I've tried bionic and sid, all of them dropped into the same error. Will there be sth relavant to the host system? My platform is Intel Core-m3 7y30 without Nvidia or AMD graphics. It should be of great compatibility.

sebanc, 2020-04-14:
Hi, could you try to add "console=" to the kernel command line and then reinstall crouton ?

yutouyes, 2020-04-14:
> Hi, could you try to add "console=" to the kernel command line and then reinstall crouton ?

I run `sudo edit-grub-config` and:
```
menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug
  initrd ($disk,7)/initramfs.img
}
```
What should I do is add `console=` just behind cros_debug like that?
```
menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug console=
  initrd ($disk,7)/initramfs.img
}
```
Or I should add `options=console=` like https://github.com/sebanc/brunch#framework-options ?

sebanc, 2020-04-14:
Yes, this is perfect:
```
menuentry "ChromeOS" {
  regexp --set disk "(^.+)(,gpt)" $root
  linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug console=
  initrd ($disk,7)/initramfs.img
}
```

yutouyes, 2020-04-14:
> Yes, this is perfect:
> 
> ```
> menuentry "ChromeOS" {
>   regexp --set disk "(^.+)(,gpt)" $root
>   linux ($disk,7)/kernel boot=local noresume noswap loglevel=7 disablevmx=off cros_secure cros_debug console=
>   initrd ($disk,7)/initramfs.img
> }
> ```

Sadly doesn't work. Still "xf86OpenConsole: Cannot open virtual console 2 (Permission denied)" 
I even tried 'startxfce4' in tty2, and error was simillar  "xf86OpenConsole: Cannot open virtual console 1 (Permission denied)", only 2 to 1

sebanc, 2020-04-14:
Could you attach a dmesg to this issue (taken just after a reboot) ?

yutouyes, 2020-04-14:
> Could you attach a dmesg to this issue (taken just after a reboot) ?

Sure
[dmesg-host.log](https://github.com/sebanc/brunch/files/4474771/dmesg-host.log)
dmesg-host.log was catched in host system.
[dmesg-chroot.log](https://github.com/sebanc/brunch/files/4474773/dmesg-chroot.log)
dmesg-chroot.log was catched in crouton chroot system.
Wish they can help.

sebanc, 2020-04-14:
The dmesg seems fine, I actually never tried to install crouton, I will try it on my install and see if I have the same issue. It might be easier to debug in that case.

yutouyes, 2020-04-14:
FYI I tried crouton in fydeos and managed to X. All xfce componments work well

sebanc, 2020-04-15:
Hi, even though I don't really understand its origin for now, I agree that the issue is on my side. You might want to close the one opened on crouton repository.

yutouyes, 2020-04-15:
@sebanc got it. I've closed that. Does the problem appear in your device?
If anything I can help feel free to mention me.

sebanc, 2020-04-15:
Yes, the problem appears also on my device, I did not really know what was crouton before but I looked into it a bit and it theoretically should work.

For now, I think the issue might be related with the way I chroot from the initramfs to chromeos, I am looking into that.

sebanc, 2020-04-16:
> @sebanc got it. I've closed that. Does the problem appear in your device?
> If anything I can help feel free to mention me.

I am still looking into this issue but meanwhile could you try to see if it the same problem happens when using the "eve" recovery image ?

yutouyes, 2020-04-16:
> > @sebanc got it. I've closed that. Does the problem appear in your device?
> > If anything I can help feel free to mention me.
> 
> I am still looking into this issue but meanwhile could you try to see if it the same problem happens when using the "eve" recovery image ?

I built system via Brunch r80 k4.19 testing 20200413 with eve R80. 
chrome://version/ shows my platform is 12739.111.0 (Official Build) stable-channel eve.Just trie and error is 
```
(EE) 
Fatal server error:
(EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)
(EE) 
(EE) 
```
 I didn't do chmod as it's said 'useless'. Seems eve image doesn't help

sebanc, 2020-04-16:
Thanks for trying it, it confirms that it's not a specific issue with rammus image.

On my end, I finished testing different things with the initramfs chroot and it seems not to be the origin of the issue as far as I can tell.
I am now looking at the kernel config.

sebanc, 2020-04-22:
Hi, just to say I am still looking into this issue, I tried many things related to initramfs, kernel, security features but for now none worked

yutouyes, 2020-04-22:
> Hi, just to say I am still looking into this issue, I tried many things related to initramfs, kernel, security features but for now none worked

Ah..Sry for helping you little. Will there be anything out of these places that blocking /dev/tty0? In fydeos, crouton can directly access tty0 without anything removed or modified.

sebanc, 2020-04-22:
I am starting to think that the chronos user permissions are not correctly set in debug mode by only adding cros_debug to the kernel command line. I also have a personal issue, not crouton related, where qemu cannot access kvm device when launched without sudo even when chronos user is added to the right group.

yutouyes, 2020-04-22:
> I am starting to think that the chronos user permissions are not correctly set in debug mode by only adding cros_debug to the kernel command line. I also have a personal issue, not crouton related, [...]

But as for crouton, the way to launch X is using sudo, like ```sudo startxfce4 ```. Does it mean the error of /dev/tty0 is not so relevant to to the chronos user permissions?
And the kvm, can qemu access kvm device with sudo?

sebanc, 2020-04-22:
I am not sure I understand correctly crouton as I only started looking into it recently but I think the sudo is only used to perform the chroot but then it maps the user profile to chronos user permissions and startxfce command within crouton is not run as root.

Yes, with sudo qemu can access kvm.

santhoshr, 2020-04-29:
After running crouton from CloudReady where native crostini does not work, I was so impressed with chroot based crouton, having multiple distros with multiple desktop environments, and gpu acceleration and usability is more than what I expected, for 12 years old MacBook, even OSX Lion crawled, so chromeos/cloudready/brunch wins! + crouton, I quickly converted my Surface Book 2 Crostini to Crouton, I'll stop my rant!! please fix this ASAP can't use Xorg way of loading distros when running brunch,

I have also ran with additional sudo from crosh shell as root user, I got no tty present error,  

Running _localhost ~ #_ `enter-chroot -t xfce \exec sudo startxfce4 -X xorg`

Entering /mnt/stateful_partition/crouton/chroots/buster...
**sudo: no tty present and no askpass program specified**
Unmounting /mnt/stateful_partition/crouton/chroots/buster...
Sending SIGTERM to processes under /mnt/stateful_partition/crouton/chroots/buster

sebanc, 2020-05-03:
Hi, I am happy to say that I have good news and that you should try the latest release.

yutouyes, 2020-05-05:
> Hi, I am happy to say that I have good news and that you should try the latest release.

Confirmed. It works flawless. Thank you very very much! You save my life!

closed 2020-05-05 by yutouyes

santhoshr, 2020-05-05:
I too confirm, it works well. Thank you.

DennisLfromGA, 2020-07-19:
Just wondering about the 'fix' noted above, is it supposed to have `/dev/tty0` now?  

I'm on an Acer C7 with a samus vers. 83.0.4103.119 and brunch r83 k4.19 20200713 and when I attempt to launch a crouton desktop I'm getting the error:
> (EE) parse_vt_settings: Cannot open /dev/tty0 (Permission denied)

Thanx,
-**DennisLfromGA**