sebanc/brunch #49: Add YogaBook Drivers

nolanchromium, 2020-04-16:
Is there any way to add Lenovo Yoga Book Halo keyboard and drivers?

https://github.com/jekhor/yogabook-linux-kernel

sebanc, 2020-04-16:
I would like to add the minimum of kernel patches as they are complicated to maintain, could you tell me which is the specific commit for the keyboard ?

sebanc, 2020-04-16:
Ok I think I got it, I was not finding the keyboard patch in the kernel as in the end it appears to be mostly a userspace binary which is built as part of the chromium source:
https://github.com/jekhor/chromiumos_touch_keyboard/

The issue is that I am not building chromiumos from source, just using google prebuilt images so I cannot build it. Nevertheless, if you provide me with the already compiled files from a chromiumos distro, we might be able to use them.

Could you attach (as zip) to this issue the 3 below files taken from a chromiumos distro which supports your keyboard (most recent chromiumos release possible) ?

/usr/bin/touch_keyboard_handler
/opt/google/touch/policies/touch_keyboard_handler.policy
/etc/init/touch_keyboard.conf

nolanchromium, 2020-04-16:
Can you have a look and see if https://github.com/jekhor/yogabook-linux or https://github.com/jekhor/yogabook-linux-kernel are useful. I appreciate you prompt reply and willingness to help out. I stumbled across your amazing project and I am still learning about how all of this works. If its useful great, if it is not I will keep researching. I reached out to that developer regarding a chromiumos distro as I couldnt find exactly which method he was using to load these drivers.

Thanks a million either way. Have a great day.

sebanc, 2020-04-17:
Hi, I had a look at those and as far as I understand, to get the keyboard working it needs this kernel patch:
https://github.com/jekhor/yogabook-linux-kernel/commit/bd3a5953126fd87e4218550c5a31baafcdc60a38

And the 3 files I previously mentioned:
/usr/bin/touch_keyboard_handler
/opt/google/touch/policies/touch_keyboard_handler.policy
/etc/init/touch_keyboard.conf

I can include the kernel patch but if I understand well, it will be useless without the binaries.

sebanc, 2020-05-18:
I close this issue for now, don't hesitate to re-open it if you have some news

closed 2020-05-18 by sebanc