sebanc/brunch #50: Unable to Boot

axeriox, 2020-04-16:
Hi there, I'm having issues booting as it kernel panics upon selecting the USB I created using the [guide](https://github.com/sebanc/brunch#install-chromeos-on-a-usb-flash-drive--sd-card-1) in the Readme.

Attached below is the verboose logs before it restarts the computer. 

[image]

System Specs:
Model: Medion Erazer - X7859 
CPU: Intel i5 8300H (TPM Enabled, Secure Boot Disabled, SGX Disabled)
RAM: Single 8GB Stick
GPU: Nvidia GTX 1060 (Wasn't running as MSHybrid was enabled in the BIOS and the laptop display **should** only run on the iGPU)
GPU2: Intel UHD 630

sebanc, 2020-04-16:
Hi,

Thanks for the printscreen and the detailed description.

Could you post a short video of the boot process (the source issue most probably occurs prior to the printscreen you posted) ?

axeriox, 2020-04-16:
Here's the video you requested. 

https://gfycat.com/uglybewitchedindianhare

sebanc, 2020-04-16:
Thanks for the video, it seems that the partition /dev/sdc5 is corrupted. Could you re-download the chromeos recovery and try again the process ?

axeriox, 2020-04-16:
Alright.

sebanc, 2020-04-16:
Forgot to ask, what size is your usb flashdrive ?

axeriox, 2020-04-16:
16 GB.

sebanc, 2020-04-16:
ok 16gb is perfect

axeriox, 2020-04-16:
Okay I have finished creating the .img file I'll proceed to using Rufus to write the file onto the USB

axeriox, 2020-04-16:
Right USB flashed I'm gonna reboot and record for any changes

axeriox, 2020-04-16:
[image] Hi again, the installation was going fine until it froze at this, any idea how to fix it?

sebanc, 2020-04-16:
Everything is normal until here, it did not display the welcome screen ?

axeriox, 2020-04-16:
I meant everything was going fine in this text console but then it just froze after that line

sebanc, 2020-04-16:
ok i was just confirming cause that is the exact point where chromeos welcome screen is supposed to be displayed.

Actually I think, chromeos i running but there is a display issue.

 Could you reboot, press "e" quickly when you see the grub screen and add "fbcon=map:1" after cros_debug ?

axeriox, 2020-04-16:
[image]
Going to try that

axeriox, 2020-04-16:
Hi sorry for not replying in a while, I'm having issues with the USB being too slow. So I'm going to retry it via the Linux [method](https://github.com/sebanc/brunch#dual-boot-chromeos-from-your-hdd).

sebanc, 2020-04-19:
Hi, I tried to improve the support for dual gpu in the new release. Could you give it a try without modifications first and if it does not work with "fbcon=map:1" added to the kernel command line ?

closed 2020-04-28 by sebanc