sebanc/brunch #54: Linux (Beta) Support

sgtstadanko, 2020-04-18:
Any tips on how to get Linux (Beta) working? I built with 80 rammus as suggested and followed all the guides, enabled VM support in chrome://flags, but no option to enable Linux in the settings.

cedricju, 2020-04-19:
I had that issue a couple of times.
Don't try to enable developper mode the first time you boot as it is
enabled per default. Then you ll have Linux beta in the settings.
C

sebanc, 2020-04-19:
Hi, the most likely reason for linux (beta) not appearing is virtualization (vt-x) not being enabled in the bios. Could you check that ?

As a side note, linux (beta) is directly available in rammus and does not need modification or flag to be changed.

sgtstadanko, 2020-04-19:
Yes! It was the virtualization settings in BIOS. Thanks so much!

closed 2020-04-19 by sgtstadanko