sebanc/brunch #56: Internal microphone not working

frigis1, 2020-04-19:
This is on an Acer Spin SP111-31. Internal microphone is not working, but if I plug in a headset with a microphone, the headset mic works fine. Currently using r80 k4.19 stable - r81 k4.19 testing does not work on this laptop.

sebanc, 2020-04-20:
Hi, the audio in ChromeOS depends on the linux kernel support and generally works the same way as in linux.
Unfortunately, browsing linux forums for your device or similar ones, I found several reports of microphone not working but no solution so it does not seem like I there is something I can do.
Don't hesitate to have a look as well and see if you find something interesting.

I am also interested by the second part of the issue as the recent testing versions are the base for future stable releases. Could you tell me what is the issue with testing versions ? (do not boot, no android apps support...)

GGorAA, 2020-04-20:
I have the same issue, device HP EliteBook 840 G3

frigis1, 2020-04-20:
Not sure how much this helps, but mic seems to work fine on Lubuntu 19.10 on a live USB. As for the testing versions, r80 k4.19 testing worked fine, but when I updated to r81 testing using the recommended method, the laptop shuts down a few seconds after the login screen. I also tried installing it from scratch with the same results (shutdown a few seconds after welcome screen). Went back to r80 stable with no problems (microphone aside).

And thank you for the prompt reply. This is amazing work!

nicktelindert, 2020-04-27:
My internal mic also isn't working at all. Which is really weird because it is linux compatible. The laptop came pre-installed with linux and i have done a lot of distro hopping and never had that problem before. It is a Dell XPS 13 9380

nicktelindert, 2020-04-27:
I did some testing, the internal mic works when i plugin a headphone(without mic). I will further look into this.

nicktelindert, 2020-04-27:
I get the internal mic working with the following:
amixer -c0 set "Internal Mic" toggle
amixer -c0 set "Capture" 40%

After reboot the Internal Mic is turned off again.

frigis1, 2020-04-27:
Interesting, this works for me, too! This setting also resets after reboot for me.

After tinkering a bit, I think I see the problem for my case. There are two toggles, the headset mic and internal mic. Enable one, and the other is automatically disabled. Presumably the headset mic should automatically enable when you plug in headphones, and the internal mic gets enabled when nothing's plugged in.

In my case, the toggle is stuck on headset mic whether or not headphones are plugged in, which means my internal mic is always disabled, but headset mic is fine if I plugged one in. I can manually enable internal mic thanks to the commands you posted, but then the headset mic is disabled.

Thank you for that post, nicktelindert!

sebanc, 2020-04-28:
> I get the internal mic working with the following:
> amixer -c0 set "Internal Mic" toggle
> amixer -c0 set "Capture" 40%
> 
> After reboot the Internal Mic is turned off again.

Great find !

If you want to automate it, you can create a bootscript by running in crosh shell:
```
sudo mkdir /var/brunch/bootscripts
echo 'amixer -c0 set "Internal Mic" toggle' | sudo tee /var/brunch/bootscripts/sound.sh
echo 'amixer -c0 set "Capture" 40%' | sudo tee -a /var/brunch/bootscripts/sound.sh
sudo chmod 0755 /var/brunch/bootscripts/sound.sh
```

jordanvanbergen, 2020-04-30:
> I get the internal mic working with the following:
> amixer -c0 set "Internal Mic" toggle
> amixer -c0 set "Capture" 40%
> 
> After reboot the Internal Mic is turned off again.

The first command works on my Lenovo Ideapad Yoga 13 and I thought YES the internal mic is going to work but the second line generates this error:
amixer: Unable to find simple control 'Capture' ,0

frigis1, 2020-04-30:
Try running:
amixer scontrols
It should come up with 'Master',0 and something else below that - try using that name (not master, but the other one) instead of "Capture."

hacker1024, 2020-05-01:
@nicktelindert, what do those commands do? They fix it, but why 40%?

jordanvanbergen, 2020-05-01:
> Try running:
> amixer scontrols
> It should come up with 'Master',0 and something else below that - try using that name (not master, but the other one) instead of "Capture."

I get this output

amixer scontrols
Simple mixer control 'Master',0
Simple mixer control 'Capture',0

So it's Capture but I get:

amixer -c0 set "Internal Mic" toggle
Simple mixer control 'Internal Mic',0
  Capabilities: cvolume cvolume-joined cswitch cswitch-joined
  Capture channels: Mono
  Limits: Capture 0 - 80
  Mono: Capture 0 [0%] [-74.00dB] [on]

and:

amixer -c0 set "Capture" 40%
amixer: Unable to find simple control 'Capture',0

frigis1, 2020-05-02:
@jordanvanbergen Maybe try: amixer set "Capture" 40% (without -c0)
@hacker1024 The commands seem to enable the internal mic, and then set "capture" (input) to 40% volume, you can set it to 100% volume if you want.

sebanc, 2020-05-03:
Could you confirm that the issue persist with the new testing release ?

jordanvanbergen, 2020-05-03:
> Hi, could you confirm that the issue persist with the new testing release ?

Will try today. This will be the 2nd time I am going to upgrade brunch & Chrome OS. I assume I will have to do this: 

sudo chromeos-update -r < path to the ChromeOS recovery image > -f < path to the Brunch release archive >

sebanc, 2020-05-03:
yes if you want to update both, if you only want to update brunch:
`sudo chromeos-update -f < path to the Brunch release archive >`

hacker1024, 2020-05-03:
The issue is still there for me.

jordanvanbergen, 2020-05-03:
> yes if you want to update both, if you only want to update brunch:
> `sudo chromeos-update -f < path to the Brunch release archive >`

Installed new Brunch release. How can I tell it's been installed correctly? Can I see a version number somewhere from command line ? 

So it's updated correctly -> Chrome OS update tells me I am on the latest version (even with or without options=enable_updates

But the internal mic is not working. 

https://imgur.com/7gwlSXA

it shows in skype for example or other programs. But it's not working.

frigis1, 2020-05-03:
@sebanc Mic issue still persists - microphone still has no jack detection. The good news is, the laptop no longer shuts down after the welcome screen (note: still using rammus).
@jordanvanbergen One more thing you can try, punch in the following:
sudo nano /etc/modprobe.d/alsa.conf
Then add the following at the end of the file:
options snd-hda-intel model=generic
Then hit ctrl+x, hit Y to save, then reboot.
For me, this disables the headset mic, but enables the internal mic.

sebanc, 2020-05-03:
@frigis1 ok thanks, i will keep looking into it

jordanvanbergen, 2020-05-03:
> @sebanc Mic issue still persists - microphone still has no jack detection. The good news is, the laptop no longer shuts down after the welcome screen (note: still using rammus).
> @jordanvanbergen One more thing you can try, punch in the following:
> sudo nano /etc/modprobe.d/alsa.conf
> Then add the following at the end of the file:
> options snd-hda-intel model=generic
> Then hit ctrl+x, hit Y to save, hit enter, then reboot.
> For me, this disables the headset mic, but enables the internal mic.

the sudo nano /etc/modprobe.d/alsa.conf 
with 
options snd-hda-intel model=generic did not work for me (Lenovo Ideapad Yoga 13)

sebanc, 2020-05-08:
Hi, I applied audio fixes on the new release, not sure which devices were actually affected by those issues though but don't hesitate to try it.

nicktelindert, 2020-05-08:
@sebanc I'll try it out

frigis1, 2020-05-09:
@sebanc Thank you, but issue still persists on my laptop.

sebanc, 2020-05-09:
@frigis1 Thanks for testing, i will keep looking into it

tungmeister, 2020-05-17:
I'm also getting this issue on my Dell E5840, running `amixer -c0 set "Internal Mic" toggle` does resolve the problem though

sebanc, 2020-05-18:
@tungmeister you have to create the /var/brunch/bootscripts directory first

tungmeister, 2020-05-18:
@sebanc I wasn't certain due to the lack of existing brunch directory, I've created the brunch and bootscripts directories however it doesn't work.

sebanc, 2020-05-18:
Try the following commands:
```
sudo mkdir -p /var/brunch/bootscripts
echo 'amixer -c0 set "Internal Mic" toggle' | sudo tee /var/brunch/bootscripts/alsa.sh
sudo chmod 0755 /var/brunch/bootscripts/alsa.sh
```

kbnhvn, 2020-05-24:
Hi, same issue for microphone on Dell venue 11 7140. Alsamixer shows a Broadwell rt286 card

sebanc, 2020-06-02:
For those of you who updated to r83, have you tried the flag "chrome://flags/system-tray-mic-gain" and if so does it help with this issue ?

kbnhvn, 2020-06-02:
Still no mic on Dell venue 7140, and I lost sound on r83

frigis1, 2020-06-06:
I have updated to r83 and the latest Brunch framework, and enabled the flag. I only get white noise.

sebanc, 2020-06-06:
@frigis1 Could you try to update to the latest build in the "brunch-testing" repo:
https://github.com/sebanc/brunch-testing/releases

frigis1, 2020-06-06:
Updated to latest testing build, still the same result.

sebanc, 2020-06-06:
Thanks for the feedback on both tests

kowalski7cc, 2020-06-21:
Same thing happens on XPS 13 9350:
Card: HDA Intel PCH
Chip: Realtek ALC3246
lspci:
1c.5-[3b]----00.0  Realtek Semiconductor Co., Ltd. Device [10ec:525a]

sebanc, 2020-07-06:
For those who use, 'amixer -c0 set "Internal Mic" toggle' to activate internal mic, could you verify if your mic works without this command if you create a file /etc/init/alsafix.conf (sudo nano /etc/init/alsafix.conf) with the below content ?
```
start on stopped udev-trigger

script
        if [ $(amixer -q -c 0 get "Internal Mic") -eq 0 ]; then amixer -q -c 0 set "Internal Mic" unmute; fi
end script
```

frigis1, 2020-07-07:
Internal mic does not switch on for me if I use this file.

sebanc, 2020-07-08:
@frigis1 Thanks! Could you try the below one ?

```
start on stopped udev-trigger and started ui

script
    if [ $(amixer -q -c 0 get "Internal Mic") -eq 0 ]; then amixer -q -c 0 set "Internal Mic" toggle; fi
end script
```

Also if your run "udevadm info" in crosh shell and then plug your headphones, do you see related lines appear in crosh shell ?

frigis1, 2020-07-09:
That script does not work for me either. I ran udevadm info and the shell listed some options: help, query, path, etc. Didn't seem to provide any information.

Shamw0w, 2020-07-15:
Hi all, on Dell Inspiron 3000. Internal mic does not work, but headset with mic works when plugged in 3.5mm jack.

I've read the above and tried the following and these are the results I get:

chronos@localhost / $ amixer
Simple mixer control 'Master',0
  Capabilities: pvolume pvolume-joined pswitch pswitch-joined
  Playback channels: Mono
  Limits: Playback 0 - 100
  Mono: Playback 100 [100%] [on]

chronos@localhost / $ amixer -c0 set "Capture" toggle
Simple mixer control 'Capture',0
  Capabilities: cvolume cswitch
  Capture channels: Front Left - Front Right
  Limits: Capture 0 - 63
  Front Left: Capture 63 [100%] [30.00dB] [on]
  Front Right: Capture 63 [100%] [30.00dB] [on]

Even with toggling the mic on and setting capture to 100% I don't seem to capture any input. 

When I run AlsaMixer it only shows 1 master channel (at 100).

Any tips / advice? Have I missed something I need to try in the thread?

Shamw0w, 2020-07-15:
Also, FWIW, I just did a reboot and tested the mic (without saving a boot script), and the mic worked without needing to toggle it on.

hacker1024, 2020-08-26:
> @frigis1 Could you try to update to the latest build in the "brunch-testing" repo:
> https://github.com/sebanc/brunch-testing/releases

Enabling the new option added fixes it for me. Thanks for your great work.

Adkatka, 2020-09-13:
Inspiron 3443 microphone not working ALC3234. 
amixer -c0 set "Internal Mic" toggle wont work
Output: Unable to find Simple Control "Internal Mic"

omminp, 2020-09-20:
Internal mic not working, but when pluged in headset mic works. Realtek ALC3204 

Fix by crosh shell 
sudo alsactl restore

To make permanent after reboot, 
create /var/brunch/bootscripts as described earlier in this thread, 
create alsa.sh with the line
sudo alsactl.restore
and make alsa.sh executable.

Works for me, so maybe worth a try.

dkannegi, 2020-09-21:
I am running Chrome 84 Rammus on Brunch r85 k4.19 testing 20200905 on the following laptops: Panasonic Toughbook CF-C2 MK1 and a MK2.  They load up without the mic control showing at all in the UI, my fix was to steal alsa-base.conf from my Linux Mint 20 x64 boot usb and plopped it into Chrome's install at /etc/modprobe.d/alsa.conf (original renamed as a .bak).  One reboot later and I got my mic working as internal mic.  Of note, on the CF-C2s and maybe most other toughbooks, the headphone port is software switchable, however the mic port only shows up as 'internal mic' and an external mic is switched in-line by the jack circuit.  My Amixer -q c0 and c1 output is attached (this computer has two, c1 is my main onboard sound).  
[amixer_c1_panasonic_CFC2MK2.txt](https://github.com/sebanc/brunch/files/5258214/amixer_c1_panasonic_CFC2MK2.txt)
[amixer_c0_panasonic_CFC2MK2.txt](https://github.com/sebanc/brunch/files/5258215/amixer_c0_panasonic_CFC2MK2.txt)
[dmesg.txt](https://github.com/sebanc/brunch/files/5258216/dmesg.txt)
[alsa.conf.txt](https://github.com/sebanc/brunch/files/5258217/alsa.conf.txt)

sebanc, 2020-09-22:
@dkannegi Thanks for the report, that's a very interesting find :) Now I would need to understand exactly what command is causing this.

I don't really see what in the linux mint alsa-base.conf file could be related to this, however it is replacing the chromeos alsa.conf which contains "options snd cards_limit=1" by default.

Could you rebuild the rootfs and try to only remove "options snd cards_limit=1" from chromeos /etc/modprobe.d/alsa.conf ?

dkannegi, 2020-09-22:
I upgraded using the chromeos-update command after the first post using ChromeOS 85 and the latest brunch (Brunch r85 k4.19 testing 20200919) which reset the asla.conf and the mic remained available in the UI upon reboot.  I still went into the alsa.conf and #remarked the line in question. Not sure if an upgrade is different from doing a full wipe of the drive.

sebanc, 2020-09-22:
An upgrade just resets the rootfs partition, several things could have happened here:
- the mic settings remained available because the setting was stored in chromeos partition
- chromeos r85 fixed it.

To directly implement a fix in brunch, I would need to know exactly what modification is necessary to make mic work on a clean install. (Without too many modifications, copying over the full Linux Mint alsa conf over the chromeos would probably have

dkannegi, 2020-09-24:
I did a test with a fresh USB install of CrOS 84 and brunch r85 k4.19 testing 20200905 with the mic not showing up, and tested commenting out "options snd cards_limit=1".  End result is that the mic still didn't show up.  I will attempt running a chromeos-update command in the USB to bring both CrOS to 85 and brunch to the latest test version and see if it catches it.  Of note,  it turns out this laptop has a Realtek ALC269 (PCIID 10EC:0269) as found out when I put Windows 10 on it, Windows 10 picked it up correctly, but I had another laptop with the same chip not correctly enumerate under Windows so I had basic audio out and no input/mic until I manually  installed the Realtek driver, which then allowed everything to work. Realtek DACs are very weird as they can act as a passthru if there is another sound controller onboard (e.g. nVidia nForce chipsets) OR they can be the sound controller also.

sebanc, 2020-09-25:
@dkannegi Thanks for trying it :) The issue must then be related to something else, I will try to see what other parameter could cause that

omminp, 2020-09-27:
@Adkatka I have the same result on my Dell 3185.  
However, I find that 
"amixer -c1 set "Internal Mic" toggle"  works without the error, and then 
"amixer -c1 set "Capture" 40%   also works.  And the mic function is restored.  
Perhaps the device 0 does not exist for this machine.

parinay602, 2020-10-14:
guys nothing is working....PLease help

CreaticaONG, 2020-11-16:
Hello,
First of all sorry for my bad English, I didn't practice for a long time...
I have the same issue on my Dell Latitude E5450. I got it working following these steps:

1. My Dell laptop has two sound cards: HDMI and PCH. You can verify your sound cards typing in a shell `aplay -l`
2. In my case HDMI sound card has the 0 id and the PCH has id 1. CRAS (Chrome OS Audio Server) use default id 0 sound card (in my case HDMI that don't have capture function) and that's why the commands above not work for me. To change default sound card I modified  `/etc/modprobe.b/alsa.conf` adding:
`
# Set this to the correct number of cards.
options snd cards_limit=2

options snd-hda-intel id=PCH index=0
options snd-hda-intel id=HDMI index=1
`
3. Save file and reboot.
4. Check the order with `aplay -l`
5.  Perform the steps described above: 
`sudo mkdir /var/brunch/bootscripts
echo 'amixer -c0 set "Internal Mic" toggle' | sudo tee /var/brunch/bootscripts/sound.sh
echo 'amixer -c0 set "Capture" 40%' | sudo tee -a /var/brunch/bootscripts/sound.sh
sudo chmod 0755 /var/brunch/bootscripts/sound.sh
`
Thanks @sebanc for the brunch project!

omminp, 2020-11-17:
@sebanc 
Folllowing on from the previous comment, I can confirm that in my case also  
aplay -l         **gives**

**** List of PLAYBACK Hardware Devices ****
card 0: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: Generic [HD-Audio Generic], device 0: ALC3204 Analog [ALC3204 Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

**AND**
chronos@localhost ~/Downloads $ amixer -c 0 set "Internal Mic" toggle
amixer: Unable to find simple control 'Internal Mic',0

**AND**
chronos@localhost ~/Downloads $ amixer -c 1 set "Internal Mic" toggle
Simple mixer control 'Internal Mic',0
  Capabilities: cswitch cswitch-joined cswitch-exclusive
  Capture exclusive group: 0
  Capture channels: Mono
  Mono: Capture [on]

With -c 0 the 'set' command returns a null , with the error message.
Perhaps the current internal_mic_fix script could include both cases by checking for a null return from 
amixer -c 0 set "Internal Mic" toggle

and in that case execute
amixer -c 1 set "Internal Mic" toggle

It will mean more users will be fixed.  It seems to be an ongoing issue for users.

sebanc, 2020-11-17:
@CreaticaONG @omminp Thanks for your investigations and feedback, it is really helpful :)

@omminp I am thinking that forcing the forcing the HDMI soundcard as 2nd might be the proper way to move on with this.
If you set only `options snd-hda-intel id=HDMI index=1` in /etc/modprobe.b/alsa.conf, does it change the soundcard order on your device ?

omminp, 2020-11-17:
As requested, added 1 option to alsa.conf

*# Set this to the correct number of cards.options snd cards_limit=1options
snd-hda-intel id=HDMI index=1*

The line snd cards_limit=1 was already in the file, and hasn't been changed.

After reboot, the soundcard order is reversed, as here:

chronos@localhost / $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: Generic [HD-Audio Generic], device 0: ALC3204 Analog [ALC3204
Analog]
  Subdevices: 1/1
  Subdevice #0: subdevice #0
card 1: HDMI [HDA ATI HDMI], device 3: HDMI 0 [HDMI 0]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

I hope this helps you work out the best way to proceed.  Let me know if I
can do any further testing for you.

[image: Mailtrack]
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
Sender
notified by
Mailtrack
<https://mailtrack.io?utm_source=gmail&utm_medium=signature&utm_campaign=signaturevirality5&>
17/11/20,
06:11:49 pm

sebanc, 2020-11-17:
@omminp Thanks for the feedback :) I just have a last doubt, could you confirm that with "options=internal_mic_fix" added to the grub config and "options snd-hda-intel id=HDMI index=1" added to /etc/modprobe.b/alsa.conf your microphone works correctly ?

(if the rootfs is rebuilt when you add "options=internal_mic_fix", you will have to add "options snd-hda-intel id=HDMI index=1" to /etc/modprobe.b/alsa.conf again)

omminp, 2020-11-18:
Well it took a while to realize that I had a file /var/brunch/bootstrap/alsa.sh which I hadn't removed and which enabled the mic...

After deleting that, here are the test results:

[ mic tests.pdf](https://github.com/sebanc/brunch/files/5563336/mic.tests.pdf)

Not sure what the doubt was, but I can confirm that mic works only with both options added.  Tested on 2 installations on the same machine.

[21 more replies]