sebanc/brunch #58: No Audio/Video Playback - Asus Chromebook C300 Quawks

Stoobacca, 2020-04-20:
Hi,

No audio as far as I can tell and video playback is not working. Videos can buffer and stills/previews load up, but video/audio won't play. Tested in YouTube web and app, also Soundcloud to check it wasn't a YouTube error.

OS smooth in Rammus, Play Store and all other components seem to run just fine.

C300 is BayTrail, I believe using max98090 audio. Booting USB using MrChromeBox UEFI. Tested GalliumOS and audio works there.

I've tried several recoveries: Rammus, Eve, Atlas, Terra (C300SA, updated version of mine) and Caroline. Also tested with original Quawks image I came from and same issue.

Thanks

sebanc, 2020-04-20:
Hi,

It seems like your audio firmware is not installed. Could you attach a dmesg to this issue ?

sebanc, 2020-04-20:
To get the dmesg, go to crosh shell and run "dmesg > ~/Downloads/dmesg.txt"

Attach dmesg.txt from your Downloads folder to this issue.

Stoobacca, 2020-04-20:
Hi.

Appreciate the quick response! I'll give it a go and load it up later on. 

Cheers

Stoobacca, 2020-04-20:
dmesg.txt attached to first post.

Thanks

sebanc, 2020-04-20:
Thanks for the dmesg, everything seems fine in it as far as i can see.

Could you run "sudo edit-grub-config" and add "options=disable_intel_hda" after cros_debug ?

Stoobacca, 2020-04-20:
No luck, still the same. I did notice now when rebooting at 'rebuilding ChromeOS rootfs' higher up there's a line 'ALSA device list: no soundcards found'
[image]

sebanc, 2020-04-21:
Hi,

The 'ALSA device list: no soundcards found' message is normal as the driver is built as a module and not yet loaded

Could you run the following commands:

sudo alsaucm -c chtmax98090 set _verb HiFi set _enadev Headphone
sudo alsaucm -c chtmax98090 set _verb HiFi set _enadev Speakers
sudo alsaucm -c chtmax98090 set _verb HiFi set _enadev HeadsetMic
sudo alsaucm -c chtmax98090 set _verb HiFi set _enadev InternalMic
sudo alsactl store

Reboot and see if sound works.

Stoobacca, 2020-04-21:
OK Cool,

I can't run the commands, I get some errors. 
[image]

Tried without 'options=disable_intel_hda' too just in case but I get the same error.

Stoobacca, 2020-04-21:
Not sure if it's useful at all but contents of /usr/share/alsa/ucm:
[image]

sebanc, 2020-04-22:
Hi,

Thanks for all the logs, I am trying to build a test image with changes included in the alsa ucm config but I am having problems with the kernel which are unrelated to this issue.
I will tell you once the test image is ready.

Stoobacca, 2020-04-22:
Update (still no luck)

I had a look at the errors and see the files don't exist, so I'm attempting to put the files where they are expected to be...

- I created the folder 'GOOGLE-Quawks-1.0-Quawks' under /user/share/alsa/ucm/
- Then copied chtmax98090.conf and HiFi.conf from /usr/share/alsa/ucm/chtmax98090/ to the new folder
- I renamed the chtmax98090.conf  to 'GOOGLE-Quawks-1.0-Quawks.conf'
- I then created the directory 'alsa' under /var/lib/ as this folder didn't exist
- I created the file 'asound.state' and used nano to paste the contents of the GalliumOS Baytrail state file (https://github.com/GalliumOS/galliumos-baytrail/blob/master/var/lib/alsa/asound.state)

This allowed your original commands to work but no change in sound. The GOOGLE-Quawks-1.0-Quawks directory was removed on a reboot but the asound.state file is still there.

Am I on the right track at all!? I will continue browsing for fixes but I'm not sure I know enough to get it working!

Stoobacca, 2020-04-22:
> Hi,
> 
> Thanks for all the logs, I am trying to build a test image with changes included in the alsa ucm config but I am having problems with the kernel which are unrelated to this issue.
> I will tell you once the test image is ready.

Oh cool thanks, really appreciate your help!

sebanc, 2020-04-22:
> Update (still no luck)
>
> I had a look at the errors and see the files don't exist, so I'm attempting to put the files where they are expected to be...
>
> * I created the folder [...]

My idea is the same as yours, copy the ucm files as for some reason chromeos expects them somewhere else.
However, if I am not mistaken the right repo for your device is "galliumos-braswell" which does not seem to need the asound.state file.

Stoobacca, 2020-04-22:
Ah OK!

So maybe I've copied the wrong config? I'll see if I can do anything about it but I'm not sure I'll manage to get it working!

Look forward to trying your fix whenever it may be available.

Thanks again

Stoobacca, 2020-04-22:
I was basing the BayTrail files on the Platform given on the Chromium dev info page. Does it not strictly relate to the platform of the device? I've tried the Braswell too but not working for me either. I'm sure I'm probably doing something wrong though!

https://www.chromium.org/chromium-os/developer-information-for-chrome-os-devices
[image]

sebanc, 2020-04-22:
My mistake, I was looking at "ASUS Chromebook C300SA"... I am going to look at Baytril repo more closely

Stoobacca, 2020-04-22:
> My mistake, I was looking at "ASUS Chromebook C300SA"... I am going to look at Baytril repo more closely

No problem! Thought I'd post just in case. Thanks

sebanc, 2020-04-22:
> Update (still no luck)
>
> I had a look at the errors and see the files don't exist, so I'm attempting to put the files where they are expected to be...
>
> * I created the folder [...]

Well, as far as I can say you are really on the right track, those steps should in theory have it working. Just make sure, you have correct permissions for the new alsa ucm folders/files.

sebanc, 2020-04-22:
The folder that got deleted is probably because the framework was rebuild which usually happens only when a chromeos or brunch update is applied.

Stoobacca, 2020-04-22:
I've run ls -l and get the following:

GOOGLE-Quawks-1.0-Quawks directory:
drwxr-xr-x. 2 root root 4096 Apr 22 10:29 GOOGLE-Quawks-1.0-Quawks

Files within directory:
-rw-r--r--. 1 root root  100 Apr 22 10:26 GOOGLE-Quawks-1.0-Quawks.conf
-rw-r--r--. 1 root root 4918 Apr 22 11:10 HiFi.conf

alsa directory:
drwxr-xr-x. 2 root                root                4096 Apr 22 10:35 alsa

asound.state:
-rw-r--r--. 1 root root 86405 Apr 22 10:35 asound.state

The file permissions look the same as the other directories/files but I'm not I'm not sure about setting permissions in shell, could you let me know if these look right or the command if the above needs modifying?

Cheers

sebanc, 2020-04-22:
Indeed, that's perfect, I don't understand why sound does not work...

sebanc, 2020-04-22:
If you have a /etc/cras directory; could you try to remove it and reboot ?

Stoobacca, 2020-04-22:
No cras file, although alsamixer does default to "cras".

When I select sound car I get the following options:

- default
0 HDA Intel PCH
1 chtmax98090
enter device name...

sebanc, 2020-04-22:
i think you run your tests with "options=disable_intel_hda"

Stoobacca, 2020-04-22:
Sadly no luck. Added that back in, rebooted and let Brunch rebuild.

Then run the following

sudo mkdir /usr/share/alsa/ucm/GOOGLE-Quawks-1.0-Quawks/
sudo cp /usr/share/alsa/ucm/chtmax98090/chtmax98090.conf /usr/share/alsa/ucm/GOOGLE-Quawks-1.0-Quawks/
sudo cp /usr/share/alsa/ucm/chtmax98090/HiFi.conf /usr/share/alsa/ucm/GOOGLE-Quawks-1.0-Quawks/
cd /usr/share/alsa/ucm/GOOGLE-Quawks-1.0-Quawks/
sudo mv chtmax98090.conf GOOGLE-Quawks-1.0-Quawks.conf

Also tried the alsaucm commands and another reboot but still no.

Alsamixer now has no HDA Intel, but defaults to CRAS and no sound

Stoobacca, 2020-04-22:
I also tried modifying asound.conf to the below (to make it default to the chtmax98090), which does change the default in alsa but no sound.. my guess is that we need to use CRAS though?

#Route all audio through the CRAS plugin.
pcm.!default {
#       type cras
        type hw
        card 0
}
ctl.!default {
#       type cras
        type hw
        card 0
}

sebanc, 2020-04-22:
I think ChromeOS sees that your device is a chromebook and applies specific rules for it.
I will try to disable CONFIG_GOOGLE_FIRMWARE in the next testing release to see if maybe it would allow your sound card to be recognized as in a standard kinux distro.

Stoobacca, 2020-04-22:
Cool, I'll give it a go once it's ready. Appreciate the help!

Stoobacca, 2020-04-23:
It's fixed! (maybe fully!!). I've played around a lot so I'm not sure if I changed anything else BUT I think it is only the asound.state file causing the problem, and then a daft error on my part with the initial fix attempt! I've checked and the new .conf files I made previously are not there, so that step appears to be unnecessary, it is just the asound.state (again.. I think). 

I've reinstalled to start fresh so had to recreate the asound directory and file, I thought I'd document the commands below, as it can be annoying when someones says "I've solved it" and then doesn't post the actual solution.. correct me if any of the commands below look wrong xD

Also worth a note to anyone who may have the same error.. After making the below changes there was no audio but videos and audio websites would actually play (just silently), but then I went to double check mid post and it's all magically working. Can't say if the fixes will stick or comment on stability yet, but it's working.

sudo mkdir /var/lib/alsa/
cd /var/lib/alsa/
sudo touch asound.state
sudo nano asound.state
copy contents from 'working' statefile.. save (ctrl X > Y > Enter)

I noticed the statefile says 'state.byt-max98090', so I've corrected to 'state.chtmax98090'
reinitialised with sudo alsactl restore

The restore gives errors but not posted as sound is working!

Stoobacca, 2020-04-23:
OK so doesn't persist on reboot, running sudo alsactl restore makes it start again, but still silent.

I'm not sure what makes the sound burst into working, but I'm getting close!

sebanc, 2020-04-23:
Congrats on your findings and being very close to solving this issue :)

I am thinking the remaining issue might be related to the audio card autosuspend not working properly, what you can try is to find your soundcard in attributes in /sys (probably something like `sudo find /sys/devices | grep max98090`), locating the power/control file and doing `echo "on" | sudo tee /sys/devices/...../power/control`

sebanc, 2020-04-23:
Forgot to say, it won't survive a reboot but there are ways to automate it

Stoobacca, 2020-04-23:
> Forgot to say, it won't survive a reboot but there are ways to automate it

No worries, well I'm happy having to run it if I need to. I've run your command so I'll see if that at least makes it simpler!

Also noticed the headphone jack doesn't work (I imagine neither does HDMI audio). Headphones get identified by ChromeOS but sound doesn't switch to the Jack.

I might fiddle with it but for now I'm happy it works mostly!

sebanc, 2020-05-08:
Hi,

I have found the fix for the audio card to be correcly recognized as "chtmax98090" and included the configuration files.

Could you try the new release with "options=baytrail_chromebook" added to the kernel command line ? (audio should work out of the box)

Not sure if the headphones, mic issues are still present though

Stoobacca, 2020-05-14:
Hi,

Thanks for the update.

I'm downloading the r81 branch now and will give it a go.

I'll let you know how I get on.

Cheers

Stoobacca, 2020-05-14:
So I created a new Rammus USB with the latest r81 Brunch release. I booted the USB and added options=baytrail_chromebook to the grub launcher. Still no audio.

Should the command work from USB boot?

Stoobacca, 2020-05-14:
> So I created a new Rammus USB with the latest r81 Brunch release. I booted the USB and added options=baytrail_chromebook to the grub launcher. Still no audio.
> 
> Should the command work from USB boot?

Scrap that. Seemed to work on second try. It's laggy on my USB so I'm installing and will test from proper install

Stoobacca, 2020-05-14:
Sound is working! Thanks so much for sorting it out!

Just a note.. Headphones work but the jack doesn't mute the Chromebook speakers, but I can get around it by turning Chromebook volume fully down and then plugging in headphones.

Thanks for your work on my problem and on the project generally!

Much appreciated.

Stoobacca, 2020-05-15:
So I've looked into the headphone issue.

I can see that the headphone jack is being muted correctly when plugged and unplugged.

However the speakers are not being muted correctly.

I've checked the /chtmax98090/HiFi.conf and it looks fine as far as I can tell. I've also tried including the command to mute the speakers in the headphone enable and disable sequences but no luck.

Any idea how I might check this? I'll keep researching otherwise!

Cheers

sebanc, 2020-05-15:
Sorry I am not very experienced with debugging audio, I was just able to revert a modification in the chromeos kernel that was causing the main issue but i don't think i will be able to help much more. The driver that's used in the 4.19 linux kernel is also not the same as the original chromeos one so I am not even sure it could work.

Maybe someone on the linux/galliumos forums has solved this issue or might be able to help.

Stoobacca, 2020-05-15:
No worries, thanks anyway!

I'm looking at Gallium at the moment to see if I can find a similar issue. I'm fairly certain I found something on this error when I was looking up my first problem so hopefully I can find it again!

Thanks

djamesrock, 2020-05-15:
Can something in this bug fix help?
https://bugzilla.kernel.org/show_bug.cgi?id=115531#c122

sebanc, 2020-06-02:
I don't think I can help further on this issue so I will close it, don't hesitate to re-open it if needed.

closed 2020-06-02 by sebanc

mohamedh648, 2022-03-06:
I got the same problem running samus 91 how do I run that command : options:baytrail_chromebook or is that a modified version of chrome os Im a newbie so I don't know alot I would appreciate it if somebody help me

Stoobacca, 2023-05-23:
> I got the same problem running samus 91 how do I run that command : options:baytrail_chromebook or is that a modified version of chrome os Im a newbie so I don't know alot I would appreciate it if somebody help me

So it's been well over a year since this message.. but if you're still trying this or if anyone happens to be searching for baytrail chromebooks or chtmax max98090..

You want to run the terminal with Ctrl + Alt + T, then type "shell" and hit enter. Once in the shell, type "sudo edit-grub-config" and hit enter. In the grub config, add "baytrail_chromebook" after where it says options= and then exit with Ctrl + X

Reboot and hopefully you'll have audio. I haven't got it working on newer versions with the chromebook audio flag

chuc2rk, 2023-06-28:
> baytrail_chromebook

same no sound on my HP 845 Ryzen 7. I try Chrome OS flex, my laptop have sound