sebanc/brunch #60: Black screen on first boot

apomalyn, 2020-04-22:
Hi! 
First of all thank you for this work!
I'm trying to install chromeOS 80 (brunch: 80 stable, recovery image: rammus 80) in dual boot but after installing it from Linux when I selection the option on the grub menu I got a black screen even when I set the log level to 0.

Any idea ?

CPU: i7
Secure boot: disabled

sebanc, 2020-04-22:
Hi,

Setting the log level to 0 might actually result in black screen, 7 is the verbose level.

Which linux distro are you using ? Also, could you post your grub config file ?

Thanks

apomalyn, 2020-04-22:
Hi,
Thanks for answering. Even when I reset the verbose level to 7 nothing change.
My ubuntu distro is Ubuntu 19.4, and my grub config file:
```bash
#!/bin/sh
exec tail -n +3 $0
# This file provides an easy way to add custom menu entries.  Simply type the
# menu entries you want to add after this comment.  Be careful not to change
# the 'exec tail' line above.

menuentry "ChromeOS (boot from disk image)" {
	img_part=/dev/sda6
	img_path=/chromeos.img
	search --no-floppy --set=root --file $img_path
	loopback loop $img_path
	linux (loop,gpt7)/kernel boot=local noresume noswap loglevel=1 disablevmx=off \
		cros_secure cros_debug loop.max_part=16 img_part=$img_part img_path=$img_path
	initrd (loop,gpt7)/initramfs.img
}```

apomalyn, 2020-04-22:
Ok, from the grub command line, if I do the `ls` command I don't see any gpt7 but:
(hd0) (hd1) (hd1/gpt6) (hd1/gpt5) (hd1/gpt4) (hd1/gpt3) (hd1/gpt2) (hd1/gpt1)

Don't know if that help

sebanc, 2020-04-22:
Well, I have been looking at the config and it looks fine, the partition gpt7 is within the chromeos.img image that is mounted as loop device so it is normal that you do not see if before the line `loopback loop $img_path` is run.

All I can think of is still a grub issue, I see that you have replaced the ubuntu grubx64.efi but are you sure it was the right one, sometime there are several on the efi partition ?

apomalyn, 2020-04-22:
I changed the one in /boot/grub/x86_64-efi but I just found there is two other in lib/grub one signed and the not signed I will try to changes the unsigned one

apomalyn, 2020-04-22:
Even after changing the grubx64.efi nothing changed. 
By the way it's like grub don't load anything. I change the background color of grub and after choosing the chrome os option the background don't even change so I guess grub don't load anything

sebanc, 2020-04-22:
The grubx64.efi file you have to change is in "/boot/efi/EFI/ubuntu/grubx64.efi", you need to be root (sudo su) to access it.

apomalyn, 2020-04-22:
Ok I've got something thank you! 
But I stay stuck on the Chrome logo 😅

sebanc, 2020-04-22:
the attempts to boot with broken grub must have messed up the image, try to rebuild one last time

apomalyn, 2020-04-22:
I retry a few more time and each time I finish stuck with everything freeze on the connect to wifi screen.

Each time I use my keyboard (or a external mouse) the OS freeze...

sebanc, 2020-04-22:
Have you tried it on usb first or are you trying directly in dual boot ?

sebanc, 2020-04-22:
Also, what's your laptop model ?

apomalyn, 2020-04-22:
Directly on dual boot 😅 maybe I should have tried on usb stick first 😂

I have a MSI MS-16J9
CPU: i7-7700HQ
Graphics chipset (used with ubuntu) Intel HD graphics 630
GPU: Nvidia GTX 1050

sebanc, 2020-04-22:
No worries, it was just to know, so it's probably another problem not related to dual boot.

Could you try to add "module_blacklist=nouveau" to the kernel command line in the grub config (after cros_debug) ?

apomalyn, 2020-04-22:
Seems like it's worked!! 🎉🎉🎉

closed 2020-04-22 by apomalyn

sebanc, 2020-04-22:
Nice :) Happy that it works

apomalyn, 2020-04-22:
All thanks to you!

ghritik, 2020-05-05:
i am getting error .it says chromos not found .. at last line the error is need to load kernel first... how to solve this plz hlp