sebanc/brunch #61: Anyway to automatically restore mounted drives?

jhedfors, 2020-04-23:
I am really enjoying this project 👍 

Is there a way to automatically restore a mounted drive? Currently, when I update Chrome OS, I have to go through the process (creating directories & mounting them wiith fstab) all over again to mount my "storage" partition to a folder named "Storage" in My Files.  I was hoping there would be a Files app "service" to mount this partition that could be automatically restored with the update.

Any suggestion on how I can more easily accomplish this?

sebanc, 2020-04-23:
Hi,

If I understand correctly what you want to do, I would create a script (ending in .sh and with execute permission) in "/var/brunch/bootscripts" containing something like:
`mount /dev/xxxxxxx /home/user/xxxxxxx/storage`

Bootscipts can survive both ChromeOS and brunch framework updates.

jhedfors, 2020-04-23:
Awesome. Thanks for the suggestion, and for your hard work!

jhedfors, 2020-04-23:
I have my executable script in place  (/var/brunch/bootscripts/mymounts.sh), with the following:

mount /dev/sda4 /home/chronos/u-cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage;
mount /dev/sda4 /mnt/storage;

I have verified the correct path to my folder in MyFiles, but only the second mount (/mnt/storage) works. This mount does work in fstab. Does this script run too early or late compared to fstab for it to work for me? Any ideas? Thanks again!

jhedfors, 2020-04-23:
I just realized the path that you suggested was under /home/user. I tried that, but it did not work for me either...
mount /dev/sda4 /home/user/cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage;

sebanc, 2020-04-23:
Well, the script seems ok. Bootscripts triggered after module loading which might be too early to access user directories.

Could you try to edit /etc/init/bootscripts.conf and replace:
`start on stopped udev-trigger`
by
`start on stopped login`

And see if your script works

jhedfors, 2020-04-23:
That didn't work for me, unfortunately...

start on stopped login

script
        if [ $(ls -1q /var/brunch/bootscripts/*.sh 2>/dev/null | wc -l) -gt 0 ]; then
                for patch in /var/brunch/bootscripts/*.sh
                do
                        /bin/bash "$patch"
                done
        fi
end script

sebanc, 2020-04-23:
Another attempt, try replacing:
`start on stopped login`

with the 2 lines below:
```
start on start-user-session
task
```

jhedfors, 2020-04-23:
That didn't work either... this is what bootscripts.conf file looks like now:

tart on start-user-session
task

script
        if [ $(ls -1q /var/brunch/bootscripts/*.sh 2>/dev/null | wc -l) -gt 0 ]; then
                for patch in /var/brunch/bootscripts/*.sh
                do
                        /bin/bash "$patch"
                done
        fi
end script

sebanc, 2020-04-23:
I have another idea, could you post your /etc/fstab ?

jhedfors, 2020-04-23:
Hi had to rename it with .txt extention. Also, I had deleted it when testing your solution above. Here is a copy I had saved.

[fstab.txt](https://github.com/sebanc/brunch/files/4524869/fstab.txt)

closed 2020-04-23 by jhedfors

sebanc, 2020-04-24:
Could you try to replace the content of your /var/brunch/bootscripts/mymounts.sh with:
```
cat >/etc/fstab <<FSTAB
/dev/sda4       /mnt/storage                          auto    defaults   0      0
/mnt/storage    /home/chronos/u-cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage                     none    bind       0      0
FSTAB
```

jhedfors, 2020-04-24:
I assume that was to create the fstab? It didn't work. This is how the file looks:

#!/bin/bash
cat >/etc/fstab <<FSTAB
/dev/sda4       /mnt/storage                          auto    defaults   0      0
/mnt/storage    /home/chronos/u-cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage                     none    bind>
FSTAB

sebanc, 2020-04-24:
Sorry I never use fstab in chromeos, try to replace /etc/fstab in `cat >/etc/fstab <<FSTAB` to the right location

sebanc, 2020-04-24:
Also there is a weird ">" on the previous to last line but i guess it is a copy paste issue

jhedfors, 2020-04-24:
It did create the fstab, but was not mounted in My Files.  We had changed the bootscripts.conf earlier -  could that be affecting this?

#!/bin/bash
cat >/usr/local/fstab <<FSTAB
/dev/sda4       /mnt/storage                          auto    defaults   0      0
/mnt/storage    /home/chronos/u-cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage     none    bind       0      0
FSTAB

sebanc, 2020-04-24:
you mean bootscripts.conf ?

jhedfors, 2020-04-24:
Ah... that is right. I got that from here:
https://github.com/MuntashirAkon/Chrome-OS-Multiboot/blob/master/README.MD

That is how I have been mounting my drives... perhaps you have a different suggested method? Sorry I did not remember getting that from somewhere else.

sebanc, 2020-04-25:
Could you try replacing the content of "/var/brunch/bootscripts/mymounts.sh" with the below lines ?
```
cat >/etc/init/mount-internals.conf <<MOUNTINTERNALS
start on start-user-session
task

script
mkdir -p /home/chronos/u-cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage
mount /dev/sda4 /home/chronos/u-cc29900fb3b9d770f0d85db4eb4dfe60bb2bf9b2/MyFiles/Storage
mkdir -p /mnt/storage
mount /dev/sda4 /mnt/storage
end script
MOUNTINTERNALS
```

jhedfors, 2020-04-25:
I powerwashed and tried the script above (with the updated ID), but did not mount or create the Storage mount point in MyFiles. I will keep experimenting...

cat >/etc/init/mount-internals.conf <<MOUNTINTERNALS
start on start-user-session
task script
mkdir -p /home/chronos/u-e3e2dcb85c22a9e4d35b37bf041e1a2ead0a6949/MyFiles/Storage
mount /dev/sda4 /home/chronos/e3e2dcb85c22a9e4d35b37bf041e1a2ead0a6949/MyFiles/Storage
mkdir -p /mnt/storage
mount /dev/sda4 /mnt/storage
end script

jhedfors, 2020-04-27:
I have as system that is working pretty well for me, so I am good. I copy the fstab and mount-internals.conf from my GoogleDrive with the following:

sudo mount -o rw,remount / && sudo cp /media/fuse/drivefs*/root/CROS/mount-internals.conf /etc/init/ && sudo cp /media/fuse/drivefs*/root/CROS/fstab /usr/local/

One thing that does not seem to work is accessing mounted drives from the Android or Crostini containers. I am assuming there is nothing that can be done about that, as I am sure that is intentional. :)  Thanks for your help on this!