sebanc/brunch #62: Device Compatibility Report: Success with Microsoft Surface Pro 3 64GB

SiemdeNijs, 2020-04-23:
Instead of a issue I'd like to make a good-news report!

I've spend the last weeks tempering with images and trying (what felt like) every single chromium based OS in combination with my Surface Pro 3 device, absolutely nothing has been as stable & pleasing to use as a image compiled with brunch ! I want to thank everyone who is working on this project or in any way contributing to it.

Here's what I've got so far & small tweaks which made huge differences:
Currently writing from a stable  ChromeOS 81 install with the latest 81 brunch. Compiled image with brunch 80 latest & rammus 80 recovery image and enabled_updates kernel flags to get on v81. 

Thanks to the included surface tweaks in this project almost every single sensor/feature seems to work:
Microphone: Working
Cameras: Working
Typecover and touchpad: Working
Auto-rotation: Working
Touchscreen: Working
Volume keys: Working
Powerbutton: Hold for power-menu working
Bluetooth: Working
Wi-Fi: Surprisingly stable whereas other builds have problems with turning on/off wi-fi + when going into sleep.
Battery: I've installed, updated, played around & have now been using the Surface Pro 3 with v80/81 for over 5 hours. Beats Windows/Ubuntu's battery life on this particulair device!

Still checking:
Closing the lid & sleeping 
Headphone / mic port (3.5mm jack).

Also, Android-app support & Linux Beta is working flawlessly, device (phone) connected including smart-lock and such funny features appear to all work too! Even chromecasting and screen mirroring work. Had to uninstall both & once updated the needed components were also updated & everything installed as suppose to (see chrome://components)

Now here are the tweaks I've made and why:
The device seemed to boot and work with the created image, but shutdown & rebooting would not work. The device would still be powered on (screens backlight lit & surface cover illumination gave that away). The simple solution was to set the "acpi=force" kernel flag. Rebooting & shutting down are now working. Also I'm using the kernel options "enable_updates" and "advanced_als. Besides this I'm using the "i915.modset=true" flag.

So that's that, The Surface Pro 3 seems to be compatible with small effort. A huge thank you to everyone working on this amazing project.

SiemdeNijs, 2020-04-23:
chrome://flag
#enable-generic-sensor-extra-classes seems to make the ambient sensor work in combination with "advanced_als" ! Marvelous.

sebanc, 2020-04-23:
Hi,

Thanks a lot for this nice feedback :)

You have totally understood the philosophy of this project which is to provide an as solid as possible base kernel and to let everyone complete it with their specific device tweaks to have it working 100%.

Also, I am sure that other Surface Pro 3 device owners will find it very helpful.

Regarding wake from suspend, I think it is standard linux kernel behavior, my SP5 also only wakes with power button in both ChromeOS/Linux (actually i find it quite useful as it does not wake up when it's moved or in a bag...).

Thanks for the "#enable-generic-sensor-extra-classes" flag, I have to give it a try, it seems like a very interesting feature.

SiemdeNijs, 2020-04-23:
Glad to hear that the waking from sleep seems to be normal behavior, and indeed it's very it doesn't wake-up at just any given command 👍 .

Beside the #enable-generic-sensors-extra-classes I've enabled the flag "#auto-screen-brightness" now also, it seems to be more responsive but I'm unsure if this replaces the functionality now that advanced_als is enabled in kernel flags ? 

Beside those, I've come across a small bug: I've got no media controls in the ChromeOS media notification when playing from my main streaming (android) app. The album/song title show, but no media controls. 

I have found a sort of work around, I've enabled "#enable-media-session-service", 
And disabled "#enable-media-session-notifications", the audio/playback apps I've used have their own controls and own notifications to be controlled, so now it shows one notification which IS controllable by it :). I've also enabled "#hardware-media-key-handling" but this has not changed behavior of the dedicated play-pause function key (guessing it's just a different bind?). 

Here's a current list of all the flags I've edited, mentioned above are the kernel flag tweaks, beside that it seems to all work best at default. 

`Native Client
Support Native Client for all web applications, even those that were not installed from the Chrome Web Store. – Mac, Windows, Linux, Chrome OS, Android

#enable-nacl

Smooth Scrolling
Animate smoothly when scrolling page content. – Windows, Linux, Chrome OS, Android

#smooth-scrolling

Enable background blur.
Enables background blur for the Launcher, Shelf, Unified System Tray etc. – Chrome OS

#enable-background-blur

Show Bluetooth device battery
Enables showing the battery level of connected and supported Bluetooth devices in the System Tray and Settings UI. – Chrome OS

#show-bluetooth-device-battery

Pull-to-refresh gesture
Pull-to-refresh gesture in response to vertical overscroll. – Windows, Linux, Chrome OS

#pull-to-refresh

Terminal System App
Enables the Terminal System App at chrome://terminal which is used for the Chrome OS Linux terminal. – Chrome OS

#terminal-system-app

Cast Media Route Provider
Enables the native Cast Media Route Provider implementation to be used instead of the implementation in the Media Router component extension. – Mac, Windows, Linux, Chrome OS

#cast-media-route-provider

Generic Sensor Extra Classes
Enables an extra set of sensor classes based on Generic Sensor API, which expose previously unavailable platform features, i.e. AmbientLightSensor and Magnetometer interfaces. – Mac, Windows, Linux, Chrome OS, Android

#enable-generic-sensor-extra-classes

Crostini Usb Allow Unsupported
Allow mounting unsupported Usb devices in Crostini. At your own risk. – Chrome OS

#crostini-usb-allow-unsupported

Parallel downloading
Enable parallel downloading to accelerate download speed. – Mac, Windows, Linux, Chrome OS, Android

#enable-parallel-downloading

Enable Play Store search
Enable Play Store search in launcher. – Chrome OS

#enable-play-store-search

Enable app data search in launcher
Allow launcher search to access data available through Firebase App Indexing – Chrome OS

#enable-app-data-search

Enable Assistant App Support
Enable the Assistant App Support feature – Chrome OS

#enable-assistant-app-support

Assistant Media Session integration
Enable Assistant Media Session Integration. – Chrome OS

#enable-assistant-media-session-integration

Enable click to call feature signals to be handled on desktop
Enables click to call feature signals to be handled on desktop by showing a list of user's available devices with telephony functionality. – Mac, Windows, Linux, Chrome OS

#click-to-call-ui

Google Password Manager UI
Enables access to the Google Password Manager UI from Chrome. – Mac, Windows, Linux, Chrome OS, Android

#google-password-manager

Media Session Service
Enables the media session mojo service and internal media session support. – Mac, Windows, Linux, Chrome OS

#enable-media-session-service`

Hoping that these posts will help other Surface owners find their way to this amazing project, once again, thanks 😄

closed 2020-05-18 by sebanc

sharry1995, 2022-04-05:
hey can you please help me , how did you get camera and microphone to work , everything else is working for me