sebanc/brunch #63: USB won't boot

Z3r0shin, 2020-04-23:
Hello!

First, many many thanks for your work on this, its much appreciated. With the quarantine, schooling is done from home and installing your Brunch on a Asus All-in-One ET1620IUTT. With its touchscreen, the Play Store works really well and really helps with the kids. Also, Croissant is a clusterfuck of lost time trying to make stuff work. Steam doesn't work though, tried a bunch of things and nothing worked. It loads, but the whole window is black. It still works though, as in, if you know where to click, it works, but the window is completely black. Didn't try to really start a game, just in case I fuck something up with my Steam settings.

On to the subject. The issue I have is when trying to boot from USB on a MSI AC6200-021US it just doesn't. Here are its specs : 
MSI BIOS Version : E1681IMR VER.10C
Build Date : 04/12/2011
EC Version : 1681EMS1 Ver4.09
Build Date 12/18/2009
CPU Information
Processor Type : Intel(R) Core(TM) i5 CPU M 430 @ 2.27GHz
Processor Speed : 2266MHz
Processor Core : Dual
Memory Information
Total Memory : 8192 MB (DDR3:1067 MHz)
Like in your FAQs, it just shows a blinking line and stops there.
I've updated and even modified the BIOS to help, changed settings to accept legacy USB, efi devices, (there is no options to disable secure boot though) etc., used Plop both from a Linux Live USB to try and boot Brunch from USB. Tried using "#sudo bash chromeos-install.sh -l -src etc. -dst chromeos.img" but to no avail. Even installed Brunch on another computer and then transfered the SSD from that computer to the offensive laptop, but it didn't boot, stating that it requires a proper boot media (classic screen). If I take the same SSD and install it in yet another computer/laptop, it boots no problem. This laptop boot a CloudReady USB no problem though, I'm using it right now, but I'm having issues with gpu acceleration of Crostini (terminal won't open if I enable it, rendering Steam games completely useless to play with). This is the main reason why I want Brunch on it so I can try to make Steam work better.

I am new to this whole GitHub thing, even if I am rather techsavvy I never really participated in something. But this interests me a lot and I want to help. If there is anything I can do to help, just ask!

Thanks!

sebanc, 2020-04-23:
Hi,

Thanks for detailed description.

Unfortunately, on your main issue, from the looks of it, I am pretty sure that your laptop does not support UEFI. Brunch contrarily to Cloudready does not support legacy boot, to be honest because I don't know how to implement it and I did not find clear enough online resources to do it (the things I tried all failed)...
If you feel like trying, you can try to modify the partition 12 to support legacy boot.

Regarding the black screen in steam, normally disabling crostini gpu acceleration in chrome://flags should solve it and in your case it might not have that much impact on performance.

sebanc, 2020-05-11:
Hi,

Actually someone apparently found a way to make brunch work on legacy bios devices using the linux dual boot method:
1) Install a linux distro on your device
2) Follow linux dual boot method

sebanc, 2020-05-18:
Hi, brunch support for legacy devices ended up being discussed in issue #77, the implementation is not complete but don't hesitate to have a look at the progress that was made there.

closed 2020-05-18 by sebanc