sebanc/brunch #64: Keep getting Missing OS

bobowin, 2020-04-23:
Hello

I recently installed the Chrome OS into my Asus UX31E laptop .  Based on installation result from the Terminal.   It shown "ChormeOS installed" at the end. 
[image]
After the system reboot, my machine keep return "Missing OS" error message. 
[image]

Here is the installation detail: 
Chrome OS Image:  chromeos_12739.111.0_rammus_recovery_stable-channel_mp-v2.bin

Machine Detail: Asus zenbook U31E 
CPU: Intel i5-2557 
Memory: 4G
BIos : American MegaTrends 214 (2119.I13UX3.011)
UEFI Boot : Enabled. 

Could you suggest / advice? 
Thanks

sebanc, 2020-04-24:
Hi,

Thanks for the detailed report.

It seems some devices do not launch the default efi boot path (/efi/boot/bootx64.efi) on an internal HDD and expect a boot manager entry to be created.

Could you try creating the boot entry manually like in the below article (Entering the UEFI Boot Entry manually in BIOS part) ?
https://www.thomas-krenn.com/en/wiki/Restoring_UEFI_boot_entry_via_motherboard_replacement_or_BIOS_update

bobowin, 2020-04-25:
Thanks sebanc.  it worked and booted.  
But another problem coming up now; the touch pad not working.  
Thank you so much!

sebanc, 2020-04-25:
Apparently, there is an issue with touchpad on latest testing, could you try the latest r80 stable ?

bobowin, 2020-04-25:
Hi Sebanc
I am currently using this version "Brunch r80 k4.19 stable 20200407" ; touch pad did not response. 
temporary using USB mouse.

sebanc, 2020-04-25:
Could you post a dmesg for the touchpad issue ?

Run "dmesg > ~/Downloads/dmesg.txt" in crosh shell and then attach dmesg.txt from your downloads folder to this issue.

bobowin, 2020-04-25:
Hi sebanc
Thank you so much! 
[dmesg.txt](https://github.com/sebanc/brunch/files/4533816/dmesg.txt)

bobowin, 2020-04-29:
Hi @sebanc 
Any way I can solve the touchpad issue?  
Thanks

sebanc, 2020-04-30:
Hi, I have tried to change something which will be in the next release and might help (the next release is currently blocked by an issue unrelated to this one). However if it does not i am not sure I will be able to do something about it.

bobowin, 2020-05-02:
@sebanc 
Thanks.. will wait for next release.

closed 2020-05-02 by bobowin